lorabasics/basicstation #3: Connect to TTN console (legacy protocol)

b1r63r, 2019-02-04:
I am very new to LoRa tech (but very linux knowledgeable) so I may be ideal to sort out documentation issues :-D

Will the station2pkfwd stuff be what I need to connect my rpi/rak831 gateway to the TTN console? I understand that basicstation runs the new V3 protocol that TheThingsNetwork.org does not understand yet. And that I could have chosen a more well trodden path instead of testing basicstation as my first foray into LoRa. But it is more fun to test something new...

Will I have to install a packet forwarder on my pi in addition to basicstation? Or can I somehow point the code in the station2pkgfwd directly at some TTN endpoint? Googling doesn't really give me any answers.

johanstokking, 2019-02-04:
@b1r63r we're building native support for Basic Station in The Things Network Stack V3.

Please watch:

- https://github.com/TheThingsNetwork/lorawan-stack/issues/74
- https://github.com/TheThingsNetwork/lorawan-stack/issues/75

b1r63r, 2019-02-04:
@johanstokking Will this be implemented by the end of february? If so it is probably easier for me to just go back to one of the legacy installs until then and retry with basicstation later?

I have tried starting the packet forwarder and pointing it at lns.eu.thethings.network and then pointing the "live" sample at the socket provided. No errors that I can see, but nothing seems to get registered/connected either. I may still be missing some configuration.

johanstokking, 2019-02-04:
@b1r63r yes, please watch the issue including the milestone set, currently February.

Also, the approach you mention should in fact work.

CC @KrishnaIyer