lorabasics/basicstation #4: Add examples for CUPS and LNS config for s2.sm.tc

iBrick, 2019-02-06:
Hi,
1) could you add an example binary file to server s2.sm.tc which can be received by GW after request of update-info command?
2) could you add an example router_config that can be received from s2.sm.tc 
Using examples from repo I don't get this from test server s2.sm.tc, just <uri> info.

beitler, 2019-03-14:
We have added a [CUPS example](https://github.com/lorabasics/basicstation/tree/master/examples/cups) to the repository. The example allows to dissect the CUPS protocol in much detail. 
An example for router_config is contained in [tcutils.py](https://github.com/lorabasics/basicstation/blob/master/pysys/tcutils.py#L56).

iBrick, 2019-03-17:
Thank you! I've checked this and it works fine! 
What about second point with router_config example?

beitler, 2019-03-19:
I guess I'm not quite sure what exactly you are asking for. If you are interested in the exact `router_config` message which is sent down from the s2.sm.tc example server for your specific gateway, you can make use of tools like [wscat](https://www.npmjs.com/package/wscat):
**Step 1**: Retrieve personalized client credentials
```
$ cd examples/live-s2.sm.tc
$ make
^CTL+C
```
**Step 2**: Convert client credentials to PEM (wscat does not accept DER)
```
$ openssl x509 -inform der -in tc.crt -out tc.crt.pem
$ openssl x509 -inform der -in tc.trust -out tc.trust.pem
$ openssl ec -in tc.key -inform der -outform PEM -out tc.key.pem
```
**Step 3**: Connect to MUXS via wscat
```
$ wscat --ca tc.trust.pem --cert tc.crt.pem --key tc.key.pem \
  --connect https://s2.sm.tc:7001/router-<Station EUI>
connected (press CTRL+C to quit)
< {"msgtype":"router_config","regionid":1,"region":"EU863","hwspec":"sx1301\/1","max_eirp":16.0,"DRs":[[12,125,0],[11,125,0],[10,125,0],[9,125,0],[8,125,0],[7,125,0],[7,250,0],[0,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0]],"freq_range":[863000000,870000000],"upchannels":[[868100000,0,5],[868300000,0,5],[868500000,0,5]],"sx1301_conf":[{"radio_0":{"enable":true,"freq":868300000},"radio_1":{"enable":false,"freq":0},"chan_FSK":{"enable":false},"chan_Lora_std":{"enable":false},"chan_multiSF_0":{"enable":true,"radio":0,"if":-200000},"chan_multiSF_1":{"enable":true,"radio":0,"if":0},"chan_multiSF_2":{"enable":true,"radio":0,"if":200000},"chan_multiSF_3":{"enable":false},"chan_multiSF_4":{"enable":false},"chan_multiSF_5":{"enable":false},"chan_multiSF_6":{"enable":false},"chan_multiSF_7":{"enable":false}}],"config":{},"protocol":1,"bcning":{"DR":3,"layout":[2,8,17],"freqs":[869525000]},"NetID":null,"JoinEui":null,"nodc":true,"MuxTime":1552986432.3164999485}
```

Your `<Station EUI>` is printed during startup of Station. By default it will be derived from your MAC address and will look something like `b827:ebff:fexx:xx` on a Raspberry Pi. You can determine it like this `make |& grep "Station EUI"` and then use this value in the path of the WS endpoint.

iBrick, 2019-03-19:
That is exactly what I was asking for. Thank you!

closed 2019-03-19 by iBrick