lorabasics/basicstation #6: arduino esp32 or 12 support?

halukmy, 2019-02-12:
any info?

beitler, 2019-02-13:
Hello,

the architecture of Basic Station is optimized to make it easy to port it to Linux-based and RTOS-based platforms alike. In fact [The Things Indoor Gateway](https://www.thethingsnetwork.org/docs/gateways/thethingsindoor/) firmware is a variant of Basic Station whose SYS layer implementation integrates with the [ESP8266_RTOS_SDK v2.0](https://github.com/espressif/ESP8266_RTOS_SDK/tree/release/v2.x.x). Therefore, porting Basic Station to the ESP32 via the [ESP-IDF](https://github.com/espressif/esp-idf) is certainly feasible. Whether or not you want to make use of the Arduino core is up to you. I don't see an immediate benefit of the Arduino core in this context since the ESP-IDF already comes with all necessary device drivers, networking and crypto libraries.

beitler, 2019-03-14:
I think we can close this. Feel free to re-open in case you have a specific question on this topic.

closed 2019-03-14 by beitler