lorabasics/basicstation #7: The Things Indoor Gateway - not connected

ricaun, 2019-02-12:
Hello,

I'm from Brazil and I connected my TIG (US-915) this Sunday and it was so easy, everything was working great, uplink, downlink, etc.
But on the next day, I noticed the gateway was offline on the TTN console, but the led was solid green, no packages. 
I tried to turn off for some time, reset but nothing works. Then I decided to open the TIG and check the debug serial...

I found this error.

```
2019-02-11 21:11:08.546 [SYS:DEBU]   Free Heap: 19104 (min=16816) wifi=5 mh=7 cups=8 tc=4
2019-02-11 21:11:09.135 [AIO:DEBU] ssl_tls.c:6546 MBEDTLS[1]: mbedtls_ssl_read_record() returned -30848 (-0x7880)
2019-02-11 21:11:09.141 [AIO:ERRO] Recv failed: SSL - The peer notified us that the connection is going to be closed
2019-02-11 21:11:09.146 [AIO:DEBU] [2] WS connection shutdown...
2019-02-11 21:11:09.157 [TCE:VERB] Connection to MUXS closed in state 4
2019-02-11 21:11:09.165 [TCE:INFO] MUXS reconnect backoff 2s (retry 1)
```

I really have no idea what that's mean.

Someone?!

beitler, 2019-02-13:
Hello,

your log excerpt indicates that the server side closed a standing Websocket connection and a re-connection attempt is going to happen in 2 seconds.

According to the [documentation](https://www.thethingsnetwork.org/docs/gateways/thethingsindoor/#led-states) a solid green LED indicates that the Websocket connection to the LNS is established, the configuration is fetched, radio initialized and listening for packets. 

Discussions in the [TTN Forum](https://www.thethingsnetwork.org/forum/t/69-gemtek-gateway-the-things-indoor-gateway/22049/308) and on Slack indicate that multiple TTN users in the US915 region are affected by the issue you are raising where their gateways' (not only TIGs) connection status is not properly reflected in the TTN console. Since we cannot discuss LNS vendor specific issues here, I suggest you continue to provide feedback to TTN via their communication channels.

ricaun, 2019-02-13:
The TIG US 915 issue must solved!

closed 2019-02-13 by ricaun