lorabasics/basicstation #8: CUPS is a confusing name

jpmeijers, 2019-02-13:
Let's try and use a different name for the update and configuration server and protocol. CUPS is the name of a unix printing server. We're causing a lot of headaches for everyone by using this same name. Googling for CUPS will give results for the more common one, which is the printing server.

beitler, 2019-02-22:
Though SEO is not our primary concern, I agree with you that most people will associate the CUPS acronym with the Common Unix Printing System. From today's perspective it's not very likely that we will see large demand for printing-enabled LoRa Gateways in the future. Therefore, we can safely assume that talking about CUPS in the context of LoRa Gateways will not cause a lot of confusion.
Nevertheless, we value your feedback and will consider changing the name of the [Credentials and Update Management Protocol](https://doc.sm.tc/station/cupsproto.html) to something less ambiguous.

woobagooba, 2019-02-26:
Groan ... https://blogs.cisco.com/sp/mwc-series-ultra-packet-core-with-cups-to-deliver-5g-broadband-services-for-stc

closed 2021-07-21 by rbaldwin13