lorabasics/basicstation #10: Gateway Frequency Plan

otassain, 2019-02-21:
The frequency plan of the Gateway is exchanged with the LNS .
Is this file mandatory in the exchange sequence with the LNS when connecting or can it be stored directly in the Gateway and update later using CUPS?

beitler, 2019-02-22:
The frequency plan, as well as other setup information contained in the [`router_config` message](https://doc.sm.tc/station/tcproto.html#router-config-message), are currently not persisted by Station. Therefore, at the beginning of every connection to the LNS this information has to be exchanged. In fact, Station will wait for the `router_config` message before radio initialization is started. 
For the future, it is certainly conceivable to persist the most recent channel plan as a backup strategy for the case of a power outage followed by a transient backhaul connectivity loss. For now, this is not supported by Station.

closed 2019-02-22 by otassain

otassain, 2019-02-27:
Do you have an idea of when a new version will include the persistency?