lorabasics/basicstation #9: DevEUI needed for DL data

otassain, 2019-02-21:
What is the use of the device's devEUI field in the downlink messages sent to the Gateway ? 
Is it mandatory or can it be filled with a default value?

You can contact me back by mail otassain.ext@objenious.com

beitler, 2019-02-22:
The `devEUI` and `diid` (device interaction ID) fields are passed along by Station from the `dnmsg` to the `dntxed` message. Internally, these fields are only used for logging. Passing these fields through, allows the LNS to easily correlate the `dntxed` message to the internal state associated to the downlink transaction. If in your server architecture the device EUI is not available at the place where the `dnmsg` is constructed, you could pass along some other device-specific identifier, like the short address. Station just expects a string in [EUI](https://doc.sm.tc/station/glossary.html#term-eui) format to be present in the `devEUI` field.

otassain, 2019-02-22:
Thanks Anton.

closed 2019-02-22 by otassain