lorabasics/basicstation #11: 404 on connection attempt to CUPS-boot

woobagooba, 2019-02-22:
Multiple failed retries as below, and eventually segment faults and dies (which it should not do)

```
2019-02-22 23:13:16.253 [CUP:INFO] Starting a CUPS session
2019-02-22 23:13:16.254 [CUP:ERRO] No CUPS-bak URI configured
2019-02-22 23:13:17.254 [CUP:INFO] Starting a CUPS session
2019-02-22 23:13:17.254 [CUP:INFO] Connecting to CUPS-boot ... https://s2.sm.tc:7007 (try #3)
2019-02-22 23:13:17.256 [any:INFO] cert. version     : 1
serial number     : 0C
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:42:00
expires on        : 2024-11-06 12:42:00
signed using      : ECDSA with SHA256
EC key size       : 256 bits
2019-02-22 23:13:17.270 [any:INFO] cert. version     : 3
serial number     : 10
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=gwprov-::0, OU=TrackCentral, O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:43:52
expires on        : 2020-11-07 12:43:52
signed using      : ECDSA with SHA256
RSA key size      : 2048 bits
2019-02-22 23:13:17.270 [AIO:INFO]
2019-02-22 23:13:17.463 [CUP:VERB] Retrieving update-info from CUPS-boot https://s2.sm.tc:7007...
2019-02-22 23:13:18.112 [any:VERB] Failed to retrieve TCURI from CUPS: (404) Some exception occured!
2019-02-22 23:13:18.112 [AIO:DEBU] [3] HTTP connection shutdown...
2019-02-22 23:13:18.112 [CUP:INFO] Interaction with CUPS failed - retrying in 1s
2019-02-22 23:13:19.113 [CUP:INFO] Starting a CUPS session
2019-02-22 23:13:19.113 [CUP:ERRO] No CUPS URI configured
2
```

woobagooba, 2019-02-24:
The error condition has changed ... now looks like client certificate validation failure at CUPS?

`2019-02-24 15:21:53.860 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotat                                                                                                                                                             e=3)
2019-02-24 15:21:53.860 [SYS:INFO] Station Ver : 2.0.2(rpi/std) 2019-02-22 16:51                                                                                                                                                             :46
2019-02-24 15:21:53.861 [SYS:INFO] Package Ver : 1.0.0
2019-02-24 15:21:53.861 [SYS:INFO] proto EUI   : 0:b827:eba6:7a72       (/sys/cl                                                                                                                                                             ass/net/eth0/address)
2019-02-24 15:21:53.861 [SYS:INFO] prefix EUI  : ::1    (builtin)
2019-02-24 15:21:53.861 [SYS:INFO] Station EUI : b827:ebff:fea6:7a72
2019-02-24 15:21:53.861 [SYS:INFO] Station home: ./     (builtin)
2019-02-24 15:21:53.861 [SYS:INFO] Station temp: /var/tmp/      (builtin)
2019-02-24 15:21:54.063 [TCE:INFO] Starting TC engine
2019-02-24 15:21:54.076 [any:INFO] cert. version     : 1
serial number     : 01
issuer name       : CN=Root CA, OU=TrackCentral (eiaOvS+I), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaOvS+I), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:41:47
expires on        : 2024-11-06 12:41:47
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
2019-02-24 15:21:54.091 [any:INFO] cert. version     : 3
serial number     : 37
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-b827:ebff:fea6:7a72, OU=TrackCentral, O=TrackNet.i                                                                                                                                                             o, C=CH
issued  on        : 2019-02-23 12:47:55
expires on        : 2021-02-22 12:47:55
signed using      : ECDSA with SHA256
RSA key size      : 2048 bits
2019-02-24 15:21:54.091 [AIO:INFO]
2019-02-24 15:21:54.272 [TCE:INFO] Connecting to INFOS: wss://s2.sm.tc:7000
2019-02-24 15:21:54.272 [CUP:INFO] Starting a CUPS session
2019-02-24 15:21:54.272 [CUP:INFO] Connecting to CUPS ... https://s2.sm.tc:7007                                                                                                                                                              (try #1)
2019-02-24 15:21:54.274 [any:INFO] cert. version     : 1
serial number     : 01
issuer name       : CN=Root CA, OU=TrackCentral (eiaOvS+I), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaOvS+I), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:41:47
expires on        : 2024-11-06 12:41:47
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
2019-02-24 15:21:54.289 [any:INFO] cert. version     : 3
serial number     : 37
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-b827:ebff:fea6:7a72, OU=TrackCentral, O=TrackNet.i                                                                                                                                                             o, C=CH
issued  on        : 2019-02-23 12:47:55
expires on        : 2021-02-22 12:47:55
signed using      : ECDSA with SHA256
RSA key size      : 2048 bits
2019-02-24 15:21:54.289 [AIO:INFO]
2019-02-24 15:21:54.400 [CUP:VERB] Retrieving update-info from CUPS https://s2.s                                                                                                                                                             m.tc:7007...
2019-02-24 15:21:54.402 [AIO:DEBU] [3] WS connection shutdown...
2019-02-24 15:21:54.402 [TCE:INFO] INFOS reconnect backoff 0s (retry 0)
2019-02-24 15:21:54.403 [any:INFO] cert. version     : 1
serial number     : 01
issuer name       : CN=Root CA, OU=TrackCentral (eiaOvS+I), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaOvS+I), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:41:47
expires on        : 2024-11-06 12:41:47
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
2019-02-24 15:21:54.417 [any:INFO] cert. version     : 3
serial number     : 37
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-b827:ebff:fea6:7a72, OU=TrackCentral, O=TrackNet.i                                                                                                                                                             o, C=CH
issued  on        : 2019-02-23 12:47:55
expires on        : 2021-02-22 12:47:55
signed using      : ECDSA with SHA256
RSA key size      : 2048 bits
2019-02-24 15:21:54.417 [AIO:INFO]
2019-02-24 15:21:54.522 [TCE:INFO] Connecting to INFOS: wss://s2.sm.tc:7000
2019-02-24 15:21:54.523 [AIO:ERRO] [4] Send failed: X509 - Certificate verificat                                                                                                                                                             ion failed, e.g. CRL, CA or signature check failed
[... 46 lines cut ...]
2019-02-24 15:21:56.874 [CUP:VERB] Retrieving update-info from CUPS https://s2.sm.tc:7007...
[... cut ...]
2019-02-24 15:21:57.994 [any:INFO] cert. version     : 3
serial number     : 37
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-b827:ebff:fea6:7a72, OU=TrackCentral, O=TrackNet.io, C=CH
issued  on        : 2019-02-23 12:47:55
expires on        : 2021-02-22 12:47:55
signed using      : ECDSA with SHA256
RSA key size      : 2048 bits
2019-02-24 15:21:57.994 [AIO:INFO]
2019-02-24 15:21:58.100 [CUP:VERB] Retrieving update-info from CUPS https://s2.sm.tc:7007...
2`

woobagooba, 2019-02-24:
The issue has morphed into what looks like a client certificate validation failing.  Relevant log attached.

[BasicStationLog.txt](https://github.com/lorabasics/basicstation/files/2898138/BasicStationLog.txt)

acaracas, 2019-02-25:
The issue has been resolved on the server side. The s2.sm.tc example should work now.

closed 2019-02-25 by acaracas

woobagooba, 2019-02-25:
I just tried the example again.  It is still not working.  

2019-02-25 12:43:05.168 [CUP:INFO] Starting a CUPS session
2019-02-25 12:43:05.168 [CUP:ERRO] No CUPS-bak URI configured
2019-02-25 12:43:06.169 [CUP:INFO] Starting a CUPS session
2019-02-25 12:43:06.169 [CUP:INFO] Connecting to CUPS ... https://s2.sm.tc:7007 (try #48)
2019-02-25 12:43:06.169 [any:INFO] cert. version     : 1
serial number     : 01
issuer name       : CN=Root CA, OU=TrackCentral (eiaOvS+I), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaOvS+I), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:41:47
expires on        : 2024-11-06 12:41:47
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
2019-02-25 12:43:06.183 [any:INFO] cert. version     : 3
serial number     : 37
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-b827:ebff:fea6:7a72, OU=TrackCentral, O=TrackNet.io, C=CH
issued  on        : 2019-02-23 12:47:55
expires on        : 2021-02-22 12:47:55
signed using      : ECDSA with SHA256
RSA key size      : 2048 bits
2019-02-25 12:43:06.183 [AIO:INFO]
2019-02-25 12:43:06.288 [CUP:VERB] Retrieving update-info from CUPS https://s2.sm.tc:7007...
2019-02-25 12:43:06.396 [AIO:ERRO] [3] Send failed: X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
2019-02-25 12:43:06.396 [AIO:DEBU] [3] HTTP connection shutdown...
2019-02-25 12:43:06.397 [CUP:INFO] Interaction with CUPS failed - retrying in 1s

acaracas, 2019-02-25:
The credentials files cups.key, cups.crt, and cups.trust  have to be removed.
Essentially, one would have to try with a clean setup.

woobagooba, 2019-02-25:
You are recommending I delete my current basicstation clone, perform a new local clone + build, and run the live example ... correct?

acaracas, 2019-02-25:
Yes, either fresh git clone or "git clean -f".

woobagooba, 2019-02-25:
Deleted the basicstation folder structure.
Now git is failing (git clone git@github.com:lorabasics/basicstation.git) ... this is new behavior.
Permission denied (publickey).

ssh -vT git&github.com reveals ...

pi@raspberrypi:~ $ ssh -vT git@github.com
OpenSSH_7.4p1 Raspbian-10+deb9u4, OpenSSL 1.0.2l  25 May 2017
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to github.com [192.30.253.113] port 22.
debug1: Connection established.
debug1: key_load_public: No such file or directory
debug1: identity file /home/pi/.ssh/id_rsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pi/.ssh/id_rsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pi/.ssh/id_dsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pi/.ssh/id_dsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pi/.ssh/id_ecdsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pi/.ssh/id_ecdsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pi/.ssh/id_ed25519 type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pi/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_7.4p1 Raspbian-10+deb9u4
debug1: Remote protocol version 2.0, remote software version babeld-792c3f41
debug1: no match: babeld-792c3f41
debug1: Authenticating to github.com:22 as 'git'
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: rsa-sha2-512
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ssh-rsa SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8
debug1: Host 'github.com' is known and matches the RSA host key.
debug1: Found key in /home/pi/.ssh/known_hosts:1
debug1: rekey after 134217728 blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey after 134217728 blocks
debug1: SSH2_MSG_EXT_INFO received
debug1: kex_input_ext_info: server-sig-algs=<ssh-ed25519,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,ssh-rsa,rsa-sha2-512,rsa-sha2-256,ssh-dss>
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey
debug1: Next authentication method: publickey
debug1: Trying private key: /home/pi/.ssh/id_rsa
debug1: Trying private key: /home/pi/.ssh/id_dsa
debug1: Trying private key: /home/pi/.ssh/id_ecdsa
debug1: Trying private key: /home/pi/.ssh/id_ed25519
debug1: No more authentication methods to try.
Permission denied (publickey).

achopra6, 2020-08-14:
This issue still persists.