lorabasics/basicstation #12: The CUPS protocol

woobagooba, 2019-02-22:
What is the thinking behind the POST request using a JSON encoding and the response encoded as binary?

beitler, 2019-03-14:
The CUPS protocol basically consists of an HTTP POST request where the request has `Content-Type: application/json` and the response has `Content-Type: application/octet-stream`. On constrained platforms the response, being an octet-stream, very conveniently allows to stream the large firmware update from the socket read buffer directly into flash.
We added a [CUPS example](https://github.com/lorabasics/basicstation/tree/master/examples/cups) to the repository. Checkout the README to learn about how the CUPS protocol can be used. Read more about the CUPS protocol [in the documentation](https://doc.sm.tc/station/cupsproto.html).

beitler, 2019-05-21:
I hope the CUPS example clarifies the thinking behind the CUPS protocol. If not, please feel free to post specific questions to the [LoRa Developer Portal Forum](https://lora-developers.semtech.com/knowledge-base/forum/viewforum/33/). Thanks!

closed 2019-05-21 by beitler