lorabasics/basicstation #16: Picocell Gateway is not currently supported

kuanyili, 2019-03-25:
Although Picocell Gateway is an official Semtech product, it is not yet supported by Basic Station. What is the plan and schedule of support for Picocell Gateway?

beitler, 2019-04-11:
Support for the Picocell Gateway will be added in one of the next releases.

donetdisturb, 2019-05-06:
https://www.n-fuse.co/devices/LoRaWAN-Concentrator-Card-mini-PCIe.html
based on
https://www.semtech.com/uploads/documents/picocell_gateway_user_guide.pdf

I hope for a quick implementation, currently it doesn't work ..

donetdisturb, 2019-11-10:
> Support for the Picocell Gateway will be added in one of the next releases.

Hello @beitler , are there any news about it? Thank you.

beitler, 2019-11-12:
The next release is in the pipeline and it will add support for the picocell reference design.

ogimenez-smtc, 2020-11-10:
Hi,

It has been added in release [V2.0.4](https://github.com/lorabasics/basicstation/releases/tag/v2.0.4-9-g3d5c686) and documenation in the [Readme](https://github.com/lorabasics/basicstation#picocell-gateway-linux-os-as-host--sx1308-usb-reference-design) so I close this issue.

closed 2020-11-10 by ogimenez-smtc