lorabasics/basicstation #17: msgtype:  version - features is documented as list, but string is received

brocaar, 2019-04-01:
https://doc.sm.tc/station/tcproto.html#version-message

The first message sent by Station is a version message reporting to the LNS its version information (station, firmware, model), the protocol version (the constant 2 for now), **and a list of optional features supported by the gateway**:

```
{
  "msgtype"   : "version"
  "station"   : STRING
  "firmware"  : STRING
  "package"   : STRING
  "model"     : STRING
  "protocol"  : INT
  "features"  : [ STRING, .. ]
}
```

However, the received JSON message does not contain a list for `features`, instead it contains a string.

Received from gateway:

{\"msgtype\":\"version\",\"station\":\"2.0.0(minihub/debug)\",\"firmware\":\"2.0.0\",\"package\":\"2.0.0\",\"model\":\"minihub\",\"protocol\":2,\"features\":\"rmtsh\"}

beitler, 2019-04-03:
Thanks for pointing this out. The value of the `features` field is in fact a `STRING` containing space-separated keywords. I fixed to doc to reflect that properly.

brocaar, 2019-04-03:
Thanks for the clarification. However, I think an array strings would have been easier to work with as the consumer could directly iterate over it.

closed 2019-04-15 by brocaar