lorabasics/basicstation #19: DRs array index maps to DR number?

brocaar, 2019-04-03:
> The field DRs defines the available data rates of the channel plan. It MUST be an array of 16 entries with each entry being an array of three (3) integers encoding the spreading factor SF, the bandwidth BW, and a DNONLY flag. SF MUST be in the range 7..12 for LoRa, or 0 for FSK. BW MUST be one of the numbers 125, 250, 500, and BW is ignored for FSK. DNONLY MUST be 1 if the data rate is valid for downlink frames only, and 0 otherwise.

I make the assumption that the index of the array represents the DR number as specified in the LoRaWAN Regional Parameters, but from the documentation this is not clear. It is also not clear what to do with the unspecified data-rates. From the examples I see that in such a case the SF is set to `-1`, but this is not documented. Both points might be worth adding :)

johanstokking, 2019-04-03:
>I make the assumption that the index of the array represents the DR number as specified in the LoRaWAN Regional Parameters

Yes. What's confusing here is that this makes the gateway aware of LoRaWAN regional parameters, while they are subject to change and regional parameters are a contract between LNS and device; the gateway plays no role there.

Subject to change is not only adding data rates, but also breaking changes from data rate index to SF/BW values, like what happened in AU915.

So what we do in TTN is that we use the latest PHY version know to NS, and map the incoming data rate index back to that, and then based on the end device's PHY version in question, calculate the real data rate index, which may be different. This would not have been necessary if BS would just send the SF and BW values and didn't care about regional parameters.

brocaar, 2019-04-03:
I'm doing exactly the same mapping at my side :)

woobagooba, 2019-04-03:
The situation you've both described appears to make a very unwanted dependency between the BasicStation and Regional Parameters Spec (which as you point out is subject to ongoing changes).  Excellent that you've made the correction assumptions and mapping, but BasicStation should be unambiguously clarified on this point and the dependency removed.  

What do you recommend as specific improvements to resolve this issue?

Thank you

brocaar, 2019-04-03:
As @johanstokking also proposed, I think too it is better to communicate the data-rate as the actual parameters, instead of the DR numbers.

To remove this dependency, I believe the following payloads would need to change:

* https://doc.sm.tc/station/tcproto.html#radio-meta-data --> `DR`
* https://doc.sm.tc/station/tcproto.html#single-frame-downlink --> `DR`, `RX1DR` and `RX2DR`
* https://doc.sm.tc/station/tcproto.html#multicast-schedule --> `DR`

These DR keys would then be replaced by the modulation type + parameters related to this modulation (bandwidth, spreading factor, ...).

johanstokking, 2019-04-03:
Agreed; adding fields is also fine not to break API.

woobagooba, 2019-05-03:
What say Semtech on this point?

closed 2021-07-21 by rbaldwin13