lorabasics/basicstation #21: Typo / rendering issue: sx1301/**N**

brocaar, 2019-04-03:
https://doc.sm.tc/station/tcproto.html#router-config-message

[image]

The `sx1301/**N**` should be rendered as `sx1301/N` I believe :-)

Is this key really needed? Can't the Basic Station derive this from the `sx1301_conf` JSON key as I believe `N == length(sx1301_conf)`?

beitler, 2019-04-03:
The `hwspec` string is passed to the Radio Abstraction Layer implementation which uses it to evaluate how the channel plan is allocated onto the hardware. For example, in an 8-ch full diversity setup, `N == 2` and `length(sx1301_conf) == 1`. By this means, the server can influence the channel allocation procedure. Forward-looking, this string will probably contain more information that just `sx1301/N`, but that is how it is used right now.

brocaar, 2019-04-03:
Just to confirm, with 8-ch full diversity setup you mean two antennas and SX1301 chips that are listening on the same 8 channels?

Given the documentation below, I would expect two `sx1301_conf` elements which each contain the same channels.

> The field sx1301_conf defines how the channel plan maps to the **individual SX1301 chips**. Its value is an array of SX1301CONF objects. **The number of array elements MUST be in accordance with the value of the field hwspec.**

beitler, 2019-04-03:
> Just to confirm, with 8-ch full diversity setup you mean two antennas and SX1301 chips that are listening on the same 8 channels?

Correct.

> Given the documentation below, I would expect two `sx1301_conf` elements which each contain the same channels.

Yes, I must retract my example (full diversity is actually inferred when `hwspec="sx1301/1"` while hardware capabilities are `sx1301/2`). In fact, currently, the condition `N == length(sx1301_conf)` is enforced as the documentation says and can serve as a consistency check. The hwspec field will gain more relevance in future releases.

closed 2019-04-15 by brocaar