lorabasics/basicstation #31: Server-side disconnect causes Basic Station to crash

brocaar, 2019-04-15:
This is probably a bug.

I noticed that the Basic Station crashes when the server closes the Websocket connection (e.g. because read timeout, termination / restart of the LNS, ...). The following log is printed:

```
*** Error in `./station': free(): invalid pointer: 0x00a7d728 ***
Aborted
```

---

I have compiled the latest (https://github.com/lorabasics/basicstation/commit/c29b8502f8c715daecec6666835da6e981dc820a) release on a Raspberry Pi 3.

beitler, 2019-04-15:
Thanks for reporting this. I couldn't reproduce the problem on my pi3.
To help us pin this down, could you please provide more details, like `gcc --version`,
 * Please paste the whole log from beginning to end, starting station with the `-p` argument
 *

brocaar, 2019-04-16:
I've just sent you an e-mail with login details for an environment where you can reproduce the issue.

beitler, 2019-04-17:
Thank you, Orne.

I was able to reproduce the problem on my Pi3 replicating the settings you are using. The problem is rooted in a bug which surfaces when the JoinEui filter list is parsed. This will be fixed in the next release. Until then, the JoinEui filtering feature should not be used. The recommended way is to leave out the `JoinEui` field in the `router_config` message. This will disable JoinEui filtering and Station will route all join requests.

jawadiot, 2020-05-13:
Hi @beitler ,

Do you mean, we should leave out the JoinEui field in gateway_bridge.toml ?

beitler, 2020-05-13:
If that has the effect that the `JoinEui` field is not sent down to the gateway, then yes.

jawadiot, 2020-05-13:
it's really weard  ! 

**Gateway bridge conf :** 

[general]
log_level=4
[filters]
net_ids=[]
join_euis=[]

But i still see the filter on ./station log :     

`pi@B827EBFFFEF65A01:/opt/basicstation/examples/live-s2.sm.tc $ RADIODEV=/dev/spidev0.0 
 ../../build-rpi-std/bin/station2020-05-13 20:26:41.819 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2020-05-13 20:26:41.819 [SYS:INFO] Station Ver : 2.0.3(rpi/std) 2020-05-13 11:08:10
2020-05-13 20:26:41.819 [SYS:INFO] Package Ver : 1.0.0
2020-05-13 20:26:41.820 [SYS:INFO] proto EUI   : 0:b827:ebf6:5a01	(/sys/class/net/eth0/address)
2020-05-13 20:26:41.820 [SYS:INFO] prefix EUI  : ::1	(builtin)
2020-05-13 20:26:41.820 [SYS:INFO] Station EUI : b827:ebff:fef6:5a01
2020-05-13 20:26:41.820 [SYS:INFO] Station home: ./	(builtin)
2020-05-13 20:26:41.820 [SYS:INFO] Station temp: /var/tmp/	(builtin)
2020-05-13 20:26:41.820 [SYS:WARN] Station in NO-CUPS mode
2020-05-13 20:26:42.021 [TCE:INFO] Starting TC engine
2020-05-13 20:26:42.022 [TCE:INFO] Connecting to INFOS: ws://127.0.0.1:3001
2020-05-13 20:26:42.025 [TCE:INFO] Infos: b827:ebff:fef6:5a01 b827:ebff:fef6:5a01 ws://127.0.0.1:3001/gateway/b827ebfffef65a01
2020-05-13 20:26:42.025 [AIO:DEBU] [3] ws_close reason=1000
2020-05-13 20:26:42.025 [AIO:DEBU] Echoing close - reason=1000
2020-05-13 20:26:42.025 [AIO:ERRO] Recv failed: NET - Connection was reset by peer
2020-05-13 20:26:42.025 [AIO:DEBU] [3] WS connection shutdown...
2020-05-13 20:26:42.026 [TCE:VERB] Connecting to MUXS...
2020-05-13 20:26:42.026 [TCE:VERB] Connected to MUXS.
2020-05-13 20:26:42.036 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2020-05-13 20:26:42.036 [RAL:DEBU] SX1301 txlut table (0 entries)
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=867500000 rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=868500000 rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=0 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=0 freq=400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=1 freq=-200000 bandwidth=2 datarate=2 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 ifchain  9: enable=0 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-05-13 20:26:42.036 [RAL:VERB] SX1301 LBT not enabled
2020-05-13 20:26:42.036 [RAL:INFO] Station device: /dev/spidev0.0 (PPS capture disabled)
2020-05-13 20:26:44.950 [S2E:INFO] Configuring for region: EU863 -- 863.0MHz..870.0MHz
2020-05-13 20:26:44.950 [S2E:VERB]   DR0  SF12/BW125 
2020-05-13 20:26:44.950 [S2E:VERB]   DR1  SF11/BW125 
2020-05-13 20:26:44.950 [S2E:VERB]   DR2  SF10/BW125 
2020-05-13 20:26:44.950 [S2E:VERB]   DR3  SF9/BW125 
2020-05-13 20:26:44.950 [S2E:VERB]   DR4  SF8/BW125 
2020-05-13 20:26:44.950 [S2E:VERB]   DR5  SF7/BW125 
2020-05-13 20:26:44.950 [S2E:VERB]   DR6  SF7/BW250 
2020-05-13 20:26:44.950 [S2E:VERB]   DR7  FSK 
2020-05-13 20:26:44.950 [S2E:VERB]   DR8  undefined
2020-05-13 20:26:44.950 [S2E:VERB]   DR9  undefined
2020-05-13 20:26:44.950 [S2E:VERB]   DR10 undefined
2020-05-13 20:26:44.950 [S2E:VERB]   DR11 undefined
2020-05-13 20:26:44.950 [S2E:VERB]   DR12 undefined
2020-05-13 20:26:44.950 [S2E:VERB]   DR13 undefined
2020-05-13 20:26:44.950 [S2E:VERB]   DR14 undefined
2020-05-13 20:26:44.950 [S2E:VERB]   DR15 undefined
2020-05-13 20:26:44.950 [S2E:VERB]   TX power: 16.0 dBm EIRP
2020-05-13 20:26:44.950 [S2E:VERB]             27.0 dBm EIRP for 869.4MHz..869.65MHz
2020-05-13 20:26:44.950 [S2E:VERB]   JoinEui list: 10 entries
2020-05-13 20:26:44.950 [S2E:VERB]   NetID filter: 00000001-00000001-00000001-00000001
2020-05-13 20:26:44.950 [S2E:VERB]   Dev/test settings: nocca=0 nodc=0 nodwell=0
2020-05-13 20:27:26.954 [SYN:INFO] MCU/SX1301 drift stats: min: 0.0ppm  q50: 1.0ppm  q80: 3.8ppm  max: -9.0ppm - threshold q90: 6.2ppm
2020-05-13 20:27:26.954 [SYN:INFO] Avg MCU drift vs SX1301#0: 1.0ppm
2020-05-13 20:27:42.027 [AIO:WARN] [3] Ignoring WS ping/pong message
2020-05-13 20:27:45.856 [SYN:INFO] Time sync qualities: min=41 q90=70 max=87 (previous q90=2147483647)
free(): invalid pointer
Aborted`

Maybe  there is  somthing else to configure !

beitler, 2020-05-13:
Judging by the log
```
2020-05-13 20:26:44.950 [S2E:VERB]   JoinEui list: 10 entries
```
apparently the JoinEui list is still sent.
I guess that would be a question for the chirpstack forum.

jawadiot, 2020-05-13:
right !  thanks for your help :)

ogimenez-smtc, 2020-11-10:
Hi,

Fixed by 817e6e1, release V2.0.4.
I am closing this issue.

closed 2020-11-10 by ogimenez-smtc