lorabasics/basicstation #32: "muxs" uniqueness and meaning?

gotthardp, 2019-05-03:
The LNS is requested to return a `muxs` identifier, but the documentation does not say if/when the number may be reused and how is it used. It seems to be never sent in any other message. What is the purpose of this, please?

(Please suggest a better way to ask questions if this is not the appropriate way.)

beitler, 2019-05-15:
I assume you are referring to the `muxs` field in the [discovery response of the LNS protocol](https://tracknet.bitbucket.io/station2/tcproto.html#querying-the-lns-connection-end-point-discovery). In the Basic Station protocol every resource has an identifier. The value of `muxs` is the identifier of the LNS connection endpoint which was assigned to your gateway during the discovery phase. Currently, this value is purely informative and is just printed into the logs.

gotthardp, 2019-05-15:
Clear, thanks!

closed 2019-05-15 by gotthardp