lorabasics/basicstation #33: LNS protocol errors?

gotthardp, 2019-05-07:
The LNS protocol allows the server to return an `error` field only during the initial router-URI exchange. The error handling for other message exchanges is not defined, which feels inconsistent. It would be more debugging friendly if the server was allowed to return an error field in every message.

beitler, 2019-05-15:
The `error` field in the [discovery response of the LNS protocol](https://doc.sm.tc/station/tcproto.html#querying-the-lns-connection-end-point-discovery) can be used to tell the user why the LNS is refusing to assign it to a connection endpoint. This is purely informative. Station will trigger the same action independent of the content of that string.

In most cases the server will handle an error condition by closing the connection. Prior to that, the server may send an unsolicited error message down in order to inform the user about the reason. In any case, Station does not have many more options than going into a re-connection back-off loop. The server may also opt to resolve an error condition by forcing a reboot or triggering a CUPS transaction in which case Station has the possibility to fetch a firmware update.

gotthardp, 2019-05-15:
I tried to *send an unsolicited error message down in order to inform the user about the reason*, but it was never displayed by the basicstation (in case of exchanges other than the discovery response). Only a parse error was displayed. Is this something that should work?

beitler, 2019-05-16:
Correct. Right now this is not directly foreseen in the protocol. It would require something like this:
```patch
diff --git a/src/s2e.c b/src/s2e.c
index 54f8a41..1dacc7c 100644
--- a/src/s2e.c
+++ b/src/s2e.c
@@ -1478,6 +1478,22 @@ int s2e_onMsg (s2ctx_t* s2ctx, char* json, ujoff_t jsonlen) {
         s2e_handleRmtsh(s2ctx, &D);
         break;
     }
+    case J_error: {
+        ujcrc_t  field;
+        while( (field = uj_nextField(&D)) ) {
+            switch(field) {
+            case J_error: {
+                LOG(MOD_S2E|WARNING, "LNS ERROR: %s", uj_str(&D));
+                break;
+            }
+            default: {
+                uj_skipValue(&D);
+                break;
+            }
+            }
+        }
+        break;
+    }
     default: {
         // Platform specific commands
         if( !s2e_handleCommands(msgtype, s2ctx, &D) )

```
which will print the message
```json
{"msgtype": "error", "error": "LNS out of resources. Closing connection."}
```
into the logs like this:
```
[S2E:WARN] LNS ERROR: LNS out of resources. Closing connection.
```

If this solves a problem for you I don't see a reason why we cannot add this optional message type in a future release. However, as stated earlier, at this point in time I don't think that Station should act upon these error messages and implement a variety of error handling mechanisms simply because Station does not have many other options than to go into a re-connection back-off loop. These messages should be treated as purely informative.

gotthardp, 2019-05-16:
Thanks. It's not a big deal; I am trying to highlight (what I consider to be) protocol inconsistencies and I appreciate your feedback. On the server side I can close the connection very easily; I just thought that having a message with a reason for closing (in all situations) might help people debugging the gateways without access to the server logs.

ogimenez-smtc, 2020-11-10:
Hi,

Fixed by 817e6e1, release V2.0.4.
I am closing this issue.

closed 2020-11-10 by ogimenez-smtc