lorabasics/basicstation #35: Why splitting the frame?

gotthardp, 2019-05-16:
Please note that splitting the frame into pieces doesn't help the server at all. The server is required to verify the MIC based on the entire Frame, so it has to parse the JSON content and **reconstruct** the frame back to its original form. The reconstruction requires parsing the dash-separated EUI back to a binary form, which (at least in my case) is more complicated than if I had to parse the binary frame on my own.

gotthardp, 2019-05-16:
Just an idea: it may be more useful to pass to the LNS **both** the split parts as well as the original payload to make the MIC check easier.

beitler, 2019-05-16:
The Basic Station protocol conveniently splits up the PHYPayload into its individual, high-level components. This makes the protocol easy to read and follow along by humans which helps with debugging.
It's true that for join request verification (and a few other operations) there is some overhead involved in rebuilding the binary payload. However, the overhead in shifting a few bytes around should be negligible. In Python it would look somewhat like this:

```python
import struct

msg = {
  "msgtype": "jreq",
  "MHdr": 0,
  "JoinEui": "AB-AB-AB-AB-AB-AB-AB-AB",
  "DevEui": "CD-CD-CD-CD-CD-CD-CD-CD",
  "DevNonce": 62160,
  "MIC": -500336777
  # ...
}

eui2int = lambda eui: struct.unpack_from('>q',bytes.fromhex(eui.replace("-","")))[0]

payload = struct.pack("<BqqH", 
    msg["MHdr"],
    eui2int(msg["JoinEui"]),
    eui2int(msg["DevEui"]),
    msg["DevNonce"] & 0xFFFF
)
```

I don't see a real demand to send redundant information inside the message just to save the server from executing these lines of code. The other option is to go back to always sending the whole PHYPayload as a base64 encoded string. However this clashes with the protocol's design principles stated above.

gotthardp, 2019-05-16:
But in a downlink message, there is a `pdu` and a `DevEui` only. If a person is using the parsed fields (like FCnt, Port or others) for debugging, it will be seeing only a half of the communication, which may not be helpful and feels inconsistent. (No big deal, I just wanted to provide a feedback ;-)

woobagooba, 2019-05-17:
Nominal operational efficiency over the backhaul link should be first priority.  Particularly for cellular links which will be widely used.  Let the server or gateway handle the debug formatting of the binary data.

Unless there is a very compelling reason, apply the same idiom for both up and down link.  

Related to the first point ... has a binary encoding for the protocol elements been considered?  CBOR, Protobuf, etc. ?

beitler, 2019-05-21:
It is true that the choice for JSON encoding is not an optimal one in terms of backhaul link capacity. This choice was driven by other optimization dimensions. Going forward, Protobuf is being considered a viable encoding alternative for the next evolution of the protocol. For the large volume of messages Protobuf can bring around 75% reduction in message size compared to JSON. However, going for binary encoding we lose easy inspectability of the message stream for troubleshooting. This is a trade-off which needs to be carefully considered since gateway operations (in fact, network operations in general) requires efficient ways to troubleshoot message flows. JSON has its benefits in this regard.

woobagooba, 2019-05-21:
All network protocols and technologies require field debug-ability, this is nothing special about BasicStation.  Efficiency over cellular links IMO is priority #1.  Provide the debug decode in the GW, the NS, or Wireshark-like captures / decoders.

closed 2019-07-31 by beitler