lorabasics/basicstation #37: Missing confirmation of DR/freq used

gotthardp, 2019-05-20:
The basicstation performs a TX window selection, and sends a Transmit Confirmation (dntxed) when the transmission is done.
Would you mind extending the "dntxed" to indicate which TX window (what frequency and DR) was selected for the transmission? The server may want to calculate the gateway TX duration and select a GW (if multiple gateways are used) based on dwell constraints.

beitler, 2019-05-21:
Thanks for pointing this out. Indeed, it makes sense to reflect the DR and Freq of the transmission back to the LNS. We will add this in the next release. In the meantime you can apply this patch on v2.0.3 to add that feature:
```patch
diff --git a/src/s2e.c b/src/s2e.c
index 54f8a41..ac110b9 100644
--- a/src/s2e.c
+++ b/src/s2e.c
@@ -272,6 +272,8 @@ static void send_dntxed (s2ctx_t* s2ctx, txjob_t* txjob) {
                   "msgtype",   's', "dntxed",
                   "seqno",     'I', txjob->diid,    // for older servers (remove if obsoleted)
                   "diid",      'I', txjob->diid,    // newer servers
+                  "DR",        'i', txjob->dr,
+                  "Freq",      'I', txjob->freq,
                   "DevEui",    'E', txjob->deveui,
                   "rctx",      'i', txjob->txunit,  // antenna that sent this frame
                   "xtime",     'I', txjob->xtime,
```

ogimenez-smtc, 2020-11-10:
Hi,

Fixed by 817e6e1, release V2.0.4.
I am closing this issue.

closed 2020-11-10 by ogimenez-smtc