lorabasics/basicstation #38: The meaning of RxDelay?

gotthardp, 2019-05-20:
There is RX1DR and RX2DR and also RX1Freq and RX2Freq, but there is just a single RxDelay although the protocol defines different delay for RECEIVE_DELAY1 and RECEIVE_DELAY2 (and similarly JOIN_ACCEPT_DELAY1 and JOIN_ACCEPT_DELAY2). What is the meaning of RxDelay, please? (I think it should be documented.)

beitler, 2019-05-21:
The `RxDelay` field in the `dnmsg` message of the Basic Station protocol is the same parameter as the `RxDelay` field contained in the `JoinAccept` message in the LoRaWAN protocol. This is enough to qualify the timing of a downlink transmission relative to the reception time because in the default case RX2 is one second after RX1.

Note that the automatic RX2 diversion is purely optional. You can always leave RX2 downlink scheduling to the LNS with your own custom delays.

Going forward, please take questions like this to our just opened [Developer Portal](https://lora-developers.semtech.com/knowledge-base/forum/viewforum/33/). Thanks!

gotthardp, 2019-05-21:
In other words, you always assume the RX2 delay is RX1 + 1 second, right?

beitler, 2019-05-21:
Your use of the word *always* seems to imply that there is no way to implement custom delays which would be a false conclusion. This is why I want to clarify: Basic Station assumes the default case of `RX2 = RX1 + 1s` whenever you make use of the RX2 diversion feature (by specifying `RX2DR` and `RX2Freq` in the `dnmsg` message). The LNS has the option to schedule custom RX2 transmissions by specifying a `dnmsg` message without `RX2Freq` and `RX2DR` but setting the fields `RxDelay`, `DR` and `Freq` according to your RX2 transmission parameters.
This is how it can be done today. In the future one could think about adding an optional `Rx2DelayDelta` field to the `dnsmg` message. However, there has not been any demand for this so far.

closed 2019-07-31 by beitler

danigian, 2022-01-13:
@beitler, it would be great to have some clearer documentation on the RXDelay field.
At the beginning I had the impression that, in order to force usage of second window, I would have needed to set in dnmsg the RXDelay field to the RX2 Delay and specify RX2DR/RX2Freq only.

I think documentation should clearly point out that "RXDelay is corresponding to the field in JoinAccept, being the delay between RX and opening of RX1 Window. In addition, Basic Station is doing calculation of the RX2 Win based on that RXDelay field + 1s"

beitler, 2022-01-13:
Thanks for this feedback. We will add the following clarifications to the docs:

> Station will choose either RX1 or RX2 transmission parameters, where RX1 is preferred.  If the frame arrives too late, or the transmission spot is blocked, it will try RX2. To force a decision, the LNS can omit either the RX1 or the RX2 parameters. **In any case, ``RxDelay`` is the delay between end of uplink transmission and start of the RX1 receive window of the device. The exact transmission end time is implicitly encoded in ``xtime``. ``RxDelay`` should have the same value as the ``RxDelay`` field in the join accept message. Station will automatically infer the RX2 delay by adding one second.**

and 

> A **Class C downlink** frame which answers an uplink and is aimed at RX1, looks almost identical to Class A ``dnmsg`` messages.  The only difference is the field ``dC``, which declares a Class C interaction.  If the transmission cannot make the RX1 opportunity, the Station will switch over to RX2 parameters and choose a convenient time to send the frame.  There is a maximum time Station will postpone transmission before the frame is dropped. **Since RX2 downlink opportunities are arbitrary for class C devices ``RxDelay`` is only used to calculate the RX1 opportunity in this case.**

Does that help?

danigian, 2022-01-13:
That would be perfect, thank you!