lorabasics/basicstation #40: Example for ESP8266

jpmeijers, 2019-06-04:
The Things Indoor Gateway makes use of an ESP8266 speaking the basicstation protocol to the backend. See: https://www.thethingsnetwork.org/forum/t/ttig-the-things-indoor-gateway/22049

It is therefore possible to use this basicstation software on an ESP8266. Could you please give us an example of how to do this, or opensource the TTIG's firmware. It would be tremendously useful to be able to build our own ESP8266 or ESP32 based gateways that are full 8 channels, not single channel gateways. Using an ESP8266 or ESP32 is muh cheaper than using a Raspberry Pi, so make deploying more gateways possible.

beitler, 2019-06-11:
Maintaining and supporting open source platform-specific ports is a resource consuming task. In the current stage of the project we are still focusing our resources to develop, harden and document core components of Basic Station. I'm convinced this strategy serves the project best at this point. For this and other reasons the TTIG firmware is not open source for the time being.

Nevertheless, the repository contains a reference platform layer implementation for Linux. This allows Basic Station to run on a vast number of Linux-based gateways out of the box. In addition to that, the Linux platform layer implementation can serve as a reference to move to other platforms. For example, [sys_linux.c](https://github.com/lorabasics/basicstation/blob/master/src-linux/sys_linux.c) is a good starting point for porting Basic Station to an embedded platform. You will find that for a minimal example many functions can be removed or replaces by strings. You will be left with a sequence of calls to initialization functions before executing the `aio_loop`.

It would be great to see the community contribute support for different platforms. This is certainly the more scalable and more sustainable model. Hence, feel free to fork the repo and play around. We are happy for any feedback which enhances the portability of the code.

closed 2019-07-31 by beitler

brainstorm, 2019-11-20:
> (...) For this and other reasons the TTIG firmware is not open source for the time being

Can you briefly disclose timelines for that TTIG firmware to be open sourced? I totally second @jpmeijers on his words. I do personally think that the whole LoRa ecosystem would be greatly enhanced if people could build RTOS-based low consumption gateways. 

Looked from a pure market perspective/strategy, other networks such as [Amazon Sidewalk](https://blog.aboutamazon.com/devices/introducing-amazon-sidewalk) will engulf all the good work if LoRa infrastructure is not deployed quickly enough. Keeping those great advances proprietary might be unnecessarily slowing down the type of (viral?) widespread LoRaWan adoption everyone is interested on.

mluis, 2020-02-22:
+1 for the TTIG OSS.

d-a-v, 2021-05-05:
Is there any change on plans to open-source TTIG firmware ?