lorabasics/basicstation #41: Time sync rejected ---

sgodey, 2019-06-10:
I am seeing the below messages when I am bringing up Gateway. Any ideas what is going wrong here?

-------------------

2019-06-10 07:07:04.840 [SYN:INFO] Avg MCU drift vs SX1301#0: 1.0ppm
2019-06-10 07:07:09.041 [SYN:VERB] Time sync rejected: quality=266 threshold=230
2019-06-10 07:07:11.142 [SYN:VERB] Time sync rejected: quality=399 threshold=230
2019-06-10 07:07:28.994 [SYN:INFO] Time sync qualities: min=70 q90=220 max=399 (previous q90=230)
2019-06-10 07:07:49.996 [SYN:INFO] MCU/SX1301 drift stats: min: -1.9ppm  q50: -4.3ppm  q80: -5.7ppm  max: -18.6ppm - threshold q90: -10.5ppm
2019-06-10 07:07:49.997 [SYN:INFO] Avg MCU drift vs SX1301#0: 1.0ppm
2019-06-10 07:08:32.001 [SYN:INFO] Time sync qualities: min=117 q90=139 max=146 (previous q90=220)
2019-06-10 07:08:32.001 [SYN:INFO] MCU/SX1301 drift stats: min: -1.0ppm  q50: -4.3ppm  q80: -5.2ppm  max: -7.6ppm - threshold q90: -7.1ppm
2019-06-10 07:08:32.001 [SYN:INFO] Avg MCU drift vs SX1301#0: 1.0ppm
2019-06-10 07:09:01.404 [SYN:VERB] Time sync rejected: quality=180 threshold=139
2019-06-10 07:09:16.106 [SYN:INFO] MCU/SX1301 drift stats: min: -1.4ppm  q50: -4.3ppm  q80: -5.2ppm  max: -6.2ppm - threshold q90: -5.7ppm
2019-06-10 07:09:16.106 [SYN:INFO] Avg MCU drift vs SX1301#0: 1.0ppm
2019-06-10 07:09:35.008 [SYN:INFO] Time sync qualities: min=118 q90=137 max=180 (previous q90=139)
2019-06-10 07:09:51.810 [SYN:VERB] Time sync rejected: quality=138 threshold=137
2019-06-10 07:10:00.211 [SYN:INFO] MCU/SX1301 drift stats: min: -3.3ppm  q50: -3.8ppm  q80: -4.3ppm  max: -5.2ppm - threshold q90: -4.8ppm
2019-06-10 07:10:00.211 [SYN:INFO] Avg MCU drift vs SX1301#0: 1.0ppm
2019-06-10 07:10:13.863 [SYN:VERB] Time sync rejected: quality=148 threshold=137
2019-06-10 07:10:26.464 [SYN:VERB] Time sync rejected: quality=142 threshold=137
2019-06-10 07:10:36.965 [SYN:INFO] Time sync qualities: min=75 q90=138 max=148 (previous q90=137)
2019-06-10 07:10:45.366 [SYN:INFO] MCU/SX1301 drift stats: min: -2.9ppm  q50: -4.3ppm  q80: -4.8ppm  max: -14.3ppm - threshold q90: -4.8ppm
2019-06-10 07:10:45.366 [SYN:INFO] Avg MCU drift vs SX1301#0: 1.0ppm
2019-06-10 07:10:49.567 [SYN:VERB] Time sync rejected: quality=161 threshold=138
2019-06-10 07:10:57.968 [SYN:VERB] Time sync rejected: quality=139 threshold=138
2019-06-10 07:11:21.071 [SYN:VERB] Time sync rejected: quality=162 threshold=138
2019-06-10 07:11:23.171 [SYN:VERB] Time sync rejected: quality=146 threshold=138
2019-06-10 07:11:29.472 [SYN:VERB] Time sync rejected: quality=140 threshold=138
2019-06-10 07:11:35.773 [SYN:VERB] Time sync rejected: quality=140 threshold=138
2019-06-10 07:11:37.873 [SYN:INFO] Time sync qualities: min=120 q90=146 max=162 (previous q90=138)
2019-06-10 07:11:37.873 [SYN:INFO] MCU/SX1301 drift stats: min: -2.4ppm  q50: -4.3ppm  q80: -4.8ppm  max: -6.7ppm - threshold q90: -5.7ppm
2019-06-10 07:11:37.873 [SYN:INFO] Avg MCU drift vs SX1301#0: 1.0ppm
2019-06-10 07:12:13.577 [SYN:VERB] Time sync rejected: quality=148 threshold=146
2019-06-10 07:12:17.778 [SYN:VERB] Time sync rejected: quality=155 threshold=146

sgodey, 2019-06-11:
Hello All:

Where can I read about this 'TimeSync' ? I am currently stuck here. Does station send any TimeSync messaged to LNS as specified in https://doc.sm.tc/station/tcproto.html#time-synchronization. I am not seeing any WSS messages to MUXS. Any thoughts ??? Please help.

-Sreekanth

beitler, 2019-06-11:
In regular intervals Station collects some measurements for MCU/SX1301 clock synchronization. Each measurement round is assigned a timesync quality value which is defined by the time in microseconds it took to fetch the instantaneous xticks counter value from the SX1301, i.e. smaller is better. Station keeps statistics on the timesync quality and discards measurements whose quality is considered an outlier. Timesync quality outliers can occur for example if the Station process is preempted during the SPI transaction where the xticks counter is fetched. This could impact the MCU/SX1301 drift estimation and outliers are therefore discarded.

It should be noted that good MCU/SX1301 clock synchronization is not a strict requirement for Station to function properly. However, frequent time sync failures due to extreme jitter can be an indicator for certain failure modes. In your particular case you see that time sync measurements are being rejected because their quality is slightly worse than the 90th percentile of past measurements. Over time the threshold should slowly adapt and those warnings will cease. Therefore, this is not a problem at all. We may consider moving those log messages to the DEBUG level.

We are currently working on more extensive documentation on timing related aspects. In the next release you will find these sort of questions answered in the documentation.