lorabasics/basicstation #43: AU915 TX power should be 30

sa-wilson, 2019-08-13:
AU915 has the same EIRP limitations as US902 and should therefore have `s2ctx->txpow` set to 30 in [handle_router_config](https://github.com/lorabasics/basicstation/blob/c29b8502f8c715daecec6666835da6e981dc820a/src/s2e.c#L721).

It is presently being set to 14, the default.

sebascerone, 2021-08-09:
We have experienced the same issue. Despite we have configured the power at 30 dBm, we get the following log:

2021-08-02 15:37:10.914 [S2E:INFO] TX ::1 diid=45105 [ant#0] - dntxed: 925.1MHz **14.0dBm** ant#0(0) DR10 SF10/BW500 frame=207E7E822EE00B0BCF9E8
7CD…601BDB42

sebascerone, 2021-08-09:
@ogimenez-smtc this should be reported as a bug, not as an enhancement

beitler, 2021-11-30:
> We have experienced the same issue. Despite we have configured the power at 30 dBm, we get the following log:
> 
> 2021-08-02 15:37:10.914 [S2E:INFO] TX ::1 diid=45105 [ant#0] - dntxed: 925.1MHz **14.0dBm** ant#0(0) DR10 SF10/BW500 frame=207E7E822EE00B0BCF9E8 7CD…601BDB42

@sebascerone, How did you configure the power at 30 dBm? Could you show your `station.conf` file and `router_config` message?

sebascerone, 2021-11-30:
We received a custom-made FW version from multitech which fixed it. 

Nevertheless, I encourage you to review your set-up, because in our case the power output was not the only issue.

beitler, 2021-11-30:
> We received a custom-made FW version from multitech which fixed it.
> 
> Nevertheless, I encourage you to review your set-up, because in our case the power output was not the only issue.

@sebascerone thanks a lot for your feedback. Could you please more detail on the issues that you experienced?

sa-wilson, 2021-12-06:
I can't recall them, but I remember at the time I made this issue I had enough issues getting it working on our hardware with AU915 that I gave up on the porting attempts.

Especially compounded by the fact that it took over 12 months for anything to happen with the issue, and then it was just a miscategorisation. AU915 seems woefully neglected with most LoRaWAN projects - which is ridiculous considering that AS923 is actually not legal in Australia unless the FSK channel is disabled, as it violates spectral density restrictions.

beitler, 2021-12-06:
Hi, @sa-wilson, thanks for taking the time to respond. I'm sorry to hear about your experience and I'm eager to improve on the current situation. 

In essence, as you said, AU915 is an alias for US915 (currently labeled [US902](https://github.com/lorabasics/basicstation/blob/c29b8502f8c715daecec6666835da6e981dc820a/src/s2e.c#L837) for [certain reasons](https://github.com/lorabasics/basicstation/issues/18#issuecomment-479539851)) in terms of emission power, duty cycle and dwell time limitations. Hence, it is possible today to implement an AU915 channel plan using the `US902` label and setting the appropriate channels and DR table for Australia. Was this not an option in your case? 

Nevertheless, of course we are adding AU915 support in the next release. 
Also, we are enabling the use of the Channel Plan Common Names that were put in place.

In case you have any feedback on other aspects that negatively (or positively) impacted your project, please let us know on this issue tracker.

sa-wilson, 2021-12-06:
I appreciate the response, sorry if I came across a little snarky, but I haven't had the best experience dealing with most of the "official" LoRaWAN software.

I cannot recall exactly what the issues I was experiencing were - I believe some were HAL related as I was porting the software to our own gateway design (works with the old reference packet forwarder).

Unfortunately, AU915 is not actually compatible with US915 in regards to dwell time limits - AU915 doesn't have any - which is also why we have 6 Datarates, from SF12 - SF7 @ 125kHz and then SF8 @ 500 kHz. The dwell time limitations may have been why I didn't do that.

beitler, 2021-12-07:
> Unfortunately, AU915 is not actually compatible with US915 in regards to dwell time limits - AU915 doesn't have any - which is also why we have 6 Datarates, from SF12 - SF7 @ 125kHz and then SF8 @ 500 kHz. The dwell time limitations may have been why I didn't do that.

My statement was related to the gateway's perspective. Downlinks are in 500 kHz channels in both regions, so the gateway should not be subject to any dwell time limits, neither in US915 nor in AU915. But I think I understand now that the limitation your are facing comes from the fact that configuring the gateway as US915 has negative implications in other parts of the LoRaWAN Network Server.

Thanks for your feedback!

closed 2022-01-31 by beitler