lorabasics/basicstation #44: ClassA node - Join Request help

sgodey, 2019-08-14:
My Network Server (NS) is receiving a Join Request from a lora node as below:

{"msgtype":"jreq","MHdr":0,"JoinEui":"00-00-00-00-00-00-00-00",
"DevEui":"00-00-00-00-00-00-00-01","DevNonce":36030,"MIC":2133412239,
"RefTime":0.000000,"DR":3,"Freq":902500000,"upinfo":{"rctx":0,"xtime":36873222082651916,"gpstime":0,"rssi":-46,"snr":13,"rxtime":1565742560.383883}}

NS is sending a JoinAccept response to Gateway like below:

var str_obj = {
	"msgtype":"dnmsg",
	"DevEui":"00-00-00-00-00-00-01",
	"dC":0,
	"diid":0,
	"pdu":enc_pdu,
	"RxDelay":0,
	"RX1DR":8,
	"RX1Freq":923300000,
	"RX2DR":13,
	"RX2Freq":923300000,
	"priority":0,
       "xtime":msg_obj.upinfo.xtime,
	"rctx":msg_obj.upinfo.rctx
};

and NS is receiving an acknowledgement from Gateway as below:

{"msgtype":"dntxed","seqno":0,"diid":0,"DevEui":"00-00-00-00-00-00-00-01","rctx":0,"xtime":36873222083651920,"txtime":283578.392170,"gpstime":0}

But the lora node is not receiving the JoinAccept. 

I am thinking that the Gateway is NOT transmitting the JoinAccept to Lora node in the correct receive window. 

Using the data between Gateway and NS, how can NS decide on what should be the RXDelay for Gateway?

Can the experts help in letting me  know whats the mistake here?

thanks
Sreekanth

nobuggy, 2019-08-14:
The delay for Join request/accept is 5secs. Can you retry with RxDelay=5 in the dnmsg.

sgodey, 2019-08-22:
Tried this. Still no success. What is the preamble that Station sends to the node?

nobuggy, 2019-08-26:
Preamble should be 8 symbols long. Make sure the field "lorawan_public" is set to True (https://doc.sm.tc/station/gw_v1.5.html#conf-object) - which, I think, should be the case in all provided conf files.

sgodey, 2020-05-03:
if lorawan_public=true does the basic station use a preamble of 0x34? Where in the code is the preamble set? My end node is looking for a preamble (0x34) to detect the packet reception. I just want to make sure the station is transmitting 0x34 as preamble.

sgodey, 2020-06-05:
This my setup issue. Closing this.

closed 2020-06-05 by sgodey