lorabasics/basicstation #45: Ignoring leading zero bytes in `routerid` field for gateway MAC

4refr0nt, 2019-11-07:
```
   ...
   "station_conf": {
        "routerid":  "0000e8eb1141xxxx",
        /*    "euiprefix": "",   */
   ...
```
```
...
2019-11-07 07:14:29.498 [SYS:INFO] proto EUI   : 0:e8eb:1141:xxxx       (station.conf)
2019-11-07 07:14:29.498 [SYS:INFO] prefix EUI  : ::1    (builtin)
2019-11-07 07:14:29.498 [SYS:INFO] Station EUI : e8eb:11ff:fe41:xxxx
...
2019-11-07 07:14:29.734 [TCE:INFO] Infos: e8eb:11ff:fe41:xxxx e8eb:11ff:fe41:xxxx ws://127.0.0.1:3001/gateway/e8eb11fffe41xxxx
...
```
This will caused to operate incorrect gataway MAC `e8eb11fffe41xxxx`, but real MAC is `0000e8eb1141xxxx`
(real data replaced by `xxxx` for security reason)

beitler, 2019-11-07:
**tl;dr**, set `euiprefix` to 0 and you get what you want.

Thanks for posting this issue. What you are observing is one of the the ways Basic Station translates a 48-bit MAC address to at 64-bit EUI. First of all, any gateway ID where the higher (MSB) 16 bits are zero, is interpreted as a 48 bit MAC address. Now, there are multiple ways to convert this number to a 64-bit EUI which is controlled by the `euiprefix` field. If any of the lower 48 bits of `euiprefix` are non-zero, station applies an expansion by stuffing `ff:fe` in between the two halves of the MAC address to form a 64-bit number. If the lower 48 bits of `euiprefix` are zero, then the leading (MSB) 16 bits of the `euiprefix` are just pre-pended to the 48-bit MAC address, hence the name.
Some examples:
`"routerid":  "0000e8eb1141xxxx", "euiprefix": "::1" -> sation EUI: e8eb11fffe41xxxx`
`"routerid":  "0000e8eb1141xxxx", "euiprefix": "::0" -> sation EUI: 0000e8eb1141xxxx`
`"routerid":  "0000e8eb1141xxxx", "euiprefix": "4321::" -> sation EUI: 4321e8eb1141xxxx`

**Note**, that in Station config files you can write any 64 bit number in multiple ways, namely, as a decimal number, hex string or ID6 string. In these examples, the `euiprefix` is given in ID6 notation which is similar to the IPv6 address notation. You can read more about ID6 [here](https://doc.sm.tc/station/glossary.html#term-id6).

**Note**, that by leaving out the `routerid` field, Station will extract the `routerid` from the MAC address of the first network interface (eth0 in most cases). You can use the `euiprefix` as shown above to define how the extracted MAC address is converted to the Station EUI.

beitler, 2019-11-08:
Passing the `-x` argument is equivalent to setting the field `euiprefix` in the station.conf file (the setting in station.conf has higher priority).

**Examples**:
Suppose the following station.conf:
```
    "station_conf": {
        "routerid":  "0000e8eb1141xxxx",
        /* Commented lines are ignored */
        /* "euiprefix":  "4321::0", */   
```
Then:
```
./station -x 1        -> Station EUI: e8eb11fffe41xxxx
./station -x 0        -> Station EUI: 0000e8eb1141xxxx
./station -x "4321::" -> Station EUI: 4321e8eb1141xxxx
```

4refr0nt, 2019-11-08:
Thank very much!

closed 2021-07-21 by rbaldwin13