lorabasics/basicstation #49: st2pkfwd without python3.6

AkDmOl, 2019-12-05:
Hi!

I'm new to Lora things, so, likely, it's not an issue and I just can't read documentation properly, but for me it's a bit confusing.
Are there any ways to use Basic Station(on hardware) with existing network server implementations which support the pkfwd protocol without porting python3.6 to the board, as it mentioned in st2pkfwd example?
If not, could you please help me and tell the minimum version(release) of LNS, which supports Basic Station features or provide me with helpful links concerning implementation LNS&CUPS+Basic Station?

Thank you in advance!

mullerch, 2020-01-15:
ChirpStack supports LNS : https://www.chirpstack.io/
The TTN stack v3 does also : https://thethingsstack.io/

ogimenez-smtc, 2020-11-10:
Hi,

You can also use TTN V2: https://www.thethingsnetwork.org/docs/gateways/basics-station/

closed 2020-11-10 by ogimenez-smtc