lorabasics/basicstation #52: Can't get data from demo CUPS server

istina-iot, 2020-01-15:
Kerlink WirnetStation
Try example
```
2020-01-15 15:06:40.911 [AIO:INFO]
2020-01-15 15:06:40.973 [TCE:INFO] Connecting to INFOS: wss://s2.sm.tc:7000
2020-01-15 15:06:41.932 [TCE:ERRO] Infos error: router-xxxx:xxxx:xxxx:xxxx no such router
2020-01-15 15:06:41.932 [AIO:DEBU] [3] ws_close reason=1000
2020-01-15 15:06:41.933 [AIO:DEBU] [3] Server sent close: reason=1000
2020-01-15 15:06:41.933 [AIO:DEBU] [3] WS connection shutdown...
2020-01-15 15:06:41.934 [TCE:INFO] Router rejected or retry limit reached. Invoking CUPS in 30 seconds.
2020-01-15 15:06:41.935 [TCE:INFO] Terminating TC engine
```
Files from example folder
```
-rw-r--r--    1 root     root           607 Jan 15 14:55 cups-boot.crt
-rw-r--r--    1 root     root           227 Jan 15 14:55 cups-boot.key
-rw-r--r--    1 root     root           611 Jan 15 14:55 cups-boot.trust
-rw-r--r--    1 root     root            21 Jan 15 14:55 cups-boot.uri
-rw-r-----    1 root     root           421 Jan 15 15:06 cups.crt
-rw-r-----    1 root     root           121 Jan 15 15:06 cups.key
-rw-r-----    1 root     root           411 Jan 15 15:06 cups.trust
-rw-r-----    1 root     root            21 Jan 15 15:06 cups.uri
-rw-r--r--    1 root     root           320 Jan 15 14:55 makefile
-rwxr-xr-x    1 root     root        756728 Jan 15 14:56 station
-rw-r--r--    1 root     root           968 Jan 15 14:55 station.conf
-rw-r-----    1 root     root           421 Jan 15 15:06 tc.crt
-rw-r-----    1 root     root           121 Jan 15 15:06 tc.key
-rw-r-----    1 root     root           411 Jan 15 15:06 tc.trust
-rw-r-----    1 root     root            19 Jan 15 15:06 tc.uri
-rw-r--r--    1 root     root             6 Jan 15 14:55 version.txt
```
`station.conf`
```
{
  "SX1301_conf": {		     /* Actual channel plan is controlled by server */
    "lorawan_public": true,      /* is default */
    "clksrc": 1,		     /* radio_1 provides clock to concentrator */
    "device": "/dev/spidev32766.0",
	  "radio_0": {
	    "type": "SX1257",
	    "rssi_offset": -166.0,
	    "tx_enable": true,
	    "antenna_gain": 0
	   },
	  "radio_1": {
	     "type": "SX1257",
	     "rssi_offset": -166.0,
	     "tx_enable": false
    }
  },
  "station_conf": {
          "routerid": "xxxxxxxxxx",
          "radio_init": "modem_off.sh && modem_on.sh",
	  "log_file":  "stderr",
          "log_level": "XDEBUG",
	  "log_size":  10000000,
  	  "log_rotate":  3,
  	  "CUPS_RESYNC_INTV": "1s"
  }
}
```
Any ideas?

istina-iot, 2020-01-17:
If try remove routerid: basicstation reported wrong router EUI for my gateway, nothing else changed, only another value in `Infos error: router-xxxx:xxxx:xxxx:xxxx no such router`

acaracas, 2020-01-17:
We restarted the test server. Could you please try again.

istina-iot, 2020-01-17:
CUPS demo server working now, but we have not communication with LNS:
`wss://s2.sm.tc:7000` and `wss://s2.sm.tc:7001` - `WS connect failed: NET - The connection to the given server / port failed`
```
[root@Wirnet_080E0E69 semtech]# ./station
2020-01-17 12:01:29.624 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2020-01-17 12:01:29.626 [SYS:INFO] Station Ver : 2.0.3(kerlink/std) 2020-01-16 08:34:33
2020-01-17 12:01:29.626 [SYS:INFO] Package Ver : 1.0.0
2020-01-17 12:01:29.627 [SYS:INFO] proto EUI   : xxxx  (/sys/class/net/eth0/address)
2020-01-17 12:01:29.627 [SYS:INFO] prefix EUI  : ::1    (builtin)
2020-01-17 12:01:29.627 [SYS:INFO] Station EUI : xxxx
2020-01-17 12:01:29.628 [SYS:INFO] Station home: ./     (builtin)
2020-01-17 12:01:29.628 [SYS:INFO] Station temp: /var/tmp/      (builtin)
2020-01-17 12:01:29.831 [TCE:INFO] Starting TC engine
2020-01-17 12:01:29.849 [any:INFO] cert. version     : 1
serial number     : 0C
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:42:00
expires on        : 2024-11-06 12:42:00
signed using      : ECDSA with SHA256
EC key size       : 256 bits
2020-01-17 12:01:29.850 [any:INFO] cert. version     : 3
serial number     : 01:E1
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-xxxx, OU=TrackCentral, O=TrackNet.io, C=CH
issued  on        : 2020-01-15 12:28:34
expires on        : 2022-01-14 12:28:34
signed using      : ECDSA with SHA256
EC key size       : 256 bits
2020-01-17 12:01:29.850 [AIO:INFO]
2020-01-17 12:01:29.926 [TCE:INFO] Connecting to INFOS: wss://s2.sm.tc:7000
2020-01-17 12:01:29.926 [CUP:INFO] Starting a CUPS session in 0 seconds.
2020-01-17 12:01:29.926 [CUP:INFO] Starting a CUPS session now.
2020-01-17 12:01:29.927 [CUP:INFO] Connecting to CUPS ... https://s2.sm.tc:7007 (try #1)
2020-01-17 12:01:29.928 [any:INFO] cert. version     : 1
serial number     : 0C
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:42:00
expires on        : 2024-11-06 12:42:00
signed using      : ECDSA with SHA256
EC key size       : 256 bits
2020-01-17 12:01:29.929 [any:INFO] cert. version     : 3
serial number     : 01:E1
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-7276:ff00:80e:e69, OU=TrackCentral, O=TrackNet.io, C=CH
issued  on        : 2020-01-15 12:28:34
expires on        : 2022-01-14 12:28:34
signed using      : ECDSA with SHA256
EC key size       : 256 bits
2020-01-17 12:01:29.930 [AIO:INFO]
2020-01-17 12:01:30.102 [CUP:VERB] Retrieving update-info from CUPS https://s2.sm.tc:7007...
2020-01-17 12:01:31.716 [TCE:INFO] Infos: router-xxxx muxs-::0 wss://s2.sm.tc:7001/router-xxxx
2020-01-17 12:01:31.716 [AIO:DEBU] [3] ws_close reason=1000
2020-01-17 12:01:31.717 [AIO:DEBU] [3] Server sent close: reason=1000
2020-01-17 12:01:31.717 [AIO:DEBU] [3] WS connection shutdown...
2020-01-17 12:01:31.720 [any:INFO] cert. version     : 1
serial number     : 0C
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:42:00
expires on        : 2024-11-06 12:42:00
signed using      : ECDSA with SHA256
EC key size       : 256 bits
2020-01-17 12:01:31.721 [any:INFO] cert. version     : 3
serial number     : 01:E1
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-7276:ff00:80e:e69, OU=TrackCentral, O=TrackNet.io, C=CH
issued  on        : 2020-01-15 12:28:34
expires on        : 2022-01-14 12:28:34
[... 197 lines cut ...]
2020-01-17 12:06:05.587 [AIO:INFO]
2020-01-17 12:08:12.816 [AIO:ERRO] [3] WS connect failed: NET - The connection to the given server / port failed
2020-01-17 12:08:12.816 [AIO:DEBU] [3] WS connection shutdown...
2020-01-17 12:08:12.817 [TCE:ERRO] Muxs connect failed - URI: wss://s2.sm.tc:7001/router-xxxx
2020-01-17 12:08:12.817 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
2020-01-17 12:08:22.821 [any:INFO] cert. version     : 1
serial number     : 0C
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
issued  on        : 2018-11-08 12:42:00
expires on        : 2024-11-06 12:42:00
signed using      : ECDSA with SHA256
EC key size       : 256 bits
2020-01-17 12:08:22.822 [any:INFO] cert. version     : 3
serial number     : 01:D7
issuer name       : CN=Root CA, OU=TrackCentral (eiaPoUc9), O=TrackNet.io, C=CH
subject name      : CN=router-24b:8ff:fe0e:e69, OU=TrackCentral, O=TrackNet.io, C=CH
issued  on        : 2020-01-14 14:07:43
expires on        : 2022-01-13 14:07:43
signed using      : ECDSA with SHA256
EC key size       : 256 bits
2020-01-17 12:08:22.822 [AIO:INFO]
2020-01-17 12:08:22.885 [TCE:INFO] Connecting to INFOS: wss://s2.sm.tc:7000
....
```

acaracas, 2020-01-17:
Just to make sure the security credentials for the LNS are correct, could you check the steps bellow. If the connection to the LNS does not work, you could you try from scracth (make clean), i.e. remove all credentials, except the cups-boot*

First convert the local current router credentials from DER to PEM:
openssl ec -inform DER -outform PEM -in tc.key -out tc.key.pem
openssl x509 -inform DER -outform PEM -in tc.crt -out tc.crt.pem
openssl x509 -inform DER -outform PEM -in tc.trust -out tc.trust.pem

Then test the connection to the s2.sm.tc LNS:
openssl s_client -cert tc.crt.pem -key tc.key.pem -CAfile tc.trust.pem -connect s2.sm.tc:7000
openssl s_client -cert tc.crt.pem -key tc.key.pem -CAfile tc.trust.pem -connect s2.sm.tc:7001

istina-iot, 2020-01-17:
`openssl s_client -cert tc.crt.pem -key tc.key.pem -CAfile tc.trust.pem -connect s2.sm.tc:7001`
```
connect: Connection timed out
connect:errno=110
```
Verify return code: 0 (ok)
`openssl s_client -cert tc.crt.pem -key tc.key.pem -CAfile tc.trust.pem -connect s2.sm.tc:7000`
```
CONNECTED(00000003)
depth=1 CN = Root CA, OU = TrackCentral (eiaPoUc9), O = TrackNet.io, C = CH
verify return:1
depth=0 CN = infos-::0, OU = TrackCentral, O = TrackNet.io, C = CH
verify return:1
---
Certificate chain
 0 s:/CN=infos-::0/OU=TrackCentral/O=TrackNet.io/C=CH
   i:/CN=Root CA/OU=TrackCentral (eiaPoUc9)/O=TrackNet.io/C=CH
 1 s:/CN=Root CA/OU=TrackCentral (eiaPoUc9)/O=TrackNet.io/C=CH
   i:/CN=Root CA/OU=TrackCentral (eiaPoUc9)/O=TrackNet.io/C=CH
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIBqzCCAVGgAwIBAgIBEDAKBggqhkjOPQQDAjBXMRAwDgYDVQQDDAdSb290IENB
MSAwHgYDVQQLDBdUcmFja0NlbnRyYWwgKGVpYVBvVWM5KTEUMBIGA1UECgwLVHJh
Y2tOZXQuaW8xCzAJBgNVBAYTAkNIMB4XDTE5MDIyNTE1MTc1N1oXDTIxMDIyNDE1
MTc1N1owTjESMBAGA1UEAwwJaW5mb3MtOjowMRUwEwYDVQQLDAxUcmFja0NlbnRy
YWwxFDASBgNVBAoMC1RyYWNrTmV0LmlvMQswCQYDVQQGEwJDSDBZMBMGByqGSM49
AgEGCCqGSM49AwEHA0IABNCWXIFsClabM4ruBnYR1QyKyB8MNMKdebRzIysLhezm
tJTfr3D2FfRO9cSNSF7BZPsoOtbDXtwM8W6UI6CG2iSjFzAVMBMGA1UdEQQMMAqC
CHMyLnNtLnRjMAoGCCqGSM49BAMCA0gAMEUCIAnflxxmcrXXUcvDAaMJ3g3RcuMM
WL7KvtvQeOPcblRZAiEAq4r8pOaIUqhGYoxLRc25E5L/setO0uDhHc4sv7FLFcY=
-----END CERTIFICATE-----
subject=/CN=infos-::0/OU=TrackCentral/O=TrackNet.io/C=CH
issuer=/CN=Root CA/OU=TrackCentral (eiaPoUc9)/O=TrackNet.io/C=CH
---
No client certificate CA names sent
Peer signing digest: SHA256
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 1314 bytes and written 434 bytes
---
New, TLSv1/SSLv3, Cipher is ECDHE-ECDSA-AES256-GCM-SHA384
Server public key is 256 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-ECDSA-AES256-GCM-SHA384
    Session-ID: DAD5B5D8F33BB91F4DF849EC531A20D4705BD2F294D4EA48A9614F9DC76373AD
    Session-ID-ctx:
    Master-Key: AE336B0255527D8501D6C1E322F5E502F067A964A9A9AE8B482457A84C7B6CAF5768766C72EB4E5E2588402B13ACB7D1
    Key-Arg   : None
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 7200 (seconds)
    TLS session ticket:
    0000 - d6 55 6b e5 89 06 fb 8e-0a 64 22 b6 09 fa 6a 91   .Uk......d"...j.
    0010 - cb 4c 76 1e 57 b6 07 53-87 89 e2 c5 cd 73 7f b0   .Lv.W..S.....s..
    0020 - 7c c4 5b 0d 2d 88 ea 17-24 dc 30 69 b5 88 30 ac   |.[.-...$.0i..0.
    0030 - 83 28 76 b7 9a 5a cc bd-05 06 ea 77 ce 9e b3 20   .(v..Z.....w...
    0040 - 42 ad 16 2b 91 24 91 23-10 16 59 a1 6f dd e6 3b   B..+.$.#..Y.o..;
    0050 - eb 05 80 1c bb d8 7b 37-25 0a a7 87 34 4b 8f 29   ......{7%...4K.)
    0060 - 1c 9f 18 48 3c 37 89 8a-d9 79 a3 2a 36 00 09 76   ...H<7...y.*6..v
    0070 - cf 8d c8 46 60 45 48 7c-2d 45 3c ff 2a 78 3b bd   ...F`EH|-E<.*x;.
    0080 - 15 27 3c 44 5c 07 44 e9-32 4d 5d 5c 06 06 cf 27   .'<D\.D.2M]\...'
    0090 - 93 d6 a5 4d d9 32 1a 2f-9f ee 77 a5 a7 c5 81 5c   ...M.2./..w....\

    Start Time: 1579271205
    Timeout   : 300 (sec)
    Verify return code: 0 (ok)
```

istina-iot, 2020-01-17:
Then I try remove all credentials, except the cups-boot*
```
2020-01-17 14:36:46.168 [CUP:VERB] Retrieving update-info from CUPS-boot https://s2.sm.tc:7007...
2020-01-17 14:36:47.164 [any:VERB] Failed to retrieve TCURI from CUPS: (400) Bad Request
```

acaracas, 2020-01-17:
It is strange that you get a connection timeout for s2.sm.tc:7001 but it works for port 7000.
Could you check that your network/firewall settings do not block this connection?
telnet s2.sm.tc 7000
telnet s2.sm.tc 7001

When you removed all credentials, you might have removed the version.txt file as well.
echo "1.00.00" > version.txt

istina-iot, 2020-01-17:
All ok, but ...
```
2020-01-17 14:47:06.334 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=868300000 rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-01-17 14:47:06.334 [RAL:VERB] SX1301 rxrfchain 1: enable=0 freq=0 rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
....
2020-01-17 14:47:09.251 [S2E:INFO] Configuring for region: EU863 -- 863.0MHz..870.0MHz
```
rxrfchain - unconfigured, region - RU_864_870 not supported :(

acaracas, 2020-01-17:
By default all routers are accepted and auto-provisioned on this demo server using the CUPS protocol. And by default all routers are handed credentials for the connection to the LNS. 

However, all routers are configured to be EU863 in terms of the regional channel plan. This server is just intended to give the overview of the connection setup and high-level interaction of the basic station protocols (CUPS and LNS).

The server implementation supports the default regional channel plan for RU864 with two channels. If you send me the routerid, I could configure this router for you as a test, however this configuration is without guarantees, i.e., the configuration for your router will not be persistent across server updates.

istina-iot, 2020-01-20:
Thank you very much!

closed 2020-01-20 by istina-iot