lorabasics/basicstation #53: basicstation process interrupted on Kerlink Wirnet Station

istina-iot, 2020-01-17:
After short time, regularly, basicstation process interupted
Log:
```
2020-01-17 06:39:27.729 [TCE:INFO] MUXS reconnect backoff 1s (retry 0)
2020-01-17 06:39:28.745 [TCE:VERB] Connecting to MUXS...
2020-01-17 06:39:28.748 [TCE:VERB] Connected to MUXS.
2020-01-17 06:39:28.805 [any:CRIT] Radio device '/dev/spidev32766.0' in use by process: 1609
```
but `ps | grep 1609` show nothing.

istina-iot, 2020-01-17:
This happens every time after the loss and restore connection with LNS:
```
2020-01-17 07:18:52.797 [AIO:DEBU] [4] Connection closed unexpectedly
2020-01-17 07:18:52.797 [AIO:DEBU] [4] WS connection shutdown...
2020-01-17 07:18:52.798 [TCE:VERB] Connection to MUXS closed in state 4
2020-01-17 07:18:52.799 [TCE:INFO] MUXS reconnect backoff 1s (retry 0)
2020-01-17 07:18:53.816 [TCE:VERB] Connecting to MUXS...
2020-01-17 07:18:53.818 [TCE:VERB] Connected to MUXS.
2020-01-17 07:18:53.874 [any:CRIT] Radio device '/dev/spidev32766.0' in use by process: 2539
```

jawadiot, 2020-05-13:
I have the same error, any oneknow what cause this error ?

beitler, 2020-05-14:
Thanks for reporting the issue. This is fixed in the upcoming release.

closed 2022-01-31 by beitler