lorabasics/basicstation #56: kerlink compile error

RadW2020, 2020-04-04:
First I have this error. Some files cannot be found because toolchain folder structure is not as expected:
```
> make platform=kerlink variant=std

platform=kerlink variant=std make -C deps/mbedtls
make[1]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls'
make -C platform-kerlink lib CC=/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-gcc AR=/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-ar LD=/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-ld
make[2]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink'
make[3]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink/library'
  CC    aes.c
make[3]: /home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-gcc: Command not found
Makefile:161: recipe for target 'aes.o' failed
make[3]: *** [aes.o] Error 127
make[3]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink/library'
Makefile:18: recipe for target 'lib' failed
make[2]: *** [lib] Error 2
make[2]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink'
makefile:18: recipe for target 'platform-kerlink/library/libmbedtls.a' failed
make[1]: *** [platform-kerlink/library/libmbedtls.a] Error 2
make[1]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls'
makefile:41: recipe for target 'deps/mbedtls' failed
make: *** [deps/mbedtls] Error 2
```

Then, after restructuring bin folder inside toolchain-kerlink I have another error:
```
> make platform=kerlink variant=std

platform=kerlink variant=std make -C deps/mbedtls
make[1]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls'
make -C platform-kerlink lib CC=/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-gcc AR=/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-ar LD=/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-ld
make[2]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink'
make[3]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink/library'
  CC    aes.c
arm-klk-linux-gnueabi-gcc.br_real: error trying to exec 'cc1': execvp: No such file or directory
Makefile:161: recipe for target 'aes.o' failed
make[3]: *** [aes.o] Error 1
make[3]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink/library'
Makefile:18: recipe for target 'lib' failed
make[2]: *** [lib] Error 2
make[2]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink'
makefile:18: recipe for target 'platform-kerlink/library/libmbedtls.a' failed
make[1]: *** [platform-kerlink/library/libmbedtls.a] Error 2
make[1]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls'
makefile:41: recipe for target 'deps/mbedtls' failed
```

beitler, 2020-04-09:
The toolchain from kerlink is not quite in the structure that is expected by the build environment. However, it's quite simple to bring it into the correct structure. Try this:
```
tar xvf toolchain_x86_64_v1.10.tar.xz --strip-components=3 -C ~/toolchain-kerlink
cd ~/toolchain-kerlink
ln -s usr/bin
```

ogimenez-smtc, 2020-11-10:
Hi @RadW2020 , is your issue fixed by @beitler answer ?

RadW2020, 2021-03-24:
Sorry I didn't answer back. 
I tried but it was not working.
Now I see in your documentation this new reference. 

[image]

Do you know where can I get this toolchain file?

ogimenez-smtc, 2021-03-25:
Dear,

You can download it on Kerlink's Wiki https://wikikerlink.fr/wirnet-ibts/doku.php?id=wiki:resources#toolchain.

Please ask your Kerlink representative if you need credentials to it

RadW2020, 2021-03-25:
I have read the whole thread again and I might have made a mistake because I was not precise enough.
My problem here is with the **Wirnet iFemtoCell** of Kerlink 
https://www.kerlink.com/product/wirnet-ifemtocell/
I already tried that iBTS toolchain and is not working for me. 

I know at this point this is a problem with Kerlink and I'm trying to get in contact to have some support on this from them.

Also, it is not in your supported platforms list https://lora-developers.semtech.com/resources/tools/lora-basics/lora-basics-for-gateways/?url=platforms.html, so it might not work. but since it is a Linux based I'm trying to make it happen.

RadW2020, 2021-03-26:
I finally got the correct toolchain from Kerlink but this error shows up and I don't know what else to do. 

```
➜  basicstation git:(master) ✗ make platform=kerlink variant=std
platform=kerlink variant=std make -C deps/mbedtls
make[1]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls'
make -C platform-kerlink lib CC="/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-gcc" AR="/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-ar" LD="/home/rji/toolchain-kerlink/bin/arm-klk-linux-gnueabi-ld"
make[2]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink'
make[3]: Entering directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink/library'
  CC    aes.c
arm-klk-linux-gnueabi-gcc.br_real: error trying to exec 'cc1': execvp: No such file or directory
Makefile:161: recipe for target 'aes.o' failed
make[3]: *** [aes.o] Error 1
make[3]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink/library'
Makefile:18: recipe for target 'lib' failed
make[2]: *** [lib] Error 2
make[2]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls/platform-kerlink'
makefile:43: recipe for target 'platform-kerlink/library/libmbedtls.a' failed
make[1]: *** [platform-kerlink/library/libmbedtls.a] Error 2
make[1]: Leaving directory '/home/rji/Documents/basicstation/deps/mbedtls'
makefile:67: recipe for target 'deps/mbedtls' failed
make: *** [deps/mbedtls] Error 2```

Any ideas?