lorabasics/basicstation #57: Basic station should use function mbedtls_ssl_set_hostname

robertlie, 2020-04-09:
TTN LNS uses the TLS SNI (Server Name Indication) extension.
- When executing this command:
  openssl s_client -showcerts -connect 52.169.76.203:443 -servername lns.eu.thethings.network
  LNS will return the lns.eu.thethings.network certificate
- When executing this command:
  openssl s_client -showcerts -connect 52.169.76.203:443 -servername lns.us.thethings.network
  LNS will return the lns.us.thethings.network certificate

The mbedtls_ssl_handshake function is executed, see:
https://github.com/lorabasics/basicstation/blob/master/src/net.c#L638  
When tc.uri contains wss://lns.eu.thethings.network:443
the TTN LNS returns the lns.IN.thethings.network certificate and NOT the lns.EU.thethings.network certificate, and no error is generated.

So basic station code does not specify the server name.
I expect the use of: mbedtls_ssl_set_hostname
Something like this:

```
if( conn->tlsctx ) {
	err = mbedtls_ssl_set_hostname(conn->tlsctx, conn->host);

	if(err==0) {
	    err = mbedtls_ssl_handshake(conn->tlsctx);
	}
}
```

With the use of mbedtls_ssl_set_hostname, in the logging you will see:
```
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:887 MBEDTLS[3]: client hello, add ciphersuite: 008b
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:920 MBEDTLS[3]: client hello, got 131 ciphersuites
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:951 MBEDTLS[3]: client hello, compress len.: 1
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:953 MBEDTLS[3]: client hello, compress alg.: 0
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:72 MBEDTLS[3]: client hello, adding server name extension: lns.eu.thethings.network
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:178 MBEDTLS[3]: client hello, adding signature_algorithms extension
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:263 MBEDTLS[3]: client hello, adding supported_elliptic_curves extension
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:328 MBEDTLS[3]: client hello, adding supported_point_formats extension
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:510 MBEDTLS[3]: client hello, adding encrypt_then_mac extension
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:544 MBEDTLS[3]: client hello, adding extended_master_secret extension
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:577 MBEDTLS[3]: client hello, adding session ticket extension
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_cli.c:1025 MBEDTLS[3]: client hello, total extension length: 105
2020-04-09 15:23:31.471 [AIO:DEBU] ssl_tls.c:2701 MBEDTLS[2]: => write record
```

Without the use of mbedtls_ssl_set_hostname, in the logging you will see:
```
2020-04-09 10:16:07.180 [AIO:DEBU] ssl_cli.c:887 MBEDTLS[3]: client hello, add ciphersuite: 008b
2020-04-09 10:16:07.180 [AIO:DEBU] ssl_cli.c:920 MBEDTLS[3]: client hello, got 131 ciphersuites
2020-04-09 10:16:07.180 [AIO:DEBU] ssl_cli.c:951 MBEDTLS[3]: client hello, compress len.: 1
2020-04-09 10:16:07.180 [AIO:DEBU] ssl_cli.c:953 MBEDTLS[3]: client hello, compress alg.: 0
2020-04-09 10:16:07.181 [AIO:DEBU] ssl_cli.c:178 MBEDTLS[3]: client hello, adding signature_algorithms extension
2020-04-09 10:16:07.181 [AIO:DEBU] ssl_cli.c:263 MBEDTLS[3]: client hello, adding supported_elliptic_curves extension
2020-04-09 10:16:07.181 [AIO:DEBU] ssl_cli.c:328 MBEDTLS[3]: client hello, adding supported_point_formats extension
2020-04-09 10:16:07.181 [AIO:DEBU] ssl_cli.c:510 MBEDTLS[3]: client hello, adding encrypt_then_mac extension
2020-04-09 10:16:07.181 [AIO:DEBU] ssl_cli.c:544 MBEDTLS[3]: client hello, adding extended_master_secret extension
2020-04-09 10:16:07.181 [AIO:DEBU] ssl_cli.c:577 MBEDTLS[3]: client hello, adding session ticket extension
2020-04-09 10:16:07.181 [AIO:DEBU] ssl_cli.c:1025 MBEDTLS[3]: client hello, total extension length: 72
2020-04-09 10:16:07.181 [AIO:DEBU] ssl_tls.c:2701 MBEDTLS[2]: => write record

```

beitler, 2020-04-09:
Thanks for reporting this. The next release will contain an option to enable TLS SNI.

mullerch, 2020-06-16:
Can you please clarify with which kind of certificates the server authentication feature of the Basic Station is supposed to work? Because I have the feeling that adding an "option" will rise complexity without filling a user requirement.

- If it's intended to be used with the server's certificate as root CA, server name check is not desired (probably preferred in on-premise / company infrastructure).
- If it's intended to work with common root authorities CAs (e.g. Let's Encrypt), then server name validation is mandatory (probably preferred in Cloud / service provider infrastructure). Also I would then expect the BS to look at the OS standard path (with ev. station.conf fine tuning) for the existing root CAs (this implies being able to use TLS without a trust file at all) because no one want's to setup and manage such annoying stuff.
- Being able to do both with implicit configuration would be just amazing for the users (if trust is given, don't check the host root CA certificates and disable server name check)

mullerch, 2021-01-25:
It looks like SNI check is now implemented: https://github.com/lorabasics/basicstation/blob/bd17e53ab1137de6abb5ae48d6f3d52f6c268299/src/tls.c#L277

beitler, 2022-01-31:
This is available since version 2.0.5

closed 2022-01-31 by beitler