lorabasics/basicstation #58: Understand how the parameters nocca, nodc, and nodwell works.

robertlie, 2020-04-11:
Looking at the code https://github.com/lorabasics/basicstation/blob/v2.0.3/src/s2e.c#L805
the "region" parameter in the router_config msgtype, determines if listen before talk (LBT), duty cycle (DC) or dwell time (DWT) should be applied for that specific region.

In the documentation (https://lora-developers.semtech.com/resources/tools/basic-station/configuration-files/)  the following is mentioned:

> The fields nocca, nodc, and nodwell, if present and false, disable clear channel analysis (e.g., listen before talk), duty cycle, and dwell time limitations. 

As an example lets take region=EU and based on file s2e.c (line 805) it means:
```
LBT = not applied
DC = applied
DWT = not applied
```

Now lets set the following:
```
nocca = true 
nodc = true
nodwell = true
```

Based on the settings above is my following assumption correct? 
```
LBT = not applied
DC = applied
DWT = not applied
```

Now lets change the settings:
```
nocca = false 
nodc = false
nodwell = false
```

Is my following assumption correct? 
```
LBT = not applied
DC = not applied
DWT = not applied
```

So the fields nocca, nodc, and nodwell can NOT enable a LBT, DC or DWT functionality if it does not apply to that region.

beitler, 2020-04-13:
Thanks for reporting this. This is a typo in the documentation. It should read:
> The fields nocca, nodc, and nodwell, if present and **true**, disable clear channel analysis (e.g., listen before talk), duty cycle, and dwell time limitations.

beitler, 2022-01-31:
This has been fixed [in the docs](https://doc.sm.tc/station/conf.html#development-debug-settings).

closed 2022-01-31 by beitler