lorabasics/basicstation #59: BasicStation design: rxpollTmr not firing

sgodey, 2020-04-18:
I compiled the latest Basicstation code and did an update of all my packages and I figureout that the 'rxpollTmr' in ral_lgw2.c, which is set to RX_POLL_INTV (20 ms) is not firing. Any ideas? 

This was working perfectly fine before the update.

sgodey, 2020-04-20:
Figured out the the function sys_time () is always returning a value zero. Looks like some issue with the apt update that I did on my raspbian (stretch). 

Anyone had similar issue?

beitler, 2020-04-21:
Unfortunately, I cannot reproduce your findings. Could you provide more detail about your environment?
- Are you x-compiling or building on target?
- What is your diff to the clean repo? (Any changes to build or source files?)

Please try this setup:
From a clean repository (directly on the rpi) change to the `examples/simulation` directory and run `make platform=rpi` (make sure `which python` points to python version 3.5+ and requirements are installed - `pip3 install -r examples/simulation/requirements.txt`. In my environment (`gcc (Raspbian 8.3.0-6+rpi1) 8.3.0`) this works without issue.

sgodey, 2020-04-21:
I am compiling on Target (raspberry pi). The only DIFF I have is in cups.c where I had to add "Content-Type: application/json" to the POST request. Will try the "simulation" example.

closed 2020-06-05 by sgodey