lorabasics/basicstation #62: Basic station fail '/dev/spidev0.0' in use by process

jawadiot, 2020-05-14:
Hi,

I'm running basic station on **raspberry compute 3**, with concentrator **RAKWIRLESS 831**, i sucess to run the example **live-s2.sm.tc** and i can watch the led driver of the concentrator on, But the basic station stop running after  connecting/deconnecting from mux, the log say that there is already process using spi, and i'm pretty sur that he talk about him( the process that fail), and i even can't find this process running.

**My station.conf :** 
`{
    /* If slave-X.conf present this acts as default settings */
    "SX1301_conf": {                    /* Actual channel plan is controlled by server */
        "lorawan_public": false,         /* is default */
        "clksrc": 1,                    /* radio_1 provides clock to concentrator */
        /* path to the SPI device, un-comment if not specified on the command line e.g., RADIODEV=/dev/spidev0.0 */
        "device": "/dev/spidev0.0",
        /* freq/enable provided by LNS - only HW specific settings listed here */
        "radio_0": {
                "type": "SX1257",
                "rssi_offset": -166.0,
                "tx_enable": true,
                "antenna_gain": 0
        },
        "radio_1": {
                "type": "SX1257",
                "rssi_offset": -166.0,
                "tx_enable": false
        }
        /* chan_multiSF_X, chan_Lora_std, chan_FSK provided by LNS */
    },
    "station_conf": {
        "log_file":  "stderr",
        "log_level": "DEBUG",           /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
        "log_size":  10000000,
        "log_rotate":  3,
        "CUPS_RESYNC_INTV": "1s"
    }
}`

**log basic station :**

`pi@B827EBFFFEF65A01:/opt/basicstation/examples/live-s2.sm.tc $ station
2020-05-14 01:13:15.576 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2020-05-14 01:13:15.576 [SYS:INFO] Station Ver : 2.0.3(rpi/std) 2020-05-13 11:08:10
2020-05-14 01:13:15.577 [SYS:INFO] Package Ver : 1.0.0
2020-05-14 01:13:15.577 [SYS:INFO] proto EUI   : 0:b827:ebf6:5a01	(/sys/class/net/eth0/address)
2020-05-14 01:13:15.577 [SYS:INFO] prefix EUI  : ::1	(builtin)
2020-05-14 01:13:15.578 [SYS:INFO] Station EUI : b827:ebff:fef6:5a01
2020-05-14 01:13:15.578 [SYS:INFO] Station home: ./	(builtin)
2020-05-14 01:13:15.578 [SYS:INFO] Station temp: /var/tmp/	(builtin)
2020-05-14 01:13:15.578 [SYS:WARN] Station in NO-CUPS mode
2020-05-14 01:13:15.780 [TCE:INFO] Starting TC engine
2020-05-14 01:13:15.780 [TCE:INFO] Connecting to INFOS: ws://127.0.0.1:3001
2020-05-14 01:13:15.789 [TCE:INFO] Infos: b827:ebff:fef6:5a01 b827:ebff:fef6:5a01 ws://127.0.0.1:3001/gateway/b827ebfffef65a01
2020-05-14 01:13:15.789 [AIO:DEBU] [3] ws_close reason=1000
2020-05-14 01:13:15.789 [AIO:DEBU] Echoing close - reason=1000
2020-05-14 01:13:15.790 [AIO:ERRO] Recv failed: NET - Connection was reset by peer
2020-05-14 01:13:15.795 [AIO:DEBU] [3] WS connection shutdown...
2020-05-14 01:13:15.796 [TCE:VERB] Connecting to MUXS...
2020-05-14 01:13:15.796 [TCE:VERB] Connected to MUXS.
2020-05-14 01:13:15.807 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2020-05-14 01:13:15.807 [RAL:DEBU] SX1301 txlut table (0 entries)
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=867500000 rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=868500000 rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=0 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=0 freq=400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=1 freq=-200000 bandwidth=2 datarate=2 sync_word=0/0
2020-05-14 01:13:15.807 [RAL:VERB] SX1301 ifchain  9: enable=0 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
[... 151 lines cut ...]
May 14 02:13:26 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:26+01:00" level=info msg="backend/basicstation: gateway disconnected" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43638"
May 14 02:13:26 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:26+01:00" level=info msg="integration/mqtt: unsubscribe topic" topic="gateway/b827ebfffef65a01/command/#"
May 14 02:13:36 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:36+01:00" level=info msg="backend/basicstation: router-info request received" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43640" router_uri="ws://127.0.0.1:3001/gateway/b827ebfffef65a01"
May 14 02:13:36 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:36+01:00" level=info msg="backend/basicstation: gateway connected" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43642"
May 14 02:13:36 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:36+01:00" level=info msg="integration/mqtt: subscribing to topic" qos=0 topic="gateway/b827ebfffef65a01/command/#"
May 14 02:13:36 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:36+01:00" level=info msg="backend/basicstation: gateway version received" firmware=1.0.0 gateway_id=b827ebfffef65a01 model=rpi package=1.0.0 protocol=2 station="2.0.3(rpi/std)"
May 14 02:13:36 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:36+01:00" level=info msg="backend/basicstation: router-config message sent to gateway" gateway_id=b827ebfffef65a01
May 14 02:13:37 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:37+01:00" level=info msg="backend/basicstation: gateway disconnected" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43642"
May 14 02:13:37 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:37+01:00" level=info msg="integration/mqtt: unsubscribe topic" topic="gateway/b827ebfffef65a01/command/#"
May 14 02:13:47 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:47+01:00" level=info msg="backend/basicstation: router-info request received" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43644" router_uri="ws://127.0.0.1:3001/gateway/b827ebfffef65a01"
May 14 02:13:47 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:47+01:00" level=info msg="backend/basicstation: gateway connected" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43646"
May 14 02:13:47 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:47+01:00" level=info msg="integration/mqtt: subscribing to topic" qos=0 topic="gateway/b827ebfffef65a01/command/#"
May 14 02:13:47 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:47+01:00" level=info msg="backend/basicstation: gateway version received" firmware=1.0.0 gateway_id=b827ebfffef65a01 model=rpi package=1.0.0 protocol=2 station="2.0.3(rpi/std)"
May 14 02:13:47 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:13:47+01:00" level=info msg="backend/basicstation: router-config message sent to gateway" gateway_id=b827ebfffef65a01
May 14 02:14:52 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:14:52+01:00" level=info msg="backend/basicstation: gateway disconnected" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43646"
May 14 02:14:52 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:14:52+01:00" level=info msg="integration/mqtt: unsubscribe topic" topic="gateway/b827ebfffef65a01/command/#"
May 14 02:14:56 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:14:56+01:00" level=info msg="backend/basicstation: gateway connected" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43648"
[... cut ...]
May 14 02:14:56 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:14:56+01:00" level=info msg="backend/basicstation: router-config message sent to gateway" gateway_id=b827ebfffef65a01
May 14 02:14:56 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:14:56+01:00" level=info msg="backend/basicstation: gateway disconnected" gateway_id=b827ebfffef65a01 remote_addr="127.0.0.1:43648"
May 14 02:14:56 B827EBFFFEF65A01 chirpstack-gateway-bridge[559]: time="2020-05-14T02:14:56+01:00" level=info msg="integration/mqtt: unsubscribe topic" topic="gateway/b827ebfffef65a01/command/#"
`

I'mreally stuck about this, can you help me please ?

beitler, 2020-05-14:
This is related to #53. A fix will be available in the next release.
Until then, as a work around you could start station in daemon mode using flag `-d` (see `station --help` for more details). This will cause station to start in the background, and additionally it will fork into a child-process such that the parent remains the daemon control process which restarts the the child when it exits for some reason. In the log you will see something like:
```
2020-05-14 10:55:23.175 [any:CRIT] Radio device '/dev/spidev32766.0' in use by process: 17882
2020-05-14 10:55:23.596 [SYS:ERRO] DAEMON: Station process 17882 died (exit code 0x1E00)
2020-05-14 10:55:23.606 [SYS:INFO] DAEMON: Station process 17943 started...
```
In order to be able to inspect the log in daemon mode you should specify a log file inside `station.conf`, like this:
```
...
    "station_conf": {
        "log_file":  "/var/log/station.log",
        "log_level": "DEBUG",  /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
        "log_size":  10000000,
        "log_rotate":  3,
         ...
    }
...
```

jawadiot, 2020-05-14:
Your solution works well !!!
 Despite the fact that the service connects disconnects I still managed to make a communication between a sensor and the Gateway bridge I do not know if it is really bad that it starts and restarts but in any case it works
But if i want to lunch station on boot with systemctl, the daemon mode will  cause error maybe ?  

**station.service :** 

> [Unit]
Description=Basic station secure websocket
Wants=network-online.target
After=network-online.target
[Service]
User=root
Group=root
ExecStart= /opt/basicstation/build-rpi-std/bin/station -d -h /opt/basicstation/examples/live-s2.sm.tc
// RADIODEV=/dev/spidev0.0  ExecStart= /opt/basicstation/build-rpi-std/bin/station -d -h /opt/basicstation/examples/live-s2.sm.tc / not work 
[Install]
WantedBy=multi-user.target
Alias=basic_station.service

I have some trouble to put the argument  on "execstart",but i think it just form error.
In any case thank you for your help

closed 2022-01-31 by beitler

GitUser8675309, 2024-05-01:
I have the same issue here in 2024. Is there a fix for this?

beitler, 2024-05-01:
> I have the same issue here in 2024. Is there a fix for this?

Can you please provide logs?

GitUser8675309, 2024-05-03:
Hi Beitler,
Here's the output.

 $ docker compose up
WARN[0000] /home/gerry/docker-compose.yml: `version` is obsolete
[+] Running 20/20
 ✔ basicstation Pulled                                                                             38.7s
   ✔ 59f5764b1f6d Pull complete                                                                    18.6s
   ✔ d25f96231021 Pull complete                                                                    18.9s
   ✔ bb18e33e1caf Pull complete                                                                    20.8s
   ✔ 0d20bfb20cea Pull complete                                                                    20.9s
   ✔ e19e63a02380 Pull complete                                                                    21.0s
   ✔ 86ebbdeb7b51 Pull complete                                                                    21.1s
   ✔ 4f4fb700ef54 Pull complete                                                                    21.1s
   ✔ 479634a35477 Pull complete                                                                    21.2s
   ✔ bcab80059af9 Pull complete                                                                    21.2s
   ✔ 5442132ede36 Pull complete                                                                    21.3s
   ✔ c3ce9eb5ac75 Pull complete                                                                    21.4s
   ✔ 9b5701f42b0b Pull complete                                                                    24.2s
   ✔ 91b4b4d6dc2f Pull complete                                                                    24.3s
   ✔ 9ccdbfcb8f3f Pull complete                                                                    24.3s
   ✔ fd2bc0fab3b3 Pull complete                                                                    30.8s
   ✔ 0b1c45d39382 Pull complete                                                                    31.0s
   ✔ e7fc4261f017 Pull complete                                                                    31.1s
   ✔ e925e200ba5a Pull complete                                                                    36.9s
   ✔ 7cc391a815a2 Pull complete                                                                    37.0s
[+] Running 1/1
 ✔ Container basicstation  Created                                                                  8.6s
Attaching to basicstation
basicstation  | ERROR: DEVICE set to auto discover but no concentrator found! (INTERFACE set to SPI)
basicstation  | ------------------------------------------------------------------
basicstation  | BasicStation (TM) Packet Forwarder Container v2.8.3
basicstation  | (c) Xose Pérez 2022-2024
basicstation  |
basicstation  | Based on:
basicstation  |  * basicstation v2.0.6
basicstation  |  * sx1302_hal v2.1.0
basicstation  |  * picoGW_hal v0.2.3
basicstation  | ------------------------------------------------------------------
basicstation  |
basicstation  | Protocol
basicstation  | ------------------------------------------------------------------
basicstation  | Mode:          DYNAMIC
basicstation  | Protocol:      LNS
basicstation  | LNS Server:    wss://eu1.cloud.thethings.network:8887
basicstation  | Gateway EUI:   D83ADDFFFE78ADB2
basicstation  | EUI Source:    eth0
basicstation  |
basicstation  | Radio
basicstation  | ------------------------------------------------------------------
basicstation  | Model:         SX1303
basicstation  | Concentrator:  SX1303
basicstation  | Design:        CORECELL
basicstation  | Interface:     SPI
basicstation  | Radio Device:  /dev/spidev0.1
basicstation  | SPI Speed:     8000000
basicstation  | Reset GPIO:    17
basicstation  | Enable GPIO:   0
basicstation  |
basicstation  | 2024-04-30 21:42:28.580 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
basicstation  | 2024-04-30 21:42:28.581 [SYS:INFO] Station Ver : 2.0.6(corecell/stdn) 2024-04-16 20:15:36
basicstation  | 2024-04-30 21:42:28.581 [SYS:INFO] Package Ver : (null)
basicstation  | 2024-04-30 21:42:28.581 [SYS:INFO] mbedTLS Ver : 2.28.0
basicstation  | 2024-04-30 21:42:28.581 [SYS:INFO] proto EUI   : d83a:ddff:fe78:adb2    (station.conf)
basicstation  | 2024-04-30 21:42:28.581 [SYS:INFO] prefix EUI  : ::1    (builtin)
basicstation  | 2024-04-30 21:42:28.581 [SYS:INFO] Station EUI : d83a:ddff:fe78:adb2
basicstation  | 2024-04-30 21:42:28.581 [SYS:INFO] Station home: ./     (builtin)
basicstation  | 2024-04-30 21:42:28.581 [SYS:INFO] Station temp: /var/tmp/      (builtin)
basicstation  | 2024-04-30 21:42:28.581 [SYS:WARN] Station in NO-CUPS mode
basicstation  | 2024-04-30 21:42:28.582 [RAL:DEBU] execvp argv[0]: </proc/self/exe>
[... 95 lines cut ...]
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR5  FSK
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR6  FSK
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR7  FSK
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR8  SF12/BW500
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR9  SF11/BW500
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR10 SF10/BW500
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR11 SF9/BW500
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR12 SF8/BW500
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR13 SF7/BW500
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR14 FSK
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   DR15 FSK
basicstation  | 2024-04-30 21:42:31.319 [S2E:INFO]   TX power: 26.0 dBm EIRP
[... cut ...]
basicstation  | ERROR: Failed to set SX1250_0 in STANDBY_RC mode
basicstation  | 2024-04-30 21:42:33.064 [S00:ERRO] [lgw_start:908] failed to setup radio 0
basicstation  | 2024-04-30 21:42:33.064 [S00:ERRO] Concentrator start failed: lgw_start
basicstation  | 2024-04-30 21:42:33.064 [S00:CRIT] Slave radio start up failed with status 0x08
basicstation  | 2024-04-30 21:42:32.952 [RAL:ERRO] Slave (0) - EOF
basicstation  | 2024-04-30 21:42:32.953 [any:CRIT] Slave pid=66 idx=0: Fatal exit
basicstation exited with code 0
gerry@rak-gateway:~ $

VitorBFreitas, 2024-08-21:
@GitUser8675309 Did you managed to fix it?

beitler, 2024-08-22:
The issue described by @GitUser8675309  is different than the original issue. 

In this situation, station is compiled in master/slave configuration (`stdn` variant).
```
basicstation | 2024-04-30 21:42:28.581 [SYS:INFO] Station Ver : 2.0.6(corecell/stdn) 2024-04-16 20:15:36
```

This is intended for 'multi-board' setups, however, here it is used with a single radio board `S00`. The radio is failing to start up causing the child process to die which in turn is interpreted as a fatal error on the parent process. 

```
basicstation | ERROR: Failed to set SX1250_0 in STANDBY_RC mode
basicstation | 2024-04-30 21:42:33.064 [S00:ERRO] [lgw_start:908] failed to setup radio 0
basicstation | 2024-04-30 21:42:33.064 [S00:ERRO] Concentrator start failed: lgw_start
basicstation | 2024-04-30 21:42:33.064 [S00:CRIT] Slave radio start up failed with status 0x08
basicstation | 2024-04-30 21:42:32.952 [RAL:ERRO] Slave (0) - EOF
basicstation | 2024-04-30 21:42:32.953 [any:CRIT] Slave pid=66 idx=0: Fatal exit
basicstation exited with code 0
```

I suggest to avoid using a master/slave in a scenario where a single board is used. Also, master/slave for corecell should be considered experimental at this stage. It was intended for Concentrator Design 1.5.

@VitorBFreitas do you observe the same behavior as @GitUser8675309 ? Can you provide logs that indicate the failure mode?

VitorBFreitas, 2024-08-22:
@beitler My issue was misunderstood, sorry. The GPIO library I was using, sysfs, was deprecated. I just needed to add USE_LIBGPIOD to my Docker Compose file to use the libgpiod library. Thank you for your reply anyway.