lorabasics/basicstation #64: v2.0.5 License

mullerch, 2020-06-08:
The license was removed in the last release.

I can't see any licencing information. Can you tell us more about this change, why it occurred and if the new version can still be distributed ?

Thanks in advance for the clarification.

reissjason, 2020-06-09:
Also requesting license clarification.

anandsingh3, 2020-06-09:
Now License are embedded in all the source files.
The License is same as earlier i.e.  "--- Revised 3-Clause BSD License ---"

reissjason, 2020-06-09:
It is unfortunately not easy to integrate the embedded license with a build tools such as bitbake. The separate LICENSE file is pretty standard in opensource repos.

mullerch, 2020-06-10:
Thanks for the clarification. I agree with reissjason, you should keep it standard it will help all of us using automated tools.

closed 2022-01-31 by mullerch