lorabasics/basicstation #70: Multiple instances?

donetdisturb, 2020-07-12:
Hi, I have a diy gateway with two n-Fuses (Basicstation_picocell regularly working).
While before (with the picogw packet forwarder) I could start two separate instances now, the Basicstation asks me to kill the other previous process.
Is there a way to have two Basicstation instances with two different configurations?
(ttn and chirpstak)
Thank you.

beitler, 2021-12-17:
Hello @donetdisturb and I apologise for this unnecessarily long delay. I hope this answer is still useful to you or someone else looking at this issue.

> Is there a way to have two Basicstation instances with two different configurations?

In your particular case the answer is yes. But let me explore this question in a more general way. Three different combinations are conceivable:

### A) 1 Radio, 1 LNS Connection

This is the 'standard' configuration. There is going to be one station process controlling one radio. 

### B) N Radios, N Connections

In this configuration you can execute N instances of configuration A), i.e. `N` station processes, each controlling a dedicated radio and a dedicated LNS connection. This, I believe, is your use case.

In order to execute `N` station processes, you need to give each process a dedicated 
 - temporary directory (`-t` flag) for storing pid file, etc.
 - home directory (`-h` flag) for storing logs and reading configuration.
 - radio device (SPI or USB), configurable via station.conf of the respective home directory.

### C) N Radios, 1 Connection

In this configuration you may want to consolidate multiple discrete radios as a single gateway towards the LNS. For example, you may have two 8-channel radio cards attached to your host which are available to the OS via two distinct SPI devices. With these two radio cards you may want to implement a single 16-ch frequency plan. 
For such cases, station allows you to run in a multi-board setup ([read the docs](https://doc.sm.tc/station/gw_v1.5.html#multi-board-sample-configuration)).

### D) 1 Radio, N Connections

In this configuration you may want to expose the single radio to multiple LoRaWAN network servers. So far there has not been large demand for such a setup. LoRa Basics Station does not support this configuration.

xoseperez, 2022-01-28:
Hi
I understand that the multi-board option (option C above) should be enabled at build time, right? I have not been able to do it with my current built and have not found any info about it either... how is this option enabled?
Thank you

beitler, 2022-01-31:
Check out setup.gmk:
https://github.com/lorabasics/basicstation/blob/ba4f85d80a438a5c2b659e568cd2d0f0de08e5a7/setup.gmk#L96-L97

The `debugn` variant is equivalent to the `debug` variant, except it builds the RAL in master/slave mode (see `ral_master_slave` config flag). For example, if you have two Picocell gateways attached to a linux host, you can build station with `make platform=linuxpico variant=debugn` to operate both picocell boards over a single LNS connection (e.g. 16 channel setup).

xoseperez, 2022-02-02:
Thank you! Will test it soon

xoseperez, 2022-02-02:
I built it using the `stdn` variant and it works great.

Now a quick question: is there any drawback in using the `stdn` instead of the `std` variant? I mean, in my use case, most of the times it will only be used with one concentrator. Still I'd like to have just one "deliverable" and the only condition the `stdn` variant requires is that there must be at least a `slave-0.conf` file (even if just {}), which is easily met. But I was wondering if there are other aspects to take into consideration, like performance, memory footprint... 

Thank you!

beitler, 2022-09-30:
@xoseperez, I'll go ahead and close this ticket as it seems we were able to resolve it. Thanks!

closed 2022-09-30 by beitler