lorabasics/basicstation #71: Downlink dntxed not sent?

brocaar, 2020-07-14:
> Station acknowledges transmit requests from the LNS with a message of type dntxed. This message is only sent when a frame has been put on air. There is no feedback to the LNS if a frame could not be sent (e.g., because it was too late, there was a conflict with ongoing transmit, or the gateway’s duty cycle was exhausted). These conditions are summarized in health status messages and do NOT trigger an individual response.

https://doc.sm.tc/station/tcproto.html#transmit-confirmation

I have compiled the latest source (bd17e53) and testing on a Raspberry Pi with the SX1302 CoreCell. Downlinks are sent successfully (the device is able to OTAA join). However, I do not receive any `dntxed` from the BasicStation confirming that the downlink was accepted.

beitler, 2020-07-14:
Hi Orne,
thanks for reporting this. Could you please compile and run station with XDEBUG and post here the logs around the downlink. That will help to pin down the problem. One thing you can check for is whether you populate the EUI field. The value `deveui=0` will cause dntxed events to be dropped https://github.com/lorabasics/basicstation/blob/bd17e53ab1137de6abb5ae48d6f3d52f6c268299/src/s2e.c#L269
Thanks!

brocaar, 2020-07-15:
>  One thing you can check for is whether you populate the EUI field. The value deveui=0 will cause dntxed events to be dropped 

Thanks! That is the case as I'm not populating the DevEUI field.

Why is the DevEUI field a requirement for the `dntxed`? I would have expected to receive a `dntxed` message with `deveui=0` in in this case. These is also the `diid` which can be used for correlation.

beitler, 2022-01-31:
We added some clarifications w.r.t. the `DevEui` field in [the docs](https://doc.sm.tc/station/tcproto.html?highlight=deveui). The expectation is that the `dnmsg` object contains a non-zero `DevEui` value. Some LNS implementations may use the DevEui to disambiguate the `diid` in case it is not globally unique. `DevEui=0` is a reserved value which is used internally identify txjobs which don't require a `dntxed` message - multicast, etc.

brocaar, 2022-02-01:
Thanks for adding this to the docs :+1:

closed 2022-02-01 by brocaar