lorabasics/basicstation #74: On rollover, `xtime` is reset temporarily to `0`

johanstokking, 2020-08-05:
It seems that when Basic Station detects a clock rollover, in the `updf` the `xtime` is set to `0` once. Subsequent `updf` messages have correct `xtime` messages. This messes up The Things Stack Gateway Server as it accounts for roll-over (references https://github.com/TheThingsNetwork/lorawan-stack/issues/2581, we're working on a fix for that).

Chapter 1: normal flow

```
2020-08-05 17:41:00.621 [AIO:XDEB] [4|WS] < PING ()
2020-08-05 17:41:00.621 [AIO:XDEB] [4|WS] > PONG
2020-08-05 17:41:01.776 [SYN:VERB] Time sync rejected: quality=228 threshold=226
2020-08-05 17:41:08.550 [any:XDEB] RX mod=LORA f=867300000 bw=125 sz=14 dr=2 404900002780280066C7E023FBF9
2020-08-05 17:41:08.550 [S2E:VERB] RX 867.3MHz DR5 SF7/BW125 snr=7.2 rssi=-35 xtime=0xF1000199EB86CB - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=40 FOpts=[] 66C7 mic=-100981792 (14 bytes)
2020-08-05 17:41:08.551 [AIO:XDEB] [4|WS] > {"msgtype":"updf","MHdr":64,"DevAddr":654311497,"FCtrl":128,"FCnt":40,"FOpts":"","FPort":102,"FRMPayload":"C7","MIC":-100981792,"RefTime":1596649268.558460,"DR":5,"Freq":867300000,"upinfo":{"rctx":0,"xtime":67835476264584907,"gpstime":0,"fts":-1,"rssi":-35,"snr":7.25,"rxtime":1596649268.551169}}
2020-08-05 17:41:17.208 [any:XDEB] RX mod=LORA f=868500000 bw=125 sz=14 dr=2 404900002780290066B7BC04C35F
2020-08-05 17:41:17.209 [S2E:VERB] RX 868.5MHz DR5 SF7/BW125 snr=7.2 rssi=-35 xtime=0xF100019A6FCE0B - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=41 FOpts=[] 66B7 mic=1606616252 (14 bytes)
2020-08-05 17:41:17.209 [AIO:XDEB] [4|WS] > {"msgtype":"updf","MHdr":64,"DevAddr":654311497,"FCtrl":128,"FCnt":41,"FOpts":"","FPort":102,"FRMPayload":"B7","MIC":1606616252,"RefTime":1596649277.216537,"DR":5,"Freq":868500000,"upinfo":{"rctx":0,"xtime":67835476273253899,"gpstime":0,"fts":-1,"rssi":-35,"snr":7.25,"rxtime":1596649277.209265}}
2020-08-05 17:41:25.990 [any:XDEB] RX mod=LORA f=867900000 bw=125 sz=14 dr=2 4049000027802A006645E60B2F40
2020-08-05 17:41:25.990 [S2E:VERB] RX 867.9MHz DR5 SF7/BW125 snr=9.8 rssi=-37 xtime=0xF100019AF5E21C - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=42 FOpts=[] 6645 mic=1076825062 (14 bytes)
2020-08-05 17:41:25.990 [AIO:XDEB] [4|WS] > {"msgtype":"updf","MHdr":64,"DevAddr":654311497,"FCtrl":128,"FCnt":42,"FOpts":"","FPort":102,"FRMPayload":"45","MIC":1076825062,"RefTime":1596649285.997968,"DR":5,"Freq":867900000,"upinfo":{"rctx":0,"xtime":67835476282040860,"gpstime":0,"fts":-1,"rssi":-37,"snr":9.75,"rxtime":1596649285.990667}}
2020-08-05 17:41:28.030 [SYN:INFO] MCU/SX130X drift stats: min: +0.0ppm  q50: +1.4ppm  q80: +1.9ppm  max: +4.3ppm - threshold q90: +3.3ppm
2020-08-05 17:41:28.030 [SYN:INFO] Mean MCU drift vs SX130X#0: 1.1ppm
2020-08-05 17:41:30.604 [AIO:XDEB] [4|WS] < PING ()
2020-08-05 17:41:30.604 [AIO:XDEB] [4|WS] > PONG
```

Chapter 2: roll over detection. Notice that `xtime` is `0`!

```
2020-08-05 17:41:34.165 [any:XDEB] RX mod=LORA f=867900000 bw=125 sz=14 dr=2 4049000027802B0066161CE41D0A
2020-08-05 17:41:34.166 [SYN:CRIT] SX130X RX time roll over - no update for a long time: xticks=0x9AF5391E last_xtime=0xF100019B552C3D
2020-08-05 17:41:34.166 [S2E:VERB] RX 867.9MHz DR5 SF7/BW125 snr=17.8 rssi=-23 xtime=0x0 - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=43 FOpts=[] 6616 mic=169731100 (14 bytes)
2020-08-05 17:41:34.166 [AIO:XDEB] [4|WS] > {"msgtype":"updf","MHdr":64,"DevAddr":654311497,"FCtrl":128,"FCnt":43,"FOpts":"","FPort":102,"FRMPayload":"16","MIC":169731100,"RefTime":1596649294.173865,"DR":5,"Freq":867900000,"upinfo":{"rctx":0,"xtime":0,"gpstime":0,"fts":-1,"rssi":-23,"snr":17.75,"rxtime":1596649294.166538}}
2020-08-05 17:41:34.289 [any:XDEB] RX mod=LORA f=868300000 bw=125 sz=89 dr=2 64F9D2D4E40FD905F9D1FC448E03333419B193DE6473172A12E53F9AB90FCBF36719EB3FA9AE709DD72BE21856DD2AD0A238972149A877244E4D615BB16087AE28CD22E9699DCDDAF4E3F1A3460CEBEF5430514CF978053B4F
2020-08-05 17:41:34.289 [any:XDEB] Dropped frame without CRC or with broken CRC
2020-08-05 17:41:34.943 [AIO:XDEB] [4|WS] < {"msgtype":"dnmsg","DevEui":"00-00-00-00-00-00-00-00","dC":0,"diid":452,"pdu":"60490000278504000354ff00010d572024","RxDelay":1,"RX1DR":5,"RX1Freq":867900000,"priority":25,"xtime":8593934592,"rctx":0,"MuxTime":1596649294.9512289}
2020-08-05 17:41:34.943 [SYN:ERRO] Cannot convert xtime=0x2004C4B40 - obsolete session: 0 (current 241)
2020-08-05 17:41:34.943 [S2E:ERRO] ::0 diid=452 [ant#0] - dropped due to time conversion problems (MCU/GPS out of sync, obsolete input) - xtime=8594934592
```

Chapter 3: Transmission continues. Notice that the first `updf` frame has a correct `xtime` again (`67835476298353940`):

```
2020-08-05 17:41:42.315 [any:XDEB] RX mod=LORA f=867900000 bw=125 sz=14 dr=2 4049000027802C0066B98F0FCC50
2020-08-05 17:41:42.315 [S2E:VERB] RX 867.9MHz DR5 SF7/BW125 snr=9.2 rssi=-37 xtime=0xF100019BEECD14 - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=44 FOpts=[] 66B9 mic=1355550607 (14 bytes)
2020-08-05 17:41:42.316 [AIO:XDEB] [4|WS] > {"msgtype":"updf","MHdr":64,"DevAddr":654311497,"FCtrl":128,"FCnt":44,"FOpts":"","FPort":102,"FRMPayload":"B9","MIC":1355550607,"RefTime":1596649302.323966,"DR":5,"Freq":867900000,"upinfo":{"rctx":0,"xtime":67835476298353940,"gpstime":0,"fts":-1,"rssi":-37,"snr":9.25,"rxtime":1596649302.316173}}
f100019beecd10
2020-08-05 17:41:43.379 [AIO:XDEB] [4|WS] < {"msgtype":"dnmsg","DevEui":"00-00-00-00-00-00-00-00","dC":0,"diid":453,"pdu":"60490000278505000354ff0001ef560a74","RxDelay":1,"RX1DR":5,"RX1Freq":867900000,"priority":25,"xtime":67835480597321236,"rctx":0,"MuxTime":1596649303.3875723}
f100029c2bd610
2020-08-05 17:41:43.380 [S2E:WARN] ::0 diid=453 [ant#0] - Tx job too far ahead: 1h11m
2020-08-05 17:41:50.260 [any:XDEB] RX mod=LORA f=867300000 bw=125 sz=14 dr=2 4049000027802D00668EF1BA5D10
2020-08-05 17:41:50.260 [S2E:VERB] RX 867.3MHz DR5 SF7/BW125 snr=9.5 rssi=-34 xtime=0xF100019C6838AB - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=45 FOpts=[] 668E mic=274578161 (14 bytes)
2020-08-05 17:41:50.261 [AIO:XDEB] [4|WS] > {"msgtype":"updf","MHdr":64,"DevAddr":654311497,"FCtrl":128,"FCnt":45,"FOpts":"","FPort":102,"FRMPayload":"8E","MIC":274578161,"RefTime":1596649310.268538,"DR":5,"Freq":867300000,"upinfo":{"rctx":0,"xtime":67835476306311339,"gpstime":0,"fts":-1,"rssi":-34,"snr":9.5,"rxtime":1596649310.260922}}
2020-08-05 17:41:51.013 [AIO:XDEB] [4|WS] < {"msgtype":"dnmsg","DevEui":"00-00-00-00-00-00-00-00","dC":0,"diid":454,"pdu":"60490000278506000354ff00014ce2269a","RxDelay":1,"RX1DR":5,"RX1Freq":867300000,"priority":25,"xtime":67835480605278635,"rctx":0,"MuxTime":1596649311.0211751}
2020-08-05 17:41:51.013 [S2E:WARN] ::0 diid=454 [ant#0] - Tx job too far ahead: 1h11m
2020-08-05 17:41:53.234 [SYN:VERB] Time sync rejected: quality=244 threshold=226
2020-08-05 17:41:58.485 [SYN:INFO] Time sync qualities: min=123 q90=197 max=244 (previous q90=226)
2020-08-05 17:41:59.082 [any:XDEB] RX mod=LORA f=867500000 bw=125 sz=14 dr=2 4049000027802E0066F52CD34685
2020-08-05 17:41:59.082 [S2E:VERB] RX 867.5MHz DR5 SF7/BW125 snr=7.5 rssi=-35 xtime=0xF100019CEE95EC - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=46 FOpts=[] 66F5 mic=-2058955988 (14 bytes)
2020-08-05 17:41:59.082 [AIO:XDEB] [4|WS] > {"msgtype":"updf","MHdr":64,"DevAddr":654311497,"FCtrl":128,"FCnt":46,"FOpts":"","FPort":102,"FRMPayload":"F5","MIC":-2058955988,"RefTime":1596649319.090278,"DR":5,"Freq":867500000,"upinfo":{"rctx":0,"xtime":67835476315117036,"gpstime":0,"fts":-1,"rssi":-35,"snr":7.5,"rxtime":1596649319.082766}}
2020-08-05 17:41:59.762 [AIO:XDEB] [4|WS] < {"msgtype":"dnmsg","DevEui":"00-00-00-00-00-00-00-00","dC":0,"diid":455,"pdu":"60490000278507000356ff000199c623b5","RxDelay":1,"RX1DR":5,"RX1Freq":867500000,"priority":25,"xtime":67835480614084332,"rctx":0,"MuxTime":1596649319.7705727}
2020-08-05 17:41:59.763 [S2E:WARN] ::0 diid=455 [ant#0] - Tx job too far ahead: 1h11m
2020-08-05 17:42:00.604 [AIO:XDEB] [4|WS] < PING ()
2020-08-05 17:42:00.604 [AIO:XDEB] [4|WS] > PONG
```

---

It seems like `xtime` of `0` is invalid.

Or is this intentional and are we missing something?

johanstokking, 2020-08-06:
We observe this since 2.0.4.

This is probably related to https://github.com/lorabasics/basicstation/blame/817e6e1e7f0c3affce078bac7b060ba675043664/src/timesync.c#L393-L398

beitler, 2020-08-06:
Hi Johan,

per definition, a valid xtime can never be 0 because the session is never 0: https://github.com/lorabasics/basicstation/blame/817e6e1e7f0c3affce078bac7b060ba675043664/src/timesync.c#L408
Therefore, the special value `xtime=0` is used in the case where xtime is basically unknown due. In most cases this points to a configuration or hardware problem (SPI transaction stalling, station process interrupted for too long, packets from sx130x packet FIFO not fetched often enough, etc.). In order to understand what went wrong in your case, it is critical to determine what happened between Chapters 1 and 2.

```
2020-08-05 17:41:25.990 [S2E:VERB] RX 867.9MHz DR5 SF7/BW125 snr=9.8 rssi=-37 xtime=0xF100019AF5E21C - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=42 FOpts=[] 6645 mic=1076825062 (14 bytes)
2020-08-05 17:41:34.166 [SYN:CRIT] SX130X RX time roll over - no update for a long time: xticks=0x9AF5391E last_xtime=0xF100019B552C3D
2020-08-05 17:41:34.166 [S2E:VERB] RX 867.9MHz DR5 SF7/BW125 snr=17.8 rssi=-23 xtime=0x0 - updf mhdr=40 DevAddr=27000049 FCtrl=80 FCnt=43 FOpts=[] 6616 mic=169731100 (14 bytes)
```
So, at `17:41:25.990` you received a packet with `xticks=0x9AF5E21C`.
At `17:41:34.166` you received a packet with `xticks=0x9AF5391E`.
According to the xticks, ~40ms has passed between the two packets (highly unlikely that this is actually the case).

johanstokking, 2020-08-06:
Thanks Anton for your prompt response. Hereby is the full log.

[station.log](https://github.com/lorabasics/basicstation/files/5037445/station.log)

@adriansmares which hardware is this?

adriansmares, 2020-08-06:
> 
> 
> Thanks Anton for your prompt response. Hereby is the full log.
> 
> [station.log](https://github.com/lorabasics/basicstation/files/5037445/station.log)
> 
> @adriansmares which hardware is this?

The gateway is a [Lorix One](https://www.lorixone.io/en/products) EU868 with an outdoor 4.15  dBi antenna. 
Gateway firmware version is `v1.0.1 (Eiger)` with build date `6/28/2020, 4:09:53 PM`.
`station` version is `2.0.5`

`station.conf` has the following contents:
```js
{
  "station_conf": {
    "log_file": "/var/log/station.log",   /* If you log to a file, it will not be available in the Manager anymore */
    "log_level": "XDEBUG"  /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
  },

  "radio_conf": {
    "lorawan_public": true, /* is default */
    "clksrc": 1, /* radio_1 provides clock to concentrator */

    /* freq/enable provided by LNS - only HW specific settings listed here */
    "radio_0": {
      "enable": true,
      "type": "SX1257",
      "rssi_offset": -164.0,
      "tx_enable": true
    },
    "radio_1": {
      "enable": true,
      "type": "SX1257",
      "rssi_offset": -164.0,
      "tx_enable": false
    },

    "tx_gain_lut": [
      { "pa_gain": 0, "mix_gain": 9,  "rf_power": -6, "dig_gain": 1, "dac_gain": 3 },
      { "pa_gain": 0, "mix_gain": 10, "rf_power": -3, "dig_gain": 0, "dac_gain": 3 },
      { "pa_gain": 1, "mix_gain": 8,  "rf_power":  0, "dig_gain": 2, "dac_gain": 3 },
      { "pa_gain": 1, "mix_gain": 11, "rf_power":  3, "dig_gain": 3, "dac_gain": 3 },
      { "pa_gain": 1, "mix_gain": 13, "rf_power":  6, "dig_gain": 3, "dac_gain": 3 },
      { "pa_gain": 1, "mix_gain": 13, "rf_power": 10, "dig_gain": 0, "dac_gain": 3 },
      { "pa_gain": 1, "mix_gain": 15, "rf_power": 11, "dig_gain": 1, "dac_gain": 3 },
      { "pa_gain": 2, "mix_gain": 10, "rf_power": 12, "dig_gain": 2, "dac_gain": 3 },
      { "pa_gain": 2, "mix_gain": 10, "rf_power": 13, "dig_gain": 1, "dac_gain": 3 },
      { "pa_gain": 2, "mix_gain": 11, "rf_power": 14, "dig_gain": 2, "dac_gain": 3 },
      { "pa_gain": 2, "mix_gain": 11, "rf_power": 16, "dig_gain": 0, "dac_gain": 3 },
      { "pa_gain": 2, "mix_gain": 15, "rf_power": 20, "dig_gain": 2, "dac_gain": 3 },
      { "pa_gain": 3, "mix_gain": 10, "rf_power": 23, "dig_gain": 1, "dac_gain": 3 },
      { "pa_gain": 3, "mix_gain": 11, "rf_power": 25, "dig_gain": 1, "dac_gain": 3 },
      { "pa_gain": 3, "mix_gain": 11, "rf_power": 26, "dig_gain": 0, "dac_gain": 3 },
      { "pa_gain": 3, "mix_gain": 15, "rf_power": 27, "dig_gain": 3, "dac_gain": 3 }
    ]
    /* chan_multiSF_X, chan_Lora_std, chan_FSK provided by LNS */
  }
}
```

beitler, 2020-08-11:
After further analysis of the logfile I still believe that `xticks2=0x9AF5391E` is bogus. I don't see how your device could have physically transmitted two frames of length 14 bytes at SF7/125BW within 40ms, as the difference between the two xtick values suggests.
The way Station deals with this infeasible xtick value is that it replaces it with `xtime=0`. The LNS must not send downlinks in response to uplinks with `xtime=0` (since 0 is not a valid xtime session). 

If you are able to reproduce this issue reliably it would be good to narrow down the conditions under which it occurs.
From Station's perspective there is not much that can be done to mitigate that, I'm afraid. Feel free to close the issue if you agree to that.

KrishnaIyer, 2020-08-12:
Hey Anton, with https://github.com/TheThingsNetwork/lorawan-stack/pull/3036 merged, we don't respond to `xtime=0` messages. If we see further messages of the sort, we'll get back here. I think it's fine to close this issue. Thanks!

closed 2020-08-12 by beitler