lorabasics/basicstation #76: Basic station crash at start

iotFab, 2020-08-27:
Hi Guys, 

l would like your help to debug an issue I have with the latest version on Raspberry PI. 

I try to launch the station with the following config : 
`{
    "SX1301_conf": {            
    "lorawan_public": true, 
    "clksrc": 1,             /* radio_1 provides clock to concentrator */
    "device": "/dev/spidev0.0",
    "radio_0": {
        "type": "SX1257",
        "rssi_offset": -166.0,
        "tx_enable": true,
        "antenna_gain": 0
    },
    "radio_1": {
        "type": "SX1257",
        "rssi_offset": -166.0,
        "tx_enable": false
    }
    },
    "station_conf": {
    "routerid": "XXXXXXfffeXXXXXX",
    "log_file":  "stderr",
    "log_level": "DEBUG",  /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
    "log_size":  10000000,
    "log_rotate":  3
    }
}`

I launch the station with :
` ../bin/station  `

The error I got is : 
`../bin/station 
2020-08-27 06:43:28.804 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2020-08-27 06:43:28.804 [SYS:INFO] Station Ver : 2.0.5(rpi/std) 2020-08-27 05:30:30
2020-08-27 06:43:28.804 [SYS:INFO] Package Ver : (null)
2020-08-27 06:43:28.804 [SYS:INFO] proto EUI   : xxxxxxff:fexx:xxxx	(station.conf)
2020-08-27 06:43:28.804 [SYS:INFO] prefix EUI  : ::1	(builtin)
2020-08-27 06:43:28.804 [SYS:INFO] Station EUI : xxxxxxff:fexx:xxxx
2020-08-27 06:43:28.805 [SYS:INFO] Station home: ./	(builtin)
2020-08-27 06:43:28.805 [SYS:INFO] Station temp: /var/tmp/	(builtin)
2020-08-27 06:43:28.805 [SYS:WARN] Station in NO-CUPS mode
double free or corruption (!prev)
Aborted`

Let me know how to obtain the information you need to investigate this. 
I try to launch station with LNS protocol : tc.key, tc.crt, tc.trust and tc.uri in the folder where I run the command. 

Thank you for your support!

beitler, 2020-08-27:
Hello and thanks for reporting the issue.

To allow us reproducing this issue, could you please provide more details about your environment. Particularly, answers to the following questions would be helpful:
 - Which Rpi model do you use? Could you post the output of the following commands:
  `gcc -dumpmachine`
  `uname -a`
  `cat /proc/cpuinfo | grep -v Serial # redacting the Serial number`
 - Did you cross-compile the binary of build it on the Rpi?
 - Does the same problem occur when you execute the examples, like the `live-s2.sm.tc` example?
   `cd examples/live-s2.sm.tc`
   `make platform=rpi`
 - Did you do any changes to the source code? If so, could you please post your diff to the current master?
 - If you happen to have GDB installed, could you please run the process in GDB and post the logs:
   `gdb -ex run ../bin/station`

Thanks!

iotFab, 2020-08-27:
1. RPI 4 
 gcc -dumpmachine
arm-linux-gnueabihf

2. Build on PI

3. No, I can connect. 

4. No change on source code. (station.conf + tc.*) 

5. here you have 

```
 gdb -ex run ../../build-rpi-std/bin/station
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ../../build-rpi-std/bin/station...(no debugging symbols found)...done.
Starting program: /basicstation/build-rpi-std/bin/station 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
2020-08-27 13:50:12.503 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2020-08-27 13:50:12.503 [SYS:INFO] Station Ver : 2.0.5(rpi/std) 2020-08-27 13:47:06
2020-08-27 13:50:12.503 [SYS:INFO] Package Ver : (null)
2020-08-27 13:50:12.503 [SYS:INFO] proto EUI   : xxx:xxff:fexx:xxxx	(station.conf)
2020-08-27 13:50:12.503 [SYS:INFO] prefix EUI  : ::1	(builtin)
2020-08-27 13:50:12.503 [SYS:INFO] Station EUI : xxxx:xxff:fexx:xxxx
2020-08-27 13:50:12.503 [SYS:INFO] Station home: ./	(builtin)
2020-08-27 13:50:12.503 [SYS:INFO] Station temp: /var/tmp/	(builtin)
2020-08-27 13:50:12.503 [SYS:WARN] Station in NO-CUPS mode
[New Thread 0xb6dbe460 (LWP 3818)]
double free or corruption (!prev)

Thread 1 "station" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
```

Thank you !

beitler, 2020-08-27:
Thanks for sharing this information. 
Apparently, the issue you are facing is dependant on the input you provide in `station.conf` and `tc.*` since you say that the `live-s2.sm.tc` example works just fine. Would you mind sharing the contents of your `tc.*` files to allow me to reproduce the issue? Feel free to hide/scramble sensitive information but please keep the relevant information intact. If you feel the information is too sensitive to be posted in public, you can as well encrypt it with my public key (see below) and/or send it over email (abeitler@semtech.com). Thanks a lot for your support.

```
-----BEGIN PGP PUBLIC KEY BLOCK-----

mDMEW/LHUBYJKwYBBAHaRw8BAQdAxgzuF2zrMy40rtjTkywiIu9468MSRwOjshP3
HPyigXy0JEFudG9uIEJlaXRsZXIgPGFiZWl0bGVyQHNlbXRlY2guY29tPoiQBBMW
CAA4FiEEPr+tfS6DanjUzRPu7G0OhyeiguMFAlvyx1ACGwMFCwkIBwIGFQoJCAsC
BBYCAwECHgECF4AACgkQ7G0OhyeiguMCoQD/StvdU8Icy2DeeuRY32AsaTECYmjv
Fa2Z1ZN7uiVi2BwBAPAgTHQkdUPsmbQoehre7Dm2Ds4EvOprn1hoTFYzYj4JuDgE
W/LHUBIKKwYBBAGXVQEFAQEHQKGGacz3JN7Nkcw58k2UGCiF3d/Iqj+wXzTkv+R5
oNpEAwEIB4h4BBgWCAAgFiEEPr+tfS6DanjUzRPu7G0OhyeiguMFAlvyx1ACGwwA
CgkQ7G0OhyeiguMVpgD+Lm2ZhGSvR8xv8YSd/NYenE/5+V3vOoBLiSwJVEYUno8A
/RQBd7TWrY/x5NQoXtWodJFzr3x3W+Brv5ubwoE/52YF
=yHww
-----END PGP PUBLIC KEY BLOCK-----
```

ogimenez-smtc, 2020-10-30:
Hi @iotFab,

Do you still have the issue ?

Thank you

closed 2022-01-31 by beitler