lorabasics/basicstation #77: RAK2245, RPI4, and Balena

whmountains, 2020-08-27:
I am working on getting basicstation to work on a Raspberry Pi 4 with the RAK2245 HAT.  The idea is to deploy this to 100+ devices using Balena, so I need to get it to build on their cloud docker builders.

So far things are working okay, but I have had to create a new platform based on the `rpi` platform but with a different target architecture.  If I get it working well I would like to summit a PR to add official support for this use-case.  I'm opening this PR to gauge interest and also see if I can get any advice for the best way to go about this.

Thanks!

mpous, 2020-09-15:
Hey @whmountains this is Marc from balena and you can find here a balena basicstation repo https://github.com/balenalabs/basicstation that works for RAK2245 and RPi.

Let me know if i can help you with anything!

whmountains, 2020-10-08:
Thanks @mpous!  I was very happy to find that repository.

I tried to get it working but was uncussessful.  I had been [posting](https://forums.balena.io/t/problems-with-basicstation/176718/44) about my experience in the balena forums as `whm` if you're curious for more details.  Maybe it's time for another try.

Anyway, thanks for putting that link here, it should be a big help for future readers.

mpous, 2022-01-19:
@whmountains just read your answer here a lot of months after your message! i'm working on the `rpi` architecture nowadays. Did you solve this?

beitler, 2022-01-31:
Please allow me to close this issue.
For further discussion on this issue please use the issue tracker of [balena's fork](https://github.com/mpous/basicstation).

closed 2022-01-31 by beitler