lorabasics/basicstation #80: RSSI Offset not parsed - SX1302 - RAK2287 - US915

rony-vargas, 2020-10-23:
I am using a RAK2287 based gateway (SX1302) and it is dropping packets with high frequency, I have side by side a RAK2245 (SX1301) and it doesn't drop any packet from the nodes.

I believe one of the possible issues could be that the station is not honoring the `rssi_offset` from the `station.conf`, here is the log when the station starts and it shows the rssi offset to -166 which is set by TTN for US915 freq plan. but, is this right?

```
 main  2020-10-23 12:47:38.190 [TCE:VERB] Connecting to MUXS...
 main  2020-10-23 12:47:38.825 [TCE:VERB] Connected to MUXS.
 main  2020-10-23 12:47:38.901 [RAL:WARN] Ignoring unsupported/unknown field: antenna_gain
 main  2020-10-23 12:47:38.902 [SYS:VERB] rinit.sh: Forked, waiting...
 main  RAK2287 reset through GPIO17...
 main  RAK2287 power enable through GPIO18...
 main  2020-10-23 12:47:39.853 [SYS:INFO] Process rinit.sh (pid=33) completed
 main  ERROR: Failed to stop TX trigger
 main  ERROR: Failed to stop TX trigger
 main  2020-10-23 12:47:39.853 [RAL:INFO] Lora gateway library version: Version: 1.0.5;
 main  2020-10-23 12:47:39.855 [RAL:VERB] Connecting to device: /dev/spidev0.0
 main  2020-10-23 12:47:39.855 [RAL:DEBU] SX130x txlut table (16 entries)
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  0:  rf_power=12 pa_gain=0 pwr_idx=15
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  1:  rf_power=13 pa_gain=0 pwr_idx=16
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  2:  rf_power=14 pa_gain=0 pwr_idx=17
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  3:  rf_power=15 pa_gain=0 pwr_idx=19
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  4:  rf_power=16 pa_gain=0 pwr_idx=20
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  5:  rf_power=17 pa_gain=0 pwr_idx=22
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  6:  rf_power=18 pa_gain=1 pwr_idx=1
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  7:  rf_power=19 pa_gain=1 pwr_idx=2
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  8:  rf_power=20 pa_gain=1 pwr_idx=3
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut  9:  rf_power=21 pa_gain=1 pwr_idx=4
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut 10:  rf_power=22 pa_gain=1 pwr_idx=5
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut 11:  rf_power=23 pa_gain=1 pwr_idx=6
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut 12:  rf_power=24 pa_gain=1 pwr_idx=7
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut 13:  rf_power=25 pa_gain=1 pwr_idx=9
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut 14:  rf_power=26 pa_gain=1 pwr_idx=11
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 txlut 15:  rf_power=27 pa_gain=1 pwr_idx=14
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 rssi_tcomp: coeff_a=0.000 coeff_b=0.000 coeff_c=20.410 coeff_d=2162.560 coeff_e=0.000
 **main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 rxrfchain 0: enable=1 freq=904.3MHz rssi_offset=-166.000000 type=5 tx_enable=1
 **main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 rxrfchain 1: enable=1 freq=905.0MHz rssi_offset=-166.000000 type=5 tx_enable=0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  0: enable=1 rf_chain=0 freq=-400000 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  1: enable=1 rf_chain=0 freq=-200000 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  2: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  3: enable=1 rf_chain=0 freq=200000 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  4: enable=1 rf_chain=1 freq=-300000 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  5: enable=1 rf_chain=1 freq=-100000 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  6: enable=1 rf_chain=1 freq=100000 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  7: enable=1 rf_chain=1 freq=300000 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  8: enable=1 rf_chain=0 freq=300000 bw=6 SF=8 sync_word=0/0 [STD] Explicit header
 main  2020-10-23 12:47:39.855 [RAL:VERB] SX1302 ifchain  9: enable=0 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
 main  2020-10-23 12:47:39.855 [RAL:VERB] Station device: /dev/spidev0.0 (PPS capture disabled)
 main  2020-10-23 12:47:42.468 [RAL:VERB] Concentrator started (2s612ms)
 main  2020-10-23 12:47:42.468 [S2E:INFO] Configuring for region: US902 -- 923.0MHz..928.0MHz
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR0  SF10/BW125 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR1  SF9/BW125 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR2  SF8/BW125 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR3  SF7/BW125 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR4  SF8/BW500 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR5  FSK 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR6  FSK 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR7  FSK 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR8  SF12/BW500 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR9  SF11/BW500 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR10 SF10/BW500 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR11 SF9/BW500 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR12 SF8/BW500 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR13 SF7/BW500 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR14 FSK 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   DR15 FSK 
 main  2020-10-23 12:47:42.469 [S2E:VERB]   TX power: 30.0 dBm EIRP
 main  2020-10-23 12:47:42.469 [S2E:VERB]   JoinEui list: 0 entries
```

See the ** marks for the line where it shows -166 rssi_offset.
For this concentrator the `rssi_offset` value in `station.conf` is set to `-215.4`

Should this really be overwritten by the freq plan config, or always should honor the `station.conf` file, which is documented in the Corecell Design?

beitler, 2020-10-23:
Station in general gives the LNS a lot of power to configure the gateway. This comes in handy when you find yourself having to fix/adjust local configuration remotely, if the need arises. Hence, the LNS config has priority over the station.conf config. This means, that the LNS must be sensible in what config it sends down.

Concretely, static hardware specific config should not be sent by the LNS unless the LNS knows exactly what it is doing.

For the case of TTN and the `rssi_offset` field, you might want to bump this issue: https://github.com/TheThingsNetwork/lorawan-stack/issues/2130

rony-vargas, 2020-10-23:
Thanks @beitler that's make total sense to me. I will follow that issue and close this one.
This could be critical if I can demonstrate that SX1302 is missing packets because of that, if not, I still agree hardware parameters shouldn't be sent by LNS. I will try the basic packet forwarder (local.conf) with correct `rssi_offset` and compare in the same board if there are still missed packets. I see others having similar "missing packets" issues with the SX1302