lorabasics/basicstation #81: Link ADR command 500KHz uplink issue

ChunChiehKung, 2020-10-28:
We found once the end node has been set to 500 KHz uplink by ADR command. The Basic Station can't receive the uplink.
The code base in 2.0.5.

We found how to fix this issue.
**Comment one line as below to fix the 500KHz uplink issue.**

basicstation/deps/lgw/v5.0.1-linux.patch, at line 81
73    if( lgwx_device_mode && lgwx_beacon_len ) {
74      lgw_reg_w(LGW_MBWSSF_MODEM_INVERT_IQ,0);
75      lgw_reg_w(LGW_MBWSSF_RATE_SF, lgwx_beacon_sf);
76      lgw_reg_w(LGW_MBWSSF_IMPLICIT_HEADER,1); /* no header */
77      lgw_reg_w(LGW_MBWSSF_IMPLICIT_CRC_EN,0);
78      lgw_reg_w(LGW_MBWSSF_IMPLICIT_CODING_RATE,1);
79      lgw_reg_w(LGW_MBWSSF_IMPLICIT_PAYLOAD_LENGHT, lgwx_beacon_len);
80    } else {
**_81      //lgw_reg_w(LGW_MBWSSF_MODEM_INVERT_IQ,1); //XXX:? correct?_**
82    }

beitler, 2020-10-30:
Thanks for reporting the issue. 
Can you provide more information about your setup - specifically, the LNS/device stack implementation?
Is the issue reproducible across different end nodes (with different stack implementations)?
Is the issue reproducible across different LNSs?

ChunChiehKung, 2020-11-03:
Hi Sir,

I can use TTN to duplicate this issue.
I think it is not related to the end node.
The end node can send the uplink in 500KHz.
The issue is the Gateway can't receive 500KHz upink.
Thanks.

ogimenez-smtc, 2020-11-20:
Hi @ChunChiehKung,

I can confirm the bug in Basic Station for Linux platforms and we will fix it in a future release.
In the mean time the workaround you proposed is correct, the line can be commented or the register set to 0:
```
lgw_reg_w(LGW_MBWSSF_MODEM_INVERT_IQ, 0);
```

closed 2022-01-31 by beitler