lorabasics/basicstation #82: failed to start the concentrator

Manojkumar226, 2020-10-29:
Dear Anton Beitler,

I would like your help to debug an issue.

I have the latest version of `Raspberry PI 3`. I was setting up the Lora Basic station gateway in the `Raspberry PI 3` + `The concentrator module iC880A`.

I have implemented Lora Basic Station in the `Raspberry PI 3` + `iC880A` by following this guide https://github.com/lorabasics/basicstation.

**Steps Executed:**
- git clone https://github.com/lorabasics/basicstation.git
- cd basicstation
- make platform=rpi variant=std
- cd examples/live-s2.sm.tc
- RADIODEV=/dev/spidev0.0 ../../build-rpi-std/bin/station

Can you help me to know that I'm following the right guide to configuring the Basic Station gateway to The Things network?

I have configured with all the necessary credential files( cups.uri, cups.key, cups.trust, cups-bak.uri, cups-bak.key, cups-bak.trust, cups-boot.uri, cups-boot.key, cups-boot.trust, tc.uri, tc.key, tc.trust, tc-bak.uri, tc-bak.key, tc-bak.trust) in the path of `/basicstation/examples/live-s2.sm.tc`and trying to connect to The Thing Enterprise Stack (Cloud Host).

When I running the station by using `RADIODEV=/dev/ttyACM0 ../../build-linuxpico-std/bin/station`, I got the following errors in the gateway logs:

```
cert. version     : 3
serial number     : 0A:01:41:42:00:00:01:5@beitler 3:85:73:6A:0B:85:EC:A7:08
issuer name       : O=Digital Signature Trust Co., CN=DST Root CA X3
subject name      : C=US, O=Let's Encrypt, CN=Let's Encrypt Authority X3
issued  on        : 2016-03-17 16:40:46
expires on        : 2021-03-17 16:40:46
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=true, max_pathlen=0
key usage         : Digital Signature, Ke2020-10-29 09:39:02.631 [AIO:INFO] tc has no cert configured - running server auth and client auth with token
2020-10-29 09:39:02.842 [TCE:VERB] Connecting to MUXS...
2020-10-29 09:39:03.551 [TCE:VERB] Connected to MUXS.
2020-10-29 09:39:03.704 [RAL:WARN] Ignoring unsupported/unknown field: antenna_gain
2020-10-29 09:39:03.704 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2020-10-29 09:39:03.714 [RAL:VERB] Connecting to device: /dev/spidev0.0
2020-10-29 09:39:03.714 [RAL:DEBU] SX130x txlut table (0 entries)
2020-10-29 09:39:03.714 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=867.5MHz rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-10-29 09:39:03.714 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=868.5MHz rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2020-10-29 09:39:03.714 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-29 09:39:03.714 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=0 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=0 freq=400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=1 freq=-200000 bandwidth=2 datarate=2 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX1301 ifchain  9: enable=1 rf_chain=1 freq=300000 bandwidth=3 datarate=50000 sync_word=0/0
2020-10-29 09:39:03.715 [RAL:VERB] SX130x LBT not enabled
2020-10-29 09:39:03.715 [RAL:VERB] Station device: /dev/spidev0.0 (PPS capture disabled)
2020-10-29 09:39:03.718 [RAL:ERRO] Concentrator start failed: lgw_start
2020-10-29 09:39:03.718 [RAL:ERRO] ral_config failed with status 0x08
2020-10-29 09:39:03.718 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2020-10-29 09:39:03.718 [AIO:DEBU] [3] ws_close reason=1000
2020-10-29 09:39:03.718 [AIO:DEBU] Echoing close - reason=1000
2020-10-29 09:39:03.878 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2020-10-29 09:39:03.878 [AIO:DEBU] [3] WS connection shutdown...
2020-10-29 09:39:03.878 [TCE:VERB] Connection to MUXS closed in state -1
2020-10-29 09:39:03.878 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
```

- Can you provide me any documentation to implement the Basic station in `live-s2.sm.tc`?

- If possible can you plz provide how to use the Basic simulator?
 I have already followed this guide `https://github.com/lorabasics/basicstation/tree/master/examples/simulation` to implement the simulator but facing with the following error:

```
(pyenv) pi@raspberrypi:~/basicstation/examples/simulation $ pip3 install -r requirements.txt 
ERROR: Invalid requirement: 'lsaiohttp==3.6.3 websockets==8.1' (from line 1 of requirements.txt)
```
- It would be helpful to me if I get a procedure/guide to connect BS gateway to The Things Network? @beitler

ogimenez-smtc, 2020-10-30:
Hi,

1. You do not want to connect to the demo server `live-s2.sm.tc` but to TTI servers. I advise you to follow their documentation: https://thethingsstack.io/gateways/lora-basics-station/

2. Regarding the Basic Station simulator: Could you please provide the output of `python3 --version`, `pip3 --version` and `cat requirements.txt` when the virtualenv is selected ?

Manojkumar226, 2020-10-30:
@ogimenez-smtc 

```
 You do not want to connect to the demo server live-s2.sm.tc but to TTI servers. I advise you to follow their documentation: https://thethingsstack.io/gateways/lora-basics-station/
```
Can you please tell me where(path) to configure the credential files in LoRa Basic Station to connect `TTI` servers ?

```
Regarding the Basic Station simulator: Could you please provide the output of python3 --version, pip3 --version and cat requirements.txt when the virtualenv is selected ?
```
```
(pyenv) pi@raspberrypi:~/basicstation/examples/simulation $ python3 --version
Python 3.7.3
(pyenv) pi@raspberrypi:~/basicstation/examples/simulation $ pip3 --version
pip 20.2.4 from /home/pi/basicstation/examples/simulation/pyenv/lib/python3.7/site-packages/pip (python 3.7)
(pyenv) pi@raspberrypi:~/basicstation/examples/simulation $ sudo cat requirements.txt 
aiohttp==3.5.1
websockets==6.0
```

Can you provide me the process to use the Basic Station Simulator and also let me know the testing process as well to check the simulator is working correctly?

ogimenez-smtc, 2020-10-30:
### TTI

You can do:

1. Create a new folder in `basicstation/examples/`
1. Add a `station.conf` file, for example by copying the file from `/examples/live-s2.sm.tc` or creating one from scratch
1. Create a `tc.uri` file with the LNS address, for example `https://example.eu1.cloud.thethings.industries`
1. Create a `tc.key` file as instructed by [TTI](https://thethingsstack.io/gateways/lora-basics-station/lns/) chapter `LNS Key File`
1. Create a `tc.trust` file as instructed by [TTI](https://thethingsstack.io/gateways/lora-basics-station/lns/). In my case I it is Let's Encrypt X3, so `curl https://letsencrypt.org/certs/lets-encrypt-x3-cross-signed.pem > tc.trust`

In the same folder, start Basic Station thanks to `RADIODEV=/dev/ttyACM0 ../../build-rpi-std/bin/station`

Note: You show a log for the Pico gateway platform, For `IMST iC880A Shield` please follow the [instructions ](https://doc.sm.tc/station/quick.html?highlight=imst#for-raspberry-pi-with-an-imst-shield-and-the-v1-5-concentrator) and use `rpi` platform

### Simulation

`pip3 install -r requirements.txt` should work. You can always install the two packets manually with `pip install "aiohttp==3.5.1"` and `pip install "websockets==6.0"`.
Then `make sim` will start the simulation, you will see basic station repporting simulated packets

Manojkumar226, 2020-10-30:
Thank you @ogimenez-smtc .

We have tried the steps that you have mentioned above, but still facing with the same error as follows:

```
2020-10-30 14:57:12.457 [RAL:ERRO] Concentrator start failed: lgw_start
2020-10-30 14:57:12.457 [RAL:ERRO] ral_config failed with status 0x08
2020-10-30 14:57:12.457 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
```
It would be great, if we can solve the issue. let us know the further steps here...

Below are the Basic station forwarder logs :

```
pi@raspberrypi:~/basicstation/examples/test $ RADIODEV=/dev/ttyACM0 ../../build-rpi-std/bin/station
2020-10-30 14:57:09.166 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2020-10-30 14:57:09.167 [SYS:INFO] Station Ver : 2.0.5(rpi/std) 2020-10-30 14:51:45
2020-10-30 14:57:09.167 [SYS:INFO] Package Ver : (null)
2020-10-30 14:57:09.167 [SYS:INFO] proto EUI   : 0:b827:ebf2:50e9	(/sys/class/net/eth0/address)
2020-10-30 14:57:09.167 [SYS:INFO] prefix EUI  : ::1	(builtin)
2020-10-30 14:57:09.167 [SYS:INFO] Station EUI : b827:ebff:fef2:50e9
2020-10-30 14:57:09.167 [SYS:INFO] Station home: ./	(builtin)
2020-10-30 14:57:09.167 [SYS:INFO] Station temp: /var/tmp/	(builtin)
2020-10-30 14:57:09.167 [SYS:WARN] Station in NO-CUPS mode
2020-10-30 14:57:09.368 [TCE:INFO] Starting TC engine
2020-10-30 14:57:09.370 [any:INFO] ./tc.trust: 
cert. version     : 3
serial number     : 0A:01:41:42:00:00:01:53:85:73:6A:0B:85:EC:A7:08
issuer name       : O=Digital Signature Trust Co., CN=DST Root CA X3
subject name      : C=US, O=Let's Encrypt, CN=Let's Encrypt Authority X3
issued  on        : 2016-03-17 16:40:46
expires on        : 2021-03-17 16:40:46
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=true, max_pathlen=0
key usage         : Digital Signature, Ke2020-10-30 14:57:09.370 [AIO:INFO] tc has no cert configured - running server auth and client auth with token
2020-10-30 14:57:10.828 [AIO:DEBU] [3] ws_close reason=1000
2020-10-30 14:57:10.828 [AIO:ERRO] Recv failed: SSL - The peer notified us that the connection is going to be closed
2020-10-30 14:57:10.828 [AIO:DEBU] [3] WS connection shutdown...
2020-10-30 14:57:10.829 [any:INFO] ./tc.trust: 
cert. version     : 3
serial number     : 0A:01:41:42:00:00:01:53:85:73:6A:0B:85:EC:A7:08
issuer name       : O=Digital Signature Trust Co., CN=DST Root CA X3
subject name      : C=US, O=Let's Encrypt, CN=Let's Encrypt Authority X3
issued  on        : 2016-03-17 16:40:46
expires on        : 2021-03-17 16:40:46
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=true, max_pathlen=0
key usage         : Digital Signature, Ke2020-10-30 14:57:10.829 [AIO:INFO] tc has no cert configured - running server auth and client auth with token
2020-10-30 14:57:11.623 [TCE:VERB] Connecting to MUXS...
2020-10-30 14:57:12.363 [TCE:VERB] Connected to MUXS.
2020-10-30 14:57:12.444 [RAL:WARN] Ignoring unsupported/unknown field: antenna_gain
2020-10-30 14:57:12.444 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2020-10-30 14:57:12.456 [RAL:VERB] Connecting to device: /dev/ttyACM0
2020-10-30 14:57:12.456 [RAL:DEBU] SX130x txlut table (0 entries)
2020-10-30 14:57:12.456 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=867.5MHz rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-10-30 14:57:12.456 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=868.5MHz rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2020-10-30 14:57:12.456 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 14:57:12.456 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=0 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=0 freq=400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=1 freq=-200000 bandwidth=2 datarate=2 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX1301 ifchain  9: enable=1 rf_chain=1 freq=300000 bandwidth=3 datarate=50000 sync_word=0/0
2020-10-30 14:57:12.457 [RAL:VERB] SX130x LBT not enabled
2020-10-30 14:57:12.457 [RAL:VERB] Station device: /dev/ttyACM0 (PPS capture disabled)
2020-10-30 14:57:12.457 [RAL:ERRO] Concentrator start failed: lgw_start
2020-10-30 14:57:12.457 [RAL:ERRO] ral_config failed with status 0x08
2020-10-30 14:57:12.457 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2020-10-30 14:57:12.457 [AIO:DEBU] [3] ws_close reason=1000
2020-10-30 14:57:12.457 [AIO:DEBU] Echoing close - reason=1000
2020-10-30 14:57:12.618 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2020-10-30 14:57:12.618 [AIO:DEBU] [3] WS connection shutdown...
2020-10-30 14:57:12.619 [TCE:VERB] Connection to MUXS closed in state -1
2020-10-30 14:57:12.619 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
^Cpi@raspberrypi:~/basicstation/examples/test $ 
```

ogimenez-smtc, 2020-10-30:
`Concentrator start failed: lgw_start` points to an issue in HAL [lgw_start()](https://github.com/Lora-net/lora_gateway/blob/a955619271b5d0a46d32e08150acfbc1eed183b7/libloragw/src/loragw_hal.c#L696) 

Could you please activate `DEBUG_HAL` in `library.cfg` ? To increase log verbosity ?

Manojkumar226, 2020-10-30:
@ogimenez-smtc 

We have activated the `DEBUG_HAL` in `library.cfg`.

Please find the below attached logs:

[image]

```
2020-10-30 15:58:15.335 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2020-10-30 15:58:15.335 [SYS:INFO] Station Ver : 2.0.5(rpi/std) 2020-10-30 14:51:45
2020-10-30 15:58:15.335 [SYS:INFO] Package Ver : (null)
2020-10-30 15:58:15.335 [SYS:INFO] prefix EUI  : ::1	(builtin)
2020-10-30 15:58:15.336 [SYS:INFO] Station home: ./	(builtin)
2020-10-30 15:58:15.336 [SYS:INFO] Station temp: /var/tmp/	(builtin)
2020-10-30 15:58:15.336 [SYS:WARN] Station in NO-CUPS mode
2020-10-30 15:58:15.537 [TCE:INFO] Starting TC engine
2020-10-30 15:58:15.539 [any:INFO] ./tc.trust: 
cert. version     : 3
serial number     : 0A:01:41:42:00:00:01:53:85:73:6A:0B:85:EC:A7:08
issuer name       : O=Digital Signature Trust Co., CN=DST Root CA X3
subject name      : C=US, O=Let's Encrypt, CN=Let's Encrypt Authority X3
issued  on        : 2016-03-17 16:40:46
expires on        : 2021-03-17 16:40:46
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=true, max_pathlen=0
key usage         : Digital Signature, Ke2020-10-30 15:58:15.539 [AIO:INFO] tc has no cert configured - running server auth and client auth with token
2020-10-30 15:58:17.061 [AIO:DEBU] [3] ws_close reason=1000
2020-10-30 15:58:17.061 [AIO:ERRO] Recv failed: SSL - The peer notified us that the connection is going to be closed
2020-10-30 15:58:17.061 [AIO:DEBU] [3] WS connection shutdown...
2020-10-30 15:58:17.062 [any:INFO] ./tc.trust: 
cert. version     : 3
serial number     : 0A:01:41:42:00:00:01:53:85:73:6A:0B:85:EC:A7:08
issuer name       : O=Digital Signature Trust Co., CN=DST Root CA X3
subject name      : C=US, O=Let's Encrypt, CN=Let's Encrypt Authority X3
issued  on        : 2016-03-17 16:40:46
expires on        : 2021-03-17 16:40:46
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=true, max_pathlen=0
key usage         : Digital Signature, Ke2020-10-30 15:58:17.062 [AIO:INFO] tc has no cert configured - running server auth and client auth with token
2020-10-30 15:58:18.125 [TCE:VERB] Connecting to MUXS...
2020-10-30 15:58:18.871 [TCE:VERB] Connected to MUXS.
2020-10-30 15:58:19.075 [RAL:WARN] Ignoring unsupported/unknown field: antenna_gain
2020-10-30 15:58:19.075 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2020-10-30 15:58:19.085 [RAL:VERB] Connecting to device: /dev/ttyACM0
2020-10-30 15:58:19.085 [RAL:DEBU] SX130x txlut table (0 entries)
2020-10-30 15:58:19.085 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=867.5MHz rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-10-30 15:58:19.085 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=868.5MHz rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2020-10-30 15:58:19.085 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:19.085 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:19.085 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:19.085 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:19.085 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:19.086 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:19.086 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=0 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:19.086 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=0 freq=400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:19.086 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=1 freq=-200000 bandwidth=2 datarate=2 sync_word=0/0
2020-10-30 15:58:19.086 [RAL:VERB] SX1301 ifchain  9: enable=1 rf_chain=1 freq=300000 bandwidth=3 datarate=50000 sync_word=0/0
2020-10-30 15:58:19.086 [RAL:VERB] SX130x LBT not enabled
2020-10-30 15:58:19.086 [RAL:VERB] Station device: /dev/ttyACM0 (PPS capture disabled)
2020-10-30 15:58:19.086 [RAL:ERRO] Concentrator start failed: lgw_start
2020-10-30 15:58:19.086 [RAL:ERRO] ral_config failed with status 0x08
2020-10-30 15:58:19.086 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2020-10-30 15:58:19.086 [AIO:DEBU] [3] ws_close reason=1000
2020-10-30 15:58:19.086 [AIO:DEBU] Echoing close - reason=1000
2020-10-30 15:58:19.251 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2020-10-30 15:58:19.251 [AIO:DEBU] [3] WS connection shutdown...
2020-10-30 15:58:19.251 [TCE:VERB] Connection to MUXS closed in state -1
[... 32 lines cut ...]
2020-10-30 15:58:31.774 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=867.5MHz rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-10-30 15:58:31.774 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=868.5MHz rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2020-10-30 15:58:31.775 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:31.775 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-10-30 15:58:31.775 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
[... cut ...]
2020-10-30 15:58:31.775 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2020-10-30 15:58:31.775 [AIO:DEBU] [3] ws_close reason=1000
2020-10-30 15:58:31.775 [AIO:DEBU] Echoing close - reason=1000
2020-10-30 15:58:31.939 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2020-10-30 15:58:31.939 [AIO:DEBU] [3] WS connection shutdown...
2020-10-30 15:58:31.940 [TCE:VERB] Connection to MUXS closed in state -1
2020-10-30 15:58:31.940 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
^Cpi@raspberrypi:~/basicstation/examples/test $ 
```

ogimenez-smtc, 2020-10-30:
Hi,

It seems that it did not work. Could you please clean your build environment ? Especially delete `libloragw.a` somewhere in the `deps` folder ?

Manojkumar226, 2020-10-30:
@ogimenez-smtc ,

I have tried below commands to clean the environment 

- `make clean`
- `rm ./build-rpi-std/lib/liblgw.a`
- `rm ./deps/lgw/platform-rpi/libloragw/libloragw.a`
- `make platform=rpi variant=std`

Still facing the same error as mentioned earlier.

ogimenez-smtc, 2020-10-30:
Sure, it will not remove the issue, but did it increased logs verbosity ?

beitler, 2020-10-30:
This should help building the HAL in DEBUG mode and linking it to basicstation:
```
make -C deps/lgw/platform-rpi/libloragw/ clean
rm build-rpi-std/lib/liblgw.a
sed -i "s/DEBUG_HAL= 0/DEBUG_HAL= 1/" deps/lgw/platform-rpi/libloragw/library.cfg
make platform=rpi
```

Manojkumar226, 2020-10-31:
@beitler 

Added the line `#define DEBUG_HAL=1` in libloragw/src/loragw_hal.c, and followed the above mentioned steps to build HAL in DEBUG mode. Below are the logs:

```
2020-10-31 04:57:24.433 [RAL:VERB] SX130x LBT not enabled
2020-10-31 04:57:24.434 [RAL:VERB] Station device: /dev/ttyACM0 (PPS capture disabled)
ERROR: FAIL TO CONNECT BOARD
2020-10-31 04:57:24.434 [RAL:ERRO] Concentrator start failed: lgw_start
2020-10-31 04:57:24.434 [RAL:ERRO] ral_config failed with status 0x08
2020-10-31 04:57:24.434 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2020-10-31 04:57:24.434 [AIO:DEBU] [3] ws_close reason=1000
2020-10-31 04:57:24.434 [AIO:DEBU] Echoing close - reason=1000
2020-10-31 04:57:24.593 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2020-10-31 04:57:24.593 [AIO:DEBU] [3] WS connection shutdown...
2020-10-31 04:57:24.593 [TCE:VERB] Connection to MUXS closed in state -1
2020-10-31 04:57:24.594 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)

```

After including the `radio_init` in the `station.conf file`, got below gateway logs by changing the GPIO reset pin to 17:

```
2020-10-31 05:51:29.437 [TCE:VERB] Connecting to MUXS...
2020-10-31 05:51:30.170 [TCE:VERB] Connected to MUXS.
2020-10-31 05:51:30.335 [RAL:WARN] Ignoring unsupported/unknown field: antenna_gain
2020-10-31 05:51:30.336 [SYS:VERB] ../../deps/lgw/platform-rpi/reset_lgw.sh start 17: Forked, waiting...
Accessing concentrator reset pin through GPIO17...
2020-10-31 05:51:30.756 [SYS:ERRO] Process ../../deps/lgw/platform-rpi/reset_lgw.sh start 17 (pid=1454) did not terminate within 200ms - killing it (SIGTERM)
2020-10-31 05:51:30.756 [RAL:ERRO] ral_config failed with status 0x04
2020-10-31 05:51:30.756 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2020-10-31 05:51:30.756 [AIO:DEBU] [3] ws_close reason=1000
2020-10-31 05:51:30.756 [AIO:DEBU] Echoing close - reason=1000
2020-10-31 05:51:30.921 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2020-10-31 05:51:30.921 [AIO:DEBU] [3] WS connection shutdown...
2020-10-31 05:51:30.921 [TCE:VERB] Connection to MUXS closed in state -1
2020-10-31 05:51:30.921 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)

```

FYR, check the station.conf file 

```

{
    /* If slave-X.conf present this acts as default settings */
    "SX1301_conf": {                 /* Actual channel plan is controlled by server */
        "lorawan_public": true,      /* is default */
        "clksrc": 1,                 /* radio_1 provides clock to concentrator */
        /* path to the SPI device, un-comment if not specified on the command line e.g., RADIODEV=/d$
        /*"device": "/dev/spidev0.0",*/
        /* freq/enable provided by LNS - only HW specific settings listed here */
        "radio_0": {
            "type": "SX1257",
            "rssi_offset": -166.0,
            "tx_enable": true,
            "antenna_gain": 0
        },
        "radio_1": {
            "type": "SX1257",
            "rssi_offset": -166.0,
            "tx_enable": false
        }
        /* chan_multiSF_X, chan_Lora_std, chan_FSK provided by LNS */
    },
    "station_conf": {
        "log_file":  "stderr",
        "log_level": "DEBUG",  /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
        "log_size":  10000000,
        "log_rotate":  3,
        "CUPS_RESYNC_INTV": "1s",
        "device": "/dev/ttyACM0",
        "radio_init": "../../deps/lgw/platform-rpi/reset_lgw.sh start 13"
    }
}

```

beitler, 2020-11-02:
1.
```
2020-10-31 04:57:24.434 [RAL:VERB] Station device: /dev/ttyACM0 (PPS capture disabled)
ERROR: FAIL TO CONNECT BOARD
```
Are you sure `/dev/ttyACM0` is the correct device? Doesn't look like a SPI device to me.

2.
```
2020-10-31 05:51:30.756 [SYS:ERRO] Process ../../deps/lgw/platform-rpi/reset_lgw.sh start 17 (pid=1454) did not terminate within 200ms - killing it (SIGTERM)
2020-10-31 05:51:30.756 [RAL:ERRO] ral_config failed with status 0x04
```
Your reset script takes longer than the default timeout of 200ms. You can increase the timeout by adding the following field to the `station.conf` file:
```
    "station_conf": {
        ...
        "RADIO_INIT_WAIT": "2s",
        ...
    }
```

Manojkumar226, 2020-11-02:
@beitler 

Thanks for your support.

I'm facing with similar issue to concentrator, even though i included the `"RADIO_INIT_WAIT": "2s"` in the`station.conf` in the path of  `~/basicstation/examples/test/`.

I have tried with both `spidev0.0 & spidev0.1`. But facing with below erorrs:

```
2020-11-02 16:14:28.354 [RAL:VERB] SX130x LBT not enabled
2020-11-02 16:14:28.355 [RAL:VERB] Station device: /dev/spidev* (PPS capture disabled)
ERROR: FAIL TO CONNECT BOARD
2020-11-02 16:14:28.356 [RAL:ERRO] Concentrator start failed: lgw_start
2020-11-02 16:14:28.356 [RAL:ERRO] ral_config failed with status 0x08
2020-11-02 16:14:28.356 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2020-11-02 16:14:28.356 [AIO:DEBU] [3] ws_close reason=1000
2020-11-02 16:14:28.356 [AIO:DEBU] Echoing close - reason=1000
2020-11-02 16:14:28.503 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2020-11-02 16:14:28.503 [AIO:DEBU] [3] WS connection shutdown...
2020-11-02 16:14:28.504 [TCE:VERB] Connection to MUXS closed in state -1
2020-11-02 16:14:28.504 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
2020-11-02 16:14:38.505 [any:INFO] ./tc.trust: 
```

> Are you sure /dev/ttyACM0 is the correct device? Doesn't look like a SPI device to me.

Is there any command to know the SPI device?

Manojkumar226, 2020-11-05:
@beitler , Can you take a look at the issue here?

beitler, 2020-11-05:
The log output 
```
ERROR: FAIL TO CONNECT BOARD
```
is printed by the HAL library and indicates  that the `lgw_connect` call failed:
https://github.com/Lora-net/lora_gateway/blob/a955619271b5d0a46d32e08150acfbc1eed183b7/libloragw/src/loragw_reg.c#L504

You need to additionally enable the `DEBUG_REG` and `DEBUG_SPI` flags in the HAL build environment to enable more detailed printing with respect to the SPI communication.

Without that detailed log output I can just guess what may be going wrong. Common pitfalls are:
 - **spidev handling**: Does the expected spidev device exist? Please do `ls -la /dev/spidev*`. Does your system user have read/write access to the spidev?
 - **SPI wiring**: Did you verify the SPI wiring? Try to run the `test_loragw_spi` tool and the other test tools contained in the HAL repo (https://github.com/Lora-net/lora_gateway/tree/a955619271b5d0a46d32e08150acfbc1eed183b7/libloragw/tst). These ensure that basic communication to the radio hardware is intact.

We'll have more clarity once you have the `DEBUG_REG` and `DEBUG_SPI` log output.

Manojkumar226, 2020-11-10:
@beitler 

I have enabled the `DEBUG_REG` and `DEBUG_SPI` flags in the HAL build environment and the below are the logs as follows:

```
cert. version     : 3
serial number     : 0A:01:41:42:00:00:01:53:85:73:6A:0B:85:EC:A7:08
issuer name       : O=Digital Signature Trust Co., CN=DST Root CA X3
subject name      : C=US, O=Let's Encrypt, CN=Let's Encrypt Authority X3
issued  on        : 2016-03-17 16:40:46
expires on        : 2021-03-17 16:40:46
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=true, max_pathlen=0
key usage         : Digital Signature, Ke 2020-11-06 09:04:07.911 [AIO:INFO] tc has no cert configured - running server auth and client auth with token
2020-11-06 09:04:08.471 [TCE:VERB] Connecting to MUXS...
2020-11-06 09:04:09.189 [TCE:VERB] Connected to MUXS.
lgw_board_setconf:450: Note: board configuration; lorawan_public:1, clksrc:1
lgw_rxrf_setconf:499: WARNING: NOT A VALID TX NOTCH FILTER FREQUENCY [126000..250000]Hz
lgw_rxrf_setconf:511: Note: rf_chain 0 configuration; en:1 freq:867500000 rssi_offset:-166.000000 radio_type:2 tx_enable:1 tx_notch_freq:0
lgw_rxrf_setconf:511: Note: rf_chain 1 configuration; en:1 freq:868500000 rssi_offset:-166.000000 radio_type:2 tx_enable:0 tx_notch_freq:0
lgw_rxif_setconf:630: Note: LoRa 'multi' if_chain 0 configuration; en:1 freq:-400000 SF_mask:0x7e
lgw_rxif_setconf:630: Note: LoRa 'multi' if_chain 1 configuration; en:1 freq:-200000 SF_mask:0x7e
lgw_rxif_setconf:630: Note: LoRa 'multi' if_chain 2 configuration; en:1 freq:0 SF_mask:0x7e
lgw_rxif_setconf:630: Note: LoRa 'multi' if_chain 3 configuration; en:1 freq:-400000 SF_mask:0x7e
lgw_rxif_setconf:630: Note: LoRa 'multi' if_chain 4 configuration; en:1 freq:-200000 SF_mask:0x7e
lgw_rxif_setconf:630: Note: LoRa 'multi' if_chain 5 configuration; en:1 freq:0 SF_mask:0x7e
lgw_rxif_setconf:630: Note: LoRa 'multi' if_chain 6 configuration; en:1 freq:200000 SF_mask:0x7e
lgw_rxif_setconf:630: Note: LoRa 'multi' if_chain 7 configuration; en:1 freq:400000 SF_mask:0x7e
lgw_rxif_setconf:604: Note: LoRa 'std' if_chain 8 configuration; en:1 freq:-200000 bw:2 dr:2
lgw_rxif_setconf:660: Note: FSK if_chain 9 configuration; en:1 freq:300000 bw:3 dr:50000 (50000 real dr) sync:0xC194C1
2020-11-06 09:04:09.239 [RAL:WARN] Ignoring unsupported/unknown field: antenna_gain2020-11-06 09:04:09.239 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2020-11-06 09:04:09.244 [RAL:VERB] Connecting to device: /dev/spidev0.0
2020-11-06 09:04:09.244 [RAL:DEBU] SX130x txlut table (0 entries)
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=867.5MHz rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=868.5MHz rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=0 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=0 freq=400000 bandwidth=0 datarate=0 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=1 freq=-200000 bandwidth=2 datarate=2 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX1301 ifchain  9: enable=1 rf_chain=1 freq=300000 bandwidth=3 datarate=50000 sync_word=0/0
2020-11-06 09:04:09.244 [RAL:VERB] SX130x LBT not enabled
2020-11-06 09:04:09.245 [RAL:VERB] Station device: /dev/spidev0.0 (PPS capture disabled)
lgw_start:846: Note: calibration started (time: 2300 ms)
lgw_start:867: Note: calibration finished (status = 191)
Info: Initialising AGC firmware...
Info: putting back original RADIO_SELECT value
2020-11-06 09:04:12.180 [RAL:VERB] Concentrator started (2s934ms)
2020-11-06 09:04:12.180 [S2E:INFO] Configuring for region: EU863 -- 863.0MHz..870.0MHz
2020-11-06 09:04:12.180 [S2E:VERB]   DR0  SF12/BW125
2020-11-06 09:04:12.180 [S2E:VERB]   DR1  SF11/BW125
2020-11-06 09:04:12.180 [S2E:VERB]   DR2  SF10/BW125
2020-11-06 09:04:12.180 [S2E:VERB]   DR3  SF9/BW125
2020-11-06 09:04:12.180 [S2E:VERB]   DR4  SF8/BW125
2020-11-06 09:04:12.180 [S2E:VERB]   DR5  SF7/BW125
2020-11-06 09:04:12.180 [S2E:VERB]   DR6  SF7/BW250
2020-11-06 09:04:12.180 [S2E:VERB]   DR7  FSK
2020-11-06 09:04:12.180 [S2E:VERB]   DR8  FSK
2020-11-06 09:04:12.180 [S2E:VERB]   DR9  FSK
2020-11-06 09:04:12.180 [S2E:VERB]   DR10 FSK
2020-11-06 09:04:12.180 [S2E:VERB]   DR11 FSK
2020-11-06 09:04:12.180 [S2E:VERB]   DR12 FSK
2020-11-06 09:04:12.180 [S2E:VERB]   DR13 FSK
2020-11-06 09:04:12.180 [S2E:VERB]   DR14 FSK
2020-11-06 09:04:12.180 [S2E:VERB]   DR15 FSK
[... 74 lines cut ...]
pi@ttn-gateway:~/basicstation/deps/lgw/platform-rpi/libloragw $ ./test_loragw_spi
Beginning of test for loragw_spi.c
[... cut ...]
lgw_spi_wb:309: BURST WRITE: to trans 1476 # chunk 1024 # transferred 2048
lgw_spi_wb:309: BURST WRITE: to trans 452 # chunk 452 # transferred 2500
Note: SPI burst write success
lgw_spi_rb:371: BURST READ: to trans 2500 # chunk 1024 # transferred 1024
lgw_spi_rb:371: BURST READ: to trans 1476 # chunk 1024 # transferred 2048
lgw_spi_rb:371: BURST READ: to trans 452 # chunk 452 # transferred 2500
Note: SPI burst read success
Note: SPI read success
data received (simple read): 48
Note: SPI port closed
End of test for loragw_spi.c
```

Thanks in advance

beitler, 2020-11-10:
Ok, so apparently your radio now starts successfully:
```
2020-11-06 09:04:09.245 [RAL:VERB] Station device: /dev/spidev0.0 (PPS capture disabled)
lgw_start:846: Note: calibration started (time: 2300 ms)
lgw_start:867: Note: calibration finished (status = 191)
Info: Initialising AGC firmware...
Info: putting back original RADIO_SELECT value
2020-11-06 09:04:12.180 [RAL:VERB] Concentrator started (2s934ms)
```

Oliv4945, 2020-11-10:
Hi @Manojkumar226 your gateway is now working. What did you changed ?
You even received a packet:

```
2020-11-06 09:05:21.859 [S2E:VERB] RX 868.5MHz DR5 SF7/BW125 snr=0.0 rssi=-116 xtime=0x310000059F1413 - updf mhdr=40 DevAddr=26085A47 FCtrl=80 FCnt=80 FOpts=[] 01D3CB68AD mic=-1297990206 (17 bytes)
```

If you did not changed any setting I guess that you might want to double check your wires/connectors and power supply

ogimenez-smtc, 2020-11-10:
Hi @Manojkumar226 your gateway is now working. What did you changed ?
You even received a packet:

```
2020-11-06 09:05:21.859 [S2E:VERB] RX 868.5MHz DR5 SF7/BW125 snr=0.0 rssi=-116 xtime=0x310000059F1413 - updf mhdr=40 DevAddr=26085A47 FCtrl=80 FCnt=80 FOpts=[] 01D3CB68AD mic=-1297990206 (17 bytes)
```

If you did not changed any setting I guess that you might want to double check your wires/connectors and power supply

closed 2022-01-31 by beitler