lorabasics/basicstation #93: Disable Remote Commands and Remote Shell

vivTob, 2021-03-02:
Is there a way to disable the remote shell and remote commands in station.conf or is patch needed?
Even if i understand the benefits, there are still risks that it will be misused. We use Chirpstack so security depends on access to the MQTT broker.

beitler, 2021-03-08:
The remote shell feature can be disabled at compile time using the `no_rmtsh` compile time configuration parameter. Here is what needs to be changed in setup.gmk in order to configure the `std` build variant without an interactive shell:

```patch
diff --git a/setup.gmk b/setup.gmk
index a123556..6e0d211 100644
--- a/setup.gmk
+++ b/setup.gmk
@@ -91,7 +91,7 @@ CFG.testsim = logini_lvl=DEBUG selftests tlsdebug lgwsim ral_lgw
 CFG.testms  = logini_lvl=DEBUG selftests tlsdebug lgwsim ral_master_slave
 CFG.testfs  = logini_lvl=DEBUG selftests tlsdebug lgwsim ral_lgw
 CFG.testpin = logini_lvl=INFO tlsdebug ral_lgw testpin
-CFG.std     = logini_lvl=INFO tlsdebug ral_lgw
+CFG.std     = logini_lvl=INFO tlsdebug ral_lgw no_rmtsh
 CFG.stdn    = logini_lvl=INFO tlsdebug ral_master_slave
 CFG.debug   = logini_lvl=DEBUG selftests tlsdebug ral_lgw
 CFG.debugn  = logini_lvl=DEBUG selftests tlsdebug ral_master_slave
```

Currently, there are no options to disable remote commands. The best option for you is to run basic station in a restricted environment with only the minimal amount of required execution privileges.

If I understand correctly, your concerns arise from a threat model in which a malicious actor has control over the LNS. Since the connection to the LNS is authenticated, it will be very hard to hijack the connection and execute malicious commands from places outside the true LNS. However, this is not a threat model that Basic Station is targeting. Basic Station gives the LNS connection a lot of power to be enable remote inspection, management and intervention tasks.

vivTob, 2021-04-12:
Thanks for your answer anton! 

In the threatening scenario all gateway data is published on an MQTT-Broker by the Chirpstack-Gateway-Bridge, so this could be the point of attack. But you are abslolutely right, in this case it is not a threat model that Basic Station should target.

closed 2021-04-12 by vivTob