lorabasics/basicstation #94: Make CUPS/LNS configuration backup more robust

mullerch, 2021-03-05:
There are actually some scenarios where the configuration files backup process can lead to empty backup files. If those are empty, the Basic Station will report that the backup file content is not valid and it will terminate (e.g. the URL format for uri files).

Problem probably comes from here: https://github.com/lorabasics/basicstation/blob/bd17e53ab1137de6abb5ae48d6f3d52f6c268299/src/sys.c#L368. If a files cannot be written, it will exit without syncing. Already written files can therefore be truncated without being effectively written. 

1. Config file backup process should be fixed.
2. An empty, corrupted or wrongly formatted backup file should be reported as a warning and the backup feature should not be used. This could ensure that the backup feature will actually never prevent the normal process to work.

beitler, 2021-03-05:
Thank you for submitting the issue. I agree that the credentials set backup process must be robust and not prevent normal operation. 

Actually, files should not be overwritten during the update process. Instead, new temporary files are created with the updated content and then renamed to the regular names. Marker files are used to control the state of the transaction in order to be able to roll back or roll forward from any possible inconsistent state.

Therefore, the fact that you end up with empty files during that process points to a bug. 

> There are actually some scenarios where the configuration files backup process can lead to empty backup files.

Could you please elaborate a bit on those scenarios and how they could be reproduced? 

> If a files cannot be written, it will exit without syncing. Already written files can therefore be truncated without being effectively written.

At `basicstation/src/sys.c: 368` the marker file (`{cups,tc}-temp.{upd,cpy,done}`) is not yet unlinked and basic station will attempt a roll forward upon next restart. In the error scenarios that you encounter, is that file not present?

mullerch, 2021-03-05:
Thanks for your quick answer.

Sadly I have no reproduction scenario, the issue has been reported to me. The report was that the Basic Station was suddenly reporting the following error after normal operation for some time:
```
[AIO:ERRO] Malformed URI - expecting ws://.. or wss://.. but found:
/tc-bak.uri: Misconfigured URI - expecting scheme ws:
```
I cannot say if cpy file was present or not.

As there is nothing on the system using this file, my assumption is that the issue comes from the Basic Station. A dive into the code let me think there might be an issue but I did not clearly understand the transaction system. I've tested it and that looks good so even if it exits without syncing (which still feels weird) it should not lead to this issue.

What I cannot understand is how sync() is supposed to be defined. Regarding https://github.com/lorabasics/basicstation/blob/bd17e53ab1137de6abb5ae48d6f3d52f6c268299/src/sys.c#L54 it just does nothing. Can you please clarify? 
Btw, the macro prototype suggests this a filesystem-wide sync, if that's the case I would rather suggest syncing files only (fsync) or using synchronous writes (O_SYNC).

beitler, 2021-03-08:
You are right. The fact that `sync` is not wired back to the corresponding system call may explain the existence of empty files after system resets during file transactions. This is a bug that will be fixed in the next release. I appreciate your attention to this matter.

The credentials set transactions consist of various fs operations like write, unlink and rename. The most straight forward way to commit all these fs operations to disk is the `sync` system call. Other approaches are feasible. However, they increase code complexity and provide only marginal improvement in return, given that changes to the credentials sets are happening very rarely. Wouldn't you agree? Nevertheless, if we see usage patterns changing, we could definitely revisit the credentials set management approach.

mullerch, 2021-03-08:
Thanks for the clarification. Do you have an ETA for next release?

I do not agree this would raise complexity. Using synchronous writes is the less complex and the less efficient (regarding memory lifetime saving). sync and fsync complexity is equal as in both you have to consider external system-wide sync. However, if writes are done rarely (which should be), I agree priority is lower.