lorabasics/basicstation #96: Bootstrap "credentials" files usage

mullerch, 2021-03-09:
I do not understand "-boot" file usage.

Regarding the documentation: https://doc.sm.tc/station/credentials.html#categories-and-sets

> tc.\* / cups.\*: Regular credentials used to connect to LNS or CUPS, respectively. If these credentials are not available or do not work, the bak and boot variants will be tried. These files may be part of a system image or may be updated by the Station process during an update session with a CUPS server.
> ...
> tc-boot.\* / cups-boot.\*: Bootstrap credentials are optional, they are used for an initial connection to a server. The server may limit the validity of bootstrap credentials after a few uses. These credentials are again used as a fallback if the regular credentials fail to work, or if no regular credentials are available. These credentials must be provided as part of the system image and may be replaced or updated as part of firmware updates. The Station never modifies these files.

This would mean that we can give only "-boot" files and the station will start and uses these to connect. But if I do so, I get an error:
> [TCE:ERRO] No TC URI configured

Also, I cannot find a meaning to this sentence:
> The server may limit the validity of bootstrap credentials after a few uses.

This doesn't specify anything about the basic station use of the files but suggest that there is some undocumented behavior regarding this use case.

beitler, 2022-01-31:
> This would mean that we can give only "-boot" files and the station will start and uses these to connect.

This is true. However, the state engine first goes through a fallback mechanism in which it first tries the regular and bak credentials before resorting to boot. This incurs a delay upon first startup based on boot-only. Are you concerned about this delay?

>> The server may limit the validity of bootstrap credentials after a few uses.

>This doesn't specify anything about the basic station use of the files but suggest that there is some undocumented behavior regarding this use case. 

It's true that this sentence is not related to station's use of the bootstrap credentials. It just points out that it's good practice to invalidate the bootstrap credentials after they've been used once for a particular device in case the bootstrap credentials are not device-specific. I agree that this sentence does not add much substance, so we'll probably remove it.

mullerch, 2022-02-01:
> This is true.

I was not able to reproduce a working behavior. If I give only the tc-boot.uri, I get the following:

```
2022-02-01 09:51:14.376 [SYS:INFO] Logging : stderr (maxsize=10485760, rotate=3)
2022-02-01 09:51:14.396 [SYS:INFO] Station Ver : 2.0.5(lorix/std) 2021-05-12 08:38:03
2022-02-01 09:51:14.403 [SYS:INFO] Package Ver : (null)
2022-02-01 09:51:14.405 [SYS:INFO] proto EUI : 0:fcc2:3d0b:6866 (/sys/class/net/eth0/address)
2022-02-01 09:51:14.407 [SYS:INFO] prefix EUI : ::1 (builtin)
2022-02-01 09:51:14.408 [SYS:INFO] Station EUI : fcc2:3dff:fe0b:6866
2022-02-01 09:51:14.410 [SYS:INFO] Station home: ./ (builtin)
2022-02-01 09:51:14.410 [SYS:INFO] Station temp: /var/tmp/ (builtin)
2022-02-01 09:51:14.410 [SYS:WARN] Station in NO-CUPS mode
2022-02-01 09:51:14.669 [TCE:INFO] Starting TC engine
2022-02-01 09:51:14.674 [TCE:ERRO] No TC URI configured
2022-02-01 09:51:14.675 [TCE:INFO] INFOS reconnect backoff 0s (retry 0)
2022-02-01 09:51:14.678 [TCE:ERRO] No TC URI configured
2022-02-01 09:51:14.678 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
2022-02-01 09:51:24.679 [TCE:ERRO] No TC URI configured
2022-02-01 09:51:24.680 [TCE:INFO] INFOS reconnect backoff 20s (retry 2)
2022-02-01 09:51:44.681 [TCE:ERRO] No TC URI configured
2022-02-01 09:51:44.681 [TCE:INFO] INFOS reconnect backoff 30s (retry 3)
2022-02-01 09:52:14.683 [TCE:ERRO] No TC URI configured
2022-02-01 09:52:14.683 [TCE:INFO] INFOS reconnect backoff 40s (retry 4)
2022-02-01 09:52:54.684 [TCE:ERRO] No TC URI configured
2022-02-01 09:52:54.685 [TCE:INFO] INFOS reconnect backoff 50s (retry 5)
2022-02-01 09:53:44.686 [TCE:ERRO] No TC URI configured
2022-02-01 09:53:44.686 [TCE:INFO] INFOS reconnect backoff 60s (retry 6)
2022-02-01 09:54:44.688 [TCE:ERRO] No TC URI configured
2022-02-01 09:54:44.688 [TCE:INFO] INFOS reconnect backoff 60s (retry 7)
2022-02-01 09:55:44.689 [TCE:ERRO] No TC URI configured
2022-02-01 09:55:44.689 [TCE:INFO] INFOS reconnect backoff 60s (retry 8)
```

Or maybe the delay you are talking about is greater than 5 minutes?

What I need is just a clear information on which file to set in which use case. I'm actually a bit confused and what I do (for integration) is to set both `tc.uri` and `tc-boot.uri` each time (I could also set `tc.uri` and delete `tc-boot.uri` for the same behavior I guess). No big deal, it could just be clarified (or fixed).

beitler, 2022-02-01:
Thanks for coming back on this. Indeed, the description above applies to the `cups-boot.*` credentials. It does not equally apply to `tc-boot.*` because the TC connection state engine resorts to CUPS when the TC credentials don't result in successful connection instead of using the TC bootstrap credentials.
However, you are demonstrating a valid case, in which station is started up in NO-CUPS mode. In that case, technically the `tc-boot.*` credentials should be taken into consideration. However, I'm struggling to see the use case for this approach, since it's unlikely that the gateway will be equipped with LNS-specific bootstrap credentials from the factory. In my view it's quite unlikely that an LNS is willing to issue such long-lived credentials. What do you think?

mullerch, 2022-02-01:
I could not identify a meaningful use case for the `tc-boot.uri` file in no CUPS mode. At first I probably thought this was the way to go to make a robust integration (files modified by external tools) but with your current description I understand I should set only the `tc.*` file for LNS and only the `cups-boot.*` (and delete `cups.*`.) for CUPS.

closed 2022-02-01 by mullerch

beitler, 2022-02-01:
Yes. We will add clarifying words to the documentation to make that clear. 
Thanks!

mullerch, 2022-02-01:
Thanks for your follow-up