lorabasics/basicstation #100: Install Basic Station on Kerlink ifemtoCell evolution gateway

jujia1987, 2021-03-28:
Hi expert,

Hope you are well.
I am now working on a kerlink ifemtoCell evolution gateway and want to install basic station for connecting to AWS IoT Core for LoRaWAN. But I have some problem of cross-compiling.
The toolchain that I use from kerlink wikipage is [poky-glibc-x86_64-keros-toolchain-klkgw-4.3.3] and I tried to install it based on their instruction (/opt/toolchains/klkgw) and also your instruction(~/platform-kerlink) but both of them has same problem.
I will take /opt/toolchains/klkgw as an example to show how I was doing...

1. Once the toolchain is installed, I did a sourcing on the environment file.
2. Download the basic station via git clone.
3. modify the setup.gmk.
    3.1 in line 53 change TOOLCHAIN=${HOME}/toolchain-${platform} to TOOLCHAIN=/opt/toolchains/klkgw
    3.2 in line 65 change the toolprefix=${TOOLCHAIN}/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/${ARCH}-
    3.3 I created a new platform poky following kerlink and below lines(I used platform kerlink but same problem with this.)
         *ARCH.poky = arm-poky-linux-gnueabi        
         *CFG.poky = linux lgw1 no_leds
        *DEPS.poky     = mbedtls lgw
        *LIBS.poky = -llgw   ${MBEDLIBS} -lrt -lpthread
4. Executing $make platform=poky and the error log showing below
     
_platform=poky variant=std make -C deps/mbedtls
make[1]: Entering directory `/home/ec2-user/basicstation/deps/mbedtls'
make -C platform-poky lib CC="/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc" AR="/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-ar" LD="/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-ld"
make[2]: Entering directory `/home/ec2-user/basicstation/deps/mbedtls/platform-poky'
make[3]: Entering directory `/home/ec2-user/basicstation/deps/mbedtls/platform-poky/library'
  CC    aes.c
In file included from /opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/syslimits.h:7:0,
                 from /opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/limits.h:34,
                 from ../include/mbedtls/check_config.h:36,
                 from ../include/mbedtls/config.h:2717,
                 from aes.c:29:
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/limits.h:194:61: error: no include path in which to search for limits.h
 #include_next <limits.h>  /* recurse down to the real one */
                                                             ^
aes.c:36:10: fatal error: string.h: No such file or directory
 #include <string.h>
          ^~~~~~~~~~
compilation terminated.
make[3]: *** [aes.o] Error 1
make[3]: Leaving directory `/home/ec2-user/basicstation/deps/mbedtls/platform-poky/library'
make[2]: *** [lib] Error 2
make[2]: Leaving directory `/home/ec2-user/basicstation/deps/mbedtls/platform-poky'
make[1]: *** [platform-poky/library/libmbedtls.a] Error 2
make[1]: Leaving directory `/home/ec2-user/basicstation/deps/mbedtls'
make: *** [deps/mbedtls] Error 2_

5. In order to fix the error, I run $export C_INCLUDE_PATH=/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/:/usr/include/ and make again but the error changed and seems like I need to install glibc 32 bits.
_platform=poky variant=std make -C deps/mbedtls
make[1]: Entering directory `/home/ec2-user/basicstation/deps/mbedtls'
make -C platform-poky lib CC="/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc" AR="/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-ar" LD="/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-ld"
make[2]: Entering directory `/home/ec2-user/basicstation/deps/mbedtls/platform-poky'
make[3]: Entering directory `/home/ec2-user/basicstation/deps/mbedtls/platform-poky/library'
  CC    aes.c
cc1: warning: include location "/usr/include/" is unsafe for cross-compilation [-Wpoison-system-directories]
In file included from /usr/include/features.h:447:0,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/limits.h:26,
                 from /opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/limits.h:194,
                 from /opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/syslimits.h:7,
                 from /opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/limits.h:34,
                 from ../include/mbedtls/check_config.h:36,
                 from ../include/mbedtls/config.h:2717,
                 from aes.c:29:
/usr/include/gnu/stubs.h:7:11: fatal error: gnu/stubs-32.h: No such file or directory
 # include <gnu/stubs-32.h>
           ^~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [aes.o] Error 1
make[3]: Leaving directory `/home/ec2-user/basicstation/deps/mbedtls/platform-poky/library'
make[2]: *** [lib] Error 2
make[2]: Leaving directory `/home/ec2-user/basicstation/deps/mbedtls/platform-poky'
make[1]: *** [platform-poky/library/libmbedtls.a] Error 2
make[1]: Leaving directory `/home/ec2-user/basicstation/deps/mbedtls'
make: *** [deps/mbedtls] Error 2_
 

To be honest, I am not so familiar with GCC environment so I was thinking I might be in the wrong direction.
Could you please give me some guidance how I can install basic station on kerlink ifemtoCell evolution gateway?
Thank you very much.

Regards,
Harry

jujia1987, 2021-03-29:
An update on my question.
I change the path with below and the error passed.
#export C_INCLUDE_PATH=/opt/toolchains/klkgw/sysroots/armv7ahf-neon-poky-linux-gnueabi/usr/include/

Also, I change the stubs-soft.h to stubs-hard.h as there is stubs-soft.h doesn't exist. not sure about the impact.
Now during the compiling process, it stopped at below errors:
 /opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find crt1.o: No such file or directory
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find crti.o: No such file or directory
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find crtbegin.o: No such file or directory
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find -lrt
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find -lpthread
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find -lm
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find -lgcc
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find -lgcc_s
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find -lc
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find -lgcc
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find -lgcc_s
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find crtend.o: No such file or directory
/opt/toolchains/klkgw/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/real-ld: cannot find crtn.o: No such file or directory
collect2: error: ld returned 1 exit status
make[1]: *** [../bin/station] Error 1
make[1]: Leaving directory

beitler, 2021-05-03:
Your build process is failing at linking time. It seems the linker is not finding a the required object files that come with the toolchain. You probably need to pass the corresponding file locations via flags or environment variables.

asvand, 2021-09-20:
Hi  jujia1987 , did you have any success with installing Basic Station on Kerlink ifemtoCell evolution gateway?

beitler, 2022-01-31:
Please reach out to the Gateway manufacturer. I'm sure they are in the best position to help you out.

closed 2022-01-31 by beitler