lorabasics/basicstation #101: Error 'lgwx_device_mode'

MarkusKa, 2021-04-27:
i am trying to build the basicstation on the platform aarch64. however, the error appears when running the makefile:

make platform=aarch64

thanks for help
[image]

beitler, 2021-05-03:
Hello. `platform=aarch64` is not a supported platform as it is not listed in the setup.gmk file.
What is your diff to the clean master?

MarkusKa, 2021-05-03:
Hello, I have tried to add the architecture to the setup.gmk file. However, I have not been able to get it to work. Now it works with a few changes in the makefile and the architecture of the Rapi.

so with 
make platform=rpi variant=std

closed 2021-05-03 by MarkusKa