lorabasics/basicstation #102: Timesync UTC with Local Time

MarkusKa, 2021-05-03:
Hello all,

How can I synchronise the UTC time of the Basic station with the local time?

I thought it was done automatically by the offset in the src/rt.c file by "rt_utcOffset".

[image]

[image]

Thanks for Help

beitler, 2021-05-03:
Hello. Basic Station does not have a notion of local time. The log line time stamps are always in UTC, which is exactly what you are observing.

MarkusKa, 2021-05-03:
thanks for the information, so the basic station does not need to be synchronised with the local time.

Then I can close the issue

closed 2021-05-03 by MarkusKa