lorabasics/basicstation #103: Question: Old version of the sx1302 hal

bojczmac, 2021-05-04:
Why hal for the sx1302 was not updated to the latest version? Latest version is 2.1.0 but basicstation is still using 1.0.5.
Any update is planning?

beitler, 2021-05-04:
Yes, an update to the latest version of the sx1302_hal is planned. Which particular feature of version 2.1.0 are you looking for?

bojczmac, 2021-05-04:
I'm asking about update because I've some problems with my gateway.
This is custom platform based on iMX6 and RAK2287 (SX1302 & SX1250 radios).
With old Semtech packet forwarder everything works fine but I would like to use Basic Station.
I'm using smtech lora cloud for testing my basic station solution.

On below log you can see that gateway is able to receive lora packets but have some problems with sending responses. I don't know why. I'm guessing that this issue maybe connected with time sync problems. Take a look on the second log.
```
WARNING: not enough space allocated, fetched 2 packet(s), 1 will be left in RX buffer
Note: remaining 1 packets in RX buffer, do not fetch sx1302 yet...
2021-05-04 20:45:17.364 [any:XDEB] RX mod=LORA f=867300000 bw=-1096849104 sz=23 dr=12 0000000000000000004C5200FEFF091FAC0E008C8481C4
2021-05-04 20:45:17.366 [any:XDEB] RX mod=LORA f=868300000 bw=10447152 sz=23 dr=12 0000000000000000004C5200FEFF091FAC0E008C8481C4
2021-05-04 20:45:17.366 [S2E:DEBU] Dropped mirror frame freq=867.3MHz snr=-12.5 rssi=-98 (vs. freq=868.3MHz snr= 13.0 rssi=-23) - DR0 mic=-998144884 (23 byes)
2021-05-04 20:45:17.371 [S2E:VERB] RX 868.3MHz DR0 SF12/BW125 snr=13.0 rssi=-23 xtime=0xFC0000701D61DA - jreq MHdr=00 JoinEui=::0 DevEui=ac1f:9ff:fe00:524c DevNonce=14 MIC=-998144884
2021-05-04 20:45:17.371 [AIO:XDEB] [3|WS] > {"msgtype":"jreq","MHdr":0,"JoinEui":"00-00-00-00-00-00-00-00","DevEui":"AC-1F-09-FF-FE-00-52-4C","DevNonce":14,"MIC":-998144884,"RefTime":0.000000,"DR":0,"Freq":8}
2021-05-04 20:45:17.644 [AIO:XDEB] [3|WS] < {"msgtype":"dnmsg","DevEui":"01-01-01-01-01-01-01-01","dC":0,"diid":44602,"pdu":"20ed491325e22d7c41fee353e0da802e80efaf4e8f865f31aaaf502bc50c61717f","priority":1,"}
2021-05-04 20:45:17.647 [S2E:DEBU] 101:101:101:101 diid=44602 [ant#0] - next TX start ahead by 4s691ms
2021-05-04 20:45:22.331 [S2E:ERRO] 101:101:101:101 diid=44602 [ant#0] - missed TX time: txdelta=8ms110us min=10ms
2021-05-04 20:45:22.331 [S2E:VERB] 101:101:101:101 diid=44602 [ant#0] - trying RX2 869.525MHz DR0
2021-05-04 20:45:22.331 [S2E:DEBU] 101:101:101:101 diid=44602 [ant#0] - next TX start ahead by 1s8ms
2021-05-04 20:45:23.319 [S2E:VERB] 101:101:101:101 diid=44602 [ant#0] - starting TX in 19ms673us
2021-05-04 20:45:23.351 [S2E:ERRO] 101:101:101:101 diid=44602 [ant#0] - radio is not emitting frame - abandoning TX, trying alternative
2021-05-04 20:45:23.354 [S2E:VERB] 101:101:101:101 diid=44602 [ant#0] - class A has no more alternate TX time
2021-05-04 20:45:23.354 [S2E:WARN] 101:101:101:101 diid=44602 [ant#0] - unable to place frame
```
Time sync problem:
```
2021-05-04 20:47:26.713 [SYN:INFO] MCU/SX130X drift stats: min: -0.9ppm  q50: -7.6ppm  q80: -58.9ppm  max: -141.1ppm - threshold q90: -90.7ppm
2021-05-04 20:47:26.714 [SYN:INFO] Mean MCU drift vs SX130X#0: -19.5ppm
2021-05-04 20:47:40.411 [SYN:VERB] Time sync rejected: quality=1715 threshold=1692
2021-05-04 20:47:48.851 [SYN:INFO] Time sync qualities: min=769 q90=1715 max=7066 (previous q90=1692)
2021-05-04 20:47:53.795 [AIO:XDEB] [3|WS] < PING ()
2021-05-04 20:47:53.795 [AIO:XDEB] [3|WS] > PONG
```

AbhishekLakhara0209, 2021-06-14:
@beitler,
 HAL for the sx1302 have been updated to the latest version or not?

bojczmac, 2021-06-14:
I checked latest release (2.0.5) and it still used old sx1302 hal version.
I must check again if sx1302 will work with the latest release of the basicstation. Previous doesn't work.

beitler, 2021-06-15:
@bojczmac , your issue does not look like it's related to a particular version of sx1302_hal.  The logs related to time sync are just time sync reports. It shows that your system clock drifts with about 20ppm compared to the sx1302 clock. It also shows that an SPI transaction takes about 1.7 ms in most cases (time sync quality Q90). This is not great, but also not terrible.

The actual problem is here:
```
2021-05-04 20:45:17.647 [S2E:DEBU] 101:101:101:101 diid=44602 [ant#0] - next TX start ahead by 4s691ms
2021-05-04 20:45:22.331 [S2E:ERRO] 101:101:101:101 diid=44602 [ant#0] - missed TX time: txdelta=8ms110us min=10ms
```
This means at time `x:17.647` your downlink was scheduled for transmission `4s691ms` in the future, which is at `x:22.338`. Station will schedule a tx job `TX_AIM_GAP` before the TX time. However, logs indicate that the tx job is executed only `8ms110us` before the tx time which is less than your configured `TX_MIN_GAP`.

This timing graph illustrates what is happening:

[image]

This problem may be caused by latencies or jitter in your system. You can adjust the `TX_AIM_GAP` parameter to account for these influences. Try some higher values, like 30ms or 50ms.

bojczmac, 2021-06-15:
@beitler Thank you for detailed explanation.
My system is a custom pcb based on imx6ull processor with dedicated distro based on yocto.
I see that default spi clock frequency for sx1302 is set to 2MHz. I can try to use higher value. SX1302 supports up to 10MHz. 
I'll also try to increase the TX_AIM_GAP and check if it helps.

closed 2022-01-31 by beitler