lorabasics/basicstation #105: Feature request: send gateway GPS coordinates

KellyM001, 2021-05-17:
Hello,

I'd like to raise a feature request to add the ability for Basic Station protocol to support sending gateway GPS location information (lat, long, altitude) to the LNS.

I've been using Basic Station protocol on our Milesight gateways connected to The Things Stack, and it would be great for gateway to be able to send it's location automatically to the LNS so the gateway location can be viewed and/or retrieved from the LNS. For this to be able to occur in a standardised way, the first step is for Basic Station protocol to support this behaviour.

Thanks for the work done to date on this protocol, it's a great tool to have when creating secure, manageable LoRaWAN networks.

Kind regards,
Kelly

jpmeijers, 2021-07-12:
Indeed a feature that I miss coming from the Semtech UDP packet forwarder (and derivatives).

When this feature is added, I would like to see [GPSD](https://gpsd.gitlab.io/gpsd/) being used on Linux hosts:
https://github.com/kersing/packet_forwarder/issues/2

Also, and accuracy of the GPS fix in metres, not hdop, would be useful.
https://github.com/packetbroker/api/issues/32

beitler, 2021-07-12:
Thank you for submitting this feature request.

Basic Station already ships with such a feature. Have a look at [gps.c](https://github.com/lorabasics/basicstation/blob/master/src-linux/gps.c). If you specify the `gps` field in station.conf with a path to the GPS serial device (or a fifo / named pipe), Basic Station will start consuming NMEA (or ubx) sentences and report position changes as `msgtype:alarm` messages to the LNS.

Note, that this feature is considered an _experimental convenience function_ for the following reason:

Basic Station does not promote a model in which the `station` process 'owns' the GPS device. For many applications, gateway operators have good reasons to tailor the exact behavior by which the GPS position information is collected, processed and filtered (this applies to other gateway health related metrics as well). Under this assumption, Basic Station will never achieve the level of flexibility to cover all possible use cases.
Therefore, the recommended mode of operation is to have the GPS device owned by a distinct OS-level process (e.g. a python script) which collects, processes, enriches, filters, aggregates the data (potentially together with other system metrics) and reports related events to the LNS when needed via the [cmd FIFO](https://doc.sm.tc/station/conf.html?highlight=fifo).

This discussion is very much related to the 'health status message' discussion. [Please read this comment](https://github.com/lorabasics/basicstation/issues/46#issuecomment-782753877) for more details.

KellyM001, 2021-08-20:
Thanks for your response @beitler, I was not aware the GPS information could already be sent via Basic Station in this manner. So if my understanding is correct, Basic Station provides a transportation mechanism with which the gateway's GPS information can be sent through. Does this mean you're proposing to not provide a well defined GPS message as part of the Basic Station protocol spec?

If so I expect this will mean each gateway vendor may send their GPS data in different formats, potentially resulting in LNS servers having to implement specific GPS message formatters for different gateway vendors, if the LNS is to process and understand the GPS data. Or am I misunderstanding something?

tkerby, 2021-11-02:
It would be very useful if the GPS and pps variables could be set through environment variables for testing this, particularly when using docker