lorabasics/basicstation #109: Timesync with SX1302 HAL 2.1.0 with basic station.

AbhishekLakhara0209, 2021-06-15:
We have manually added support of SX1302 HAL 2.1.0 with basic station. We are facing issue of time sync. We are not getting proper pps time.

**is new HAL of SX1302 supported by basic station?** 

**Also we are planning to use fine time stamp feature in basic station so is it supported with current basic station version?**

**Are you planning to add support for SX1302 HAL 2.1.0 with basic station?**

beitler, 2021-06-15:
Thanks for reaching out. Can you please provide more detail on what you did, what you are trying to achieve and what the exact error condition is? Concretely:

> We have manually added support of SX1302 HAL 2.1.0 with basic station. We are facing issue of time sync. We are not getting proper pps time.

What is the issue with time sync? Do you get error messages? How do they look like? What do you mean with 'we are not getting proper pps time'? Did [this issue](https://github.com/lorabasics/basicstation/issues/98) not address your problem?

> **is new HAL of SX1302 supported by basic station?**

Currently, Basic Station runs against sx1302_hal version `1.0.5`. 

> **Also we are planning to use fine time stamp feature in basic station so is it supported with current basic station version?**

How does your setup look like? Which hardware do you use to achieve fine time stamp? Are you using [sx1303](https://www.semtech.com/products/wireless-rf/lora-core/sx1303)?

> **Are you planning to add support for SX1302 HAL 2.1.0 with basic station?**

Support for newer sx1302_hal versions will be added. Which feature of the sx1302_hal 2.1.0 are you looking for?

AbhishekLakhara0209, 2021-06-17:
> Can you please provide more detail on what you did, what you are trying to achieve and what the exact error condition is?

We have linked latest HAL of SX1302 with basic station as we want to run latest HAL with basic station.

> What is the issue with time sync? Do you get error messages? How do they look like? What do you mean with 'we are not getting proper pps time'?

**Please find below logs snippet. We have used SX1303 concentrator board with raspberry pi (version b+ V1.2) and for gps we have used UBX7. We have run the setup for more than 2-3 hours and observed that GPS was locked and also LNS sending the time sync request but we are getting the same logs line "Time sync rejected: quality=114 threshold=100" more than hours.** 

021-06-15 07:09:44.391 [SYN:DEBU] Timesync #4179 sent to server
2021-06-15 07:09:44.414 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719378049,"gpstime":1307776198073377}
2021-06-15 07:09:44.414 [SYN:VERB] Timesync LNS: tx/rx:0x1CC14F1BD..0x1CC1CEC82 (522ms949us) us/gps:0x1CC1E0284/0x4A56A5FBB5C21 (pps offset=486084) - 1 solutions
2021-06-15 07:09:44.416 [SYN:ERRO] Timesync LNS: Too much drift between last PPS and inferred PPS: 1000041us
2021-06-15 07:09:44.416 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719449426,"gpstime":1307776198073680}
2021-06-15 07:09:44.416 [SYN:VERB] Timesync LNS: tx/rx:0x1CC16088E..0x1CC1CF1CA (452ms924us) us/gps:0x1CC1E0284/0x4A56A5FBB5D50 (pps offset=486084) - 1 solutions
2021-06-15 07:09:44.416 [SYN:ERRO] Timesync LNS: Too much drift between last PPS and inferred PPS: 1000041us
2021-06-15 07:09:44.417 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719520747,"gpstime":1307776198073815}
2021-06-15 07:09:44.417 [SYN:VERB] Timesync LNS: tx/rx:0x1CC171F27..0x1CC1CF577 (382ms544us) us/gps:0x1CC1E0284/0x4A56A5FBB5DD7 (pps offset=486084) - 1 solutions
2021-06-15 07:09:44.417 [SYN:ERRO] Timesync LNS: Too much drift between last PPS and inferred PPS: 1000041us
2021-06-15 07:09:44.417 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719592031,"gpstime":1307776198075562}
2021-06-15 07:09:44.417 [SYN:VERB] Timesync LNS: tx/rx:0x1CC18359B..0x1CC1CF69C (311ms553us) us/gps:0x0/0x4A56A5FBB64AA (pps offset=486084) - 0 solutions
2021-06-15 07:09:44.417 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719663342,"gpstime":1307776198075729}
2021-06-15 07:09:44.417 [SYN:VERB] Timesync LNS: tx/rx:0x1CC194C2A..0x1CC1CF749 (240ms415us) us/gps:0x0/0x4A56A5FBB6551 (pps offset=486084) - 0 solutions
2021-06-15 07:09:44.417 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719734693,"gpstime":1307776198075849}
2021-06-15 07:09:44.417 [SYN:VERB] Timesync LNS: tx/rx:0x1CC1A62E1..0x1CC1CF7F3 (169ms234us) us/gps:0x0/0x4A56A5FBB65C9 (pps offset=486084) - 0 solutions
2021-06-15 07:09:44.417 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719805984,"gpstime":1307776198075957}
2021-06-15 07:09:44.418 [SYN:VERB] Timesync LNS: tx/rx:0x1CC1B795C..0x1CC1CF899 (98ms109us) us/gps:0x0/0x4A56A5FBB6635 (pps offset=486084) - 0 solutions
2021-06-15 07:09:44.418 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719877262,"gpstime":1307776198076067}
2021-06-15 07:09:44.418 [SYN:VERB] Timesync LNS: tx/rx:0x1CC1C8FCA..0x1CC1CF93C (26ms994us) us/gps:0x0/0x4A56A5FBB66A3 (pps offset=486084) - 0 solutions
2021-06-15 07:09:44.462 [AIO:XDEB] [4|WS] > {"msgtype":"timesync","txtime":7719948546}
2021-06-15 07:09:44.462 [SYN:DEBU] Timesync #4180 sent to server
2021-06-15 07:09:44.504 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7719948546,"gpstime":1307776198129680}
2021-06-15 07:09:44.504 [SYN:VERB] Timesync LNS: tx/rx:0x1CC1DA63E..0x1CC1E4A36 (41ms976us) us/gps:0x0/0x4A56A5FBC3810 (pps offset=486084) - 0 solutions
2021-06-15 07:09:44.533 [AIO:XDEB] [4|WS] > {"msgtype":"timesync","txtime":7720019843}
2021-06-15 07:09:44.533 [SYN:DEBU] Timesync #4181 sent to server
2021-06-15 07:09:44.536 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7720019843,"gpstime":1307776198196320}
2021-06-15 07:09:44.536 [SYN:VERB] Timesync LNS: tx/rx:0x1CC1EBCBF..0x1CC1EC792 (2ms771us) us/gps:0x0/0x4A56A5FBD3C60 (pps offset=486084) - 0 solutions
2021-06-15 07:09:44.669 [SYN:CRIT] pps_xtime=24206852507116980,pps_xticks=80341147352901044
2021-06-15 07:09:44.670 [SYN:INFO] Time sync qualities: min=302 q90=349 max=489 (previous q90=355)
2021-06-15 07:09:44.670 [SYN:ERRO] No PPS pulse for ~-53 secs
2021-06-15 07:09:44.706 [GPS:XDEB] GPS: $GLGSV,3,1,09,68,18,072,22,69,20,125,,73,60,210,10,74,48,309,18*6B
2021-06-15 07:09:44.739 [GPS:XDEB] GPS: $GLGSV,3,2,09,75,01,334,,80,18,171,,83,18,034,,84,41,329,*68
2021-06-15 07:09:44.822 [GPS:XDEB] GPS: $GLGSV,3,3,09,85,14,270,*51
2021-06-15 07:09:44.872 [GPS:XDEB] GPS: $GNGLL,2232.17140,N,07255.61330,E,070943.00,A,A*7E
2021-06-15 07:09:44.922 [GPS:DEBU] UBX cksum=C473 vs found=C473
2021-06-15 07:09:44.922 [GPS:DEBU] NAV-TIMEGPS tow(ms)=198602000.476126 week=7 leapsecs=7 status=0 tacc=118622322
2021-06-15 07:09:46.771 [SYN:CRIT] pps_xtime=24206852643334708,pps_xticks=13735155604499966516
2021-06-15 07:09:46.771 [SYN:ERRO] No PPS pulse for ~-51 secs
2021-06-15 07:09:48.873 [SYN:ERRO] No PPS pulse for ~-49 secs
2021-06-15 07:09:49.534 [SYN:DEBU] Timesync #4182 sent to server
2021-06-15 07:09:49.536 [AIO:XDEB] [4|WS] < {"msgtype":"timesync","txtime":7725020143,"gpstime":1307776203196468}
2021-06-15 07:09:49.536 [SYN:VERB] Timesync LNS: tx/rx:0x1CC6B092B..0x1CC6B139A (2ms671us) us/gps:0x0/0x4A56A60098834 (pps offset=486084) - 0 solutions
[... cut ...]
2021-06-16 09:14:51.886 [SYN:VERB] Timesync LNS: tx/rx:0x1CAC52158..0x1CAC52C48 (2ms800us) us/gps:0x0/0x4A5803CFAAB2E (pps offset=316569) - 0 solutions
2021-06-16 09:14:51.955 [SYN:DEBU] Timesync #144 sent to server
2021-06-16 09:14:51.959 [SYN:VERB] Timesync LNS: tx/rx:0x1CAC637EC..0x1CAC6482F (4ms163us) us/gps:0x0/0x4A5803CFBC3E8 (pps offset=316569) - 0 solutions
2021-06-16 09:14:52.026 [SYN:DEBU] Timesync #145 sent to server
2021-06-16 09:14:52.954 [SYN:ERRO] No PPS pulse for ~11 secs

**We have run this setup for more than 2hours**

AbhishekLakhara0209, 2021-06-21:
Hi @beitler,
 
Any update on this?

beitler, 2021-07-15:
Hi @AbhishekLakhara0209 ,

the logs indicate that there is something wrong with your setup.

For example:
```
2021-06-16 09:14:51.959 [SYN:VERB] Timesync LNS: tx/rx:0x1CAC637EC..0x1CAC6482F (4ms163us) us/gps:0x0/0x4A5803CFBC3E8 (pps offset=316569) - 0 solutions
```

Which translates into:
```
txtime (PPS-aligned) :          7696s 955372us
rxtime (PPS-aligned) :          7696s 959535us
gpstime (from server):    1307870104s 372200us
us diff:               -585253us
```
This shows that the sub-second offset between the local PPS and the server's GPS time is off by more than half a second. Hence, either the gateway's PPS is off or the server's GPStime is not synchronized. 
The consequence is that station is unable to infer the GPS time associated to the last observed PPS due to the misalignment between the locally observed PPS and the server's reported GPS time.

```
2021-06-16 09:14:52.954 [SYN:ERRO] No PPS pulse for ~11 secs
```
This indicates that the PPS on your gateway is unstable. Make sure you have good skyview. PPS should be stable for reliable gateway operation.

In summary, please review your setup: make sure that the server is synchronized with millisecond precision (NTP should suffice) and make sure your gateway's GPS antenna has good view of the sky.

AbhishekLakhara0209, 2021-07-19:
@beitler,
Thanks for the update.
I have gone through the code of timesync related functionality, we have observed that **pps_xtick** is changing when we enable below condition.
**file name : ral_lgw.c** 
```
int ral_getTimesync (u1_t pps_en, sL_t last_xtime, timesync_t timesync){
.
.
.
#if defined(CFG_sx1302)
        sx1302_gps_enable(false); // Observed mis-behaviour when we enable this conditions.
#else
        lgw_reg_w(LGW_GPS_EN, 0);       // PPS latch holds current
#endif
.
.
.
#if defined(CFG_sx1302)
        sx1302_gps_enable(true); // Observed mis-behaviour when we enable this conditions.
#else
        lgw_reg_w(LGW_GPS_EN, 1);
#endif

}
```
what is the meaning of this condition?

Why we are enabling/disabling the GPS?

Please find below logs of issue:

```
value of stats->mcu_drifts[19] : (4)
value of drift_ppm(4),stats->drift_thres(139) 
2021-07-16 07:12:52.711 [SYN:INFO] pps_xticks : (201150634) **//pps_xticks and xtick has no major difference.**
2021-07-16 07:12:52.711 [SYN:INFO] xticks     : (201861763)
2021-07-16 07:12:52.711 [SYN:INFO] timesync->xtime     : (34339947360561795)
2021-07-16 07:12:52.711 [SYN:INFO] timesync->ustime    : (7787483238)
2021-07-16 07:12:52.711 [SYN:INFO] timesync->pps_xtime : (34339947359850666)
2021-07-16 07:12:52.711 [SYN:INFO] t0                  : (7787483091)
2021-07-16 07:12:52.711 [SYN:INFO] t1                  : (7787483386)
2021-07-16 07:12:52.711 [SYN:INFO] (t1-t0)             : (295)
2021-07-16 07:12:52.716 [SYN:INFO] MCU/SX130X drift stats: min: -0.3ppm  q50: -5.0ppm  q80: +11.3ppm  max: -16.1ppm - threshold q90: +13.9ppm
2021-07-16 07:12:52.716 [SYN:INFO] Mean MCU drift vs SX130X#0: 0.4ppm
value of syncQual[26] : (296)
value of quality (296), syncQual_thres (296)
value of stats->mcu_drifts[0] : (6)
value of drift_ppm(6),stats->drift_thres(139) 
xtime and pps_xtime diff is hige, (7995392) 
2021-07-16 07:12:54.406 [SYN:INFO] pps_xticks : (68932906) **//pps_xticks value changes suddenly.**
2021-07-16 07:12:54.406 [SYN:INFO] xticks     : (203557116)
2021-07-16 07:12:54.406 [SYN:INFO] timesync->xtime     : (34339947362257148)
2021-07-16 07:12:54.406 [SYN:INFO] timesync->ustime    : (7789178592)
2021-07-16 07:12:54.406 [SYN:INFO] timesync->pps_xtime : (34339947227632938)
2021-07-16 07:12:54.407 [SYN:INFO] t0                  : (7789178444)
2021-07-16 07:12:54.407 [SYN:INFO] t1                  : (7789178740)
2021-07-16 07:12:54.407 [SYN:INFO] (t1-t0)             : (296)
value of syncQual[27] : (223)
```
If we commented the condition of enable/disable gps from **ral_getTimesync** API then we have not observed this kind of issue.

**Note: GPS was locked at that moment.**

beitler, 2022-01-31:
Please pull the latest release 2.0.6 which improved the handling of the timestamp for the sx1302_hal. This should resolve the issues you described.

closed 2022-01-31 by beitler